introlab/rtabmap #135: standalone binary release for windows crashes on some hardware configurations.

Lhun, 2016-10-28:
Apologies if this has been raised before, I was unable to find any mention of it. 

Getting application start error on windows, usually an issue with elevation but this doesn't appear to be the case here. 
Both windows 10 pro x64
On my home rig (amd 9590, fx990 chipset, amd RX480) no crash on startup, works great.
On work rig, (amd core i52320, Nvidia 9800gtx) crashes at startup with the following: 
[image]

matlabbe, 2016-10-30:
The 0x000007b error is caused by some 32 bits and 64 bits libraries loaded at the same time. With [dependency walker](http://www.dependencywalker.com/) on RTABMap.exe, you can find if there are some 32 bits dll loaded.

For example, I did have the same error on my computer having OpenCL installed from NVIDIA driver. The `opencl.dll` used was the 32 bits version (while an `opencl64.dll` is existing in the same installation directory), so I renamed the `opencl64.dll` to `opencl.dll` and it fixed this startup error. If you don't want to change dll in your system, you can just copy/rename `opencl64.dll` in RTAB-Map's bin directory so it is loaded on startup.

cheers

Lhun, 2016-10-30:
Very cool, I'll give that a shot.

josesevilla, 2017-02-07:
Also I got an error dialog for missing concrt140.dll right after my first install of windows binaries RTABMap-0.11.14-win64-ZED12-CUDA80.exe on windows7 profesional

josesevilla, 2017-02-07:
[image]

matlabbe, 2017-02-07:
For concrt140.dll missing, install visual studio 2015 redistributable: https://github.com/introlab/rtabmap/wiki/Installation#windows