introlab/rtabmap #136: Crashing during export to mesh + textures

Lhun, 2016-10-30:
Hi there again. I have an extremely capable system, and I'm getting a full crash when attempting to export a scan to mesh. 

It could be that it's too complex, I'm not sure. I'm looking to do simple room scans for use in visualization, and I'm having no luck so far. 

Using a Kinect360, what settings should I be using for visualization? The mapping is very rapid and impressive, but I'm not sure how to take this 5m+ point cloud and get it out of the software.

matlabbe, 2016-10-30:
Which version (on which system) are you using? You can start RTAB-Map from the console/terminal and set logger level to Debug (Preferences->Logging->Level). Try exporting the mesh and copy/part the terminal output here.

matlabbe, 2016-10-30:
A print screen of your Export dialog could also help to reproduce the problem.

Lhun, 2016-10-30:
Win64.
Thanks for the swift response. Creating a log right now.

Lhun, 2016-10-30:
okay, so nothing exciting, it seems to crash exporting the temp textures: 
```

[ INFO] (2016-10-30 17:40:38) util3d_surface.cpp:530::rtabmap::util3d::createTextureCameras() Saved temporary texture: "C:/Users/Lhun/Documents/RTAB-Map/tmp_textures/texture_282.png"

[ INFO] (2016-10-30 17:40:38) util3d_surface.cpp:530::rtabmap::util3d::createTextureCameras() Saved temporary texture: "C:/Users/Lhun/Documents/RTAB-Map/tmp_textures/texture_283.png"

[ INFO] (2016-10-30 17:40:38) util3d_surface.cpp:530::rtabmap::util3d::createTextureCameras() Saved temporary texture: "C:/Users/Lhun/Documents/RTAB-Map/tmp_textures/texture_284.png"

[ INFO] (2016-10-30 17:40:38) util3d_surface.cpp:530::rtabmap::util3d::createTextureCameras() Saved temporary texture: "C:/Users/Lhun/Documents/RTAB-Map/tmp_textures/texture_285.png"

[ INFO] (2016-10-30 17:40:38) util3d_surface.cpp:530::rtabmap::util3d::createTextureCameras() Saved temporary texture: "C:/Users/Lhun/Documents/RTAB-Map/tmp_textures/texture_28

```
[image]
[image]

Lhun, 2016-10-30:
Just trying to export a reasonable mesh - if anyone has any settings config saved for kinect360 that is known to work reasonably well with a 2-5 pass scan on a 4m square space to export as obj+mtl, I would be forever in your debt. Looking to get a workflow going for VR recreation and mapping.

matlabbe, 2016-10-30:
I cannot reproduce the crash, I'll try on Windows tomorrow.

You can also try the "Organized Cloud" reconstruction flavor, which gives a lot better looking textured meshes (note that there is an [issue](https://github.com/introlab/rtabmap/issues/137) on dense texture mesh exportation). With organized clouds, I recommend to use "Min polygon cluster size" instead of "Noise filtering" to remove noisy polygons. These parameters would give you nice results:
[image]

To reduce the number of polygons, use "3D cloud decimation" under "Regenerate clouds". You can also activate Preferences->General Settings->Cloud Filtering->Pose Filtering to avoid creating meshes for supperposed nodes.

Lhun, 2016-10-30:
That's fantastic, thank you, I'll give that a shot. Would you recommend orb
over grift in this case, for a purely visual aspect?

Lhun, 2016-10-30:
So I went ahead and did this, however, the output, rather then being an obj, ended up being several hundred ply files, and they were not assembled, remained as point clouds, and had no references to texture. Maybe I'm missing a step, apologies for the wet behind the ears here, but I assumed the output would be a single mesh+texture.

matlabbe, 2016-10-31:
I tried your settings on Windows (latest release 0.11.11), and I cannot reproduce the crash. Can you share your database?

For the organized cloud setting, yes by checking "Assemble clouds/meshes", only one mesh should be saved. If "Texture mapping" is checked, then an OBJ is generated instead of PLY. I tried my settings above and I get this OBJ (opened in MeshLab):
[image]

Can you show the output of the progress dialog? Mine is:
```
17:45:47 Copied cloud 1 from cache with 19200 points and 15182 indices (1/32).
17:45:47 Copied cloud 3 from cache with 19200 points and 15717 indices (2/32).
17:45:47 Copied cloud 4 from cache with 19200 points and 15542 indices (3/32).
17:45:47 Copied cloud 5 from cache with 19200 points and 15345 indices (4/32).
17:45:47 Copied cloud 6 from cache with 19200 points and 14094 indices (5/32).
17:45:47 Copied cloud 7 from cache with 19200 points and 13076 indices (6/32).
17:45:47 Copied cloud 8 from cache with 19200 points and 11870 indices (7/32).
17:45:47 Copied cloud 9 from cache with 19200 points and 14925 indices (8/32).
17:45:47 Copied cloud 10 from cache with 19200 points and 15451 indices (9/32).
17:45:47 Copied cloud 11 from cache with 19200 points and 15846 indices (10/32).
17:45:47 Copied cloud 12 from cache with 19200 points and 15785 indices (11/32).
17:45:47 Copied cloud 13 from cache with 19200 points and 16506 indices (12/32).
17:45:48 Copied cloud 14 from cache with 19200 points and 15970 indices (13/32).
17:45:48 Copied cloud 15 from cache with 19200 points and 15776 indices (14/32).
17:45:48 Copied cloud 16 from cache with 19200 points and 14345 indices (15/32).
17:45:48 Copied cloud 17 from cache with 19200 points and 11457 indices (16/32).
17:45:48 Copied cloud 18 from cache with 19200 points and 7987 indices (17/32).
17:45:48 Copied cloud 19 from cache with 19200 points and 9320 indices (18/32).
17:45:48 Copied cloud 20 from cache with 19200 points and 12137 indices (19/32).
17:45:48 Copied cloud 21 from cache with 19200 points and 13497 indices (20/32).
17:45:48 Copied cloud 22 from cache with 19200 points and 15132 indices (21/32).
17:45:48 Copied cloud 23 from cache with 19200 points and 15062 indices (22/32).
17:45:48 Copied cloud 24 from cache with 19200 points and 14431 indices (23/32).
17:45:48 Copied cloud 26 from cache with 19200 points and 12459 indices (24/32).
17:45:48 Copied cloud 27 from cache with 19200 points and 11668 indices (25/32).
17:45:48 Copied cloud 28 from cache with 19200 points and 13602 indices (26/32).
17:45:48 Copied cloud 29 from cache with 19200 points and 14394 indices (27/32).
17:45:48 Copied cloud 30 from cache with 19200 points and 15458 indices (28/32).
17:45:48 Copied cloud 31 from cache with 19200 points and 15736 indices (29/32).
17:45:48 Copied cloud 32 from cache with 19200 points and 15616 indices (30/32).
17:45:48 Copied cloud 33 from cache with 19200 points and 16489 indices (31/32).
17:45:48 Copied cloud 34 from cache with 19200 points and 16298 indices (32/32).
17:45:48 Gain compensation of 32 clouds...
17:45:49 Applying gain compensation...
17:45:49 Cloud 1 has gain 0.861344
17:45:49 Cloud 3 has gain 0.858267
17:45:49 Cloud 4 has gain 0.853592
17:45:49 Cloud 5 has gain 1.09896
17:45:49 Cloud 6 has gain 0.649708
17:45:49 Cloud 7 has gain 0.601025
17:45:49 Cloud 8 has gain 0.653189
17:45:49 Cloud 9 has gain 1.01786
17:45:49 Cloud 10 has gain 1.33935
17:45:49 Cloud 11 has gain 0.92196
17:45:49 Cloud 12 has gain 0.911898
17:45:49 Cloud 13 has gain 1.04661
17:45:49 Cloud 14 has gain 0.936551
17:45:49 Cloud 15 has gain 0.896505
17:45:49 Cloud 16 has gain 0.826169
17:45:49 Cloud 17 has gain 1.50536
17:45:49 Cloud 18 has gain 1.29203
17:45:49 Cloud 19 has gain 1.38673
17:45:49 Cloud 20 has gain 1.32677
17:45:49 Cloud 21 has gain 1.06434
17:45:49 Cloud 22 has gain 0.729587
17:45:49 Cloud 23 has gain 0.646278
17:45:49 Cloud 24 has gain 0.642561
17:45:49 Cloud 26 has gain 0.636722
17:45:49 Cloud 27 has gain 0.64603
17:45:49 Cloud 28 has gain 0.634262
17:45:49 Cloud 29 has gain 0.976196
17:45:49 Cloud 30 has gain 0.952556
17:45:49 Cloud 31 has gain 0.879196
[... 47 lines cut ...]
17:45:50 TextureMesh 13 created [cameras=1] (12/32).
17:45:50 TextureMesh 14 created [cameras=1] (13/32).
17:45:50 TextureMesh 15 created [cameras=1] (14/32).
17:45:50 TextureMesh 16 created [cameras=1] (15/32).
17:45:50 TextureMesh 17 created [cameras=1] (16/32).
17:45:50 TextureMesh 18 created [cameras=1] (17/32).
17:45:50 TextureMesh 19 created [cameras=1] (18/32).
17:45:50 TextureMesh 20 created [cameras=1] (19/32).
17:45:50 TextureMesh 21 created [cameras=1] (20/32).
17:45:50 TextureMesh 22 created [cameras=1] (21/32).
17:45:50 TextureMesh 23 created [cameras=1] (22/32).
17:45:50 TextureMesh 24 created [cameras=1] (23/32).
17:45:50 TextureMesh 26 created [cameras=1] (24/32).
17:45:50 TextureMesh 27 created [cameras=1] (25/32).
17:45:50 TextureMesh 28 created [cameras=1] (26/32).
17:45:50 TextureMesh 29 created [cameras=1] (27/32).
17:45:50 TextureMesh 30 created [cameras=1] (28/32).
17:45:50 TextureMesh 31 created [cameras=1] (29/32).
17:45:50 TextureMesh 32 created [cameras=1] (30/32).
17:45:50 TextureMesh 33 created [cameras=1] (31/32).
17:45:50 TextureMesh 34 created [cameras=1] (32/32).
17:45:50 Assembling 32 meshes...
17:45:54 Saving the mesh (with 32 textures)...
17:46:22 Saving the mesh (with 32 textures)... done.
```

Lhun, 2016-10-31:
I don't have access to that system at the moment, but IIRC it skips the
"assembling" portion at the end.

Lhun, 2016-10-31:
If you would like, I have another instance running here on alternate hardware (also 11.11 win64) I could try the database you just meshed with the same settings and see if I'm successful,

matlabbe, 2016-10-31:
I didn't keep the previous database, so I just created a new one: [mesh_test.db.zip](https://github.com/introlab/rtabmap/files/562699/mesh_test.db.zip)

Click on this image to see the OBJ in Sketchfab
[image]

Lhun, 2016-10-31:
okay wow, that's a trip--- It's exported perfectly on my machine here at
work, as an obj. Thanks so much - I must have changed something in the
settings or the scan is simply too big? I'm not sure. I'll try another
small export, and I'll reset all the settings.

I did notice that when I imported the db it had several settings different
then the defaults (I believe it said 4) so what I'll do is globally reset,
import your db, save the settings, then try an export over there. Cheers!
I'll let you know how it goes at home.

Lhun, 2016-10-31:
[image: Inline image 1]

Go figure. Happy halloween, and thanks again. Great piece of software you
have here.

Lhun, 2016-11-01:
Okay, so still getting issues at home. I have remote access today. 

The log did indeed show something different this time. 
Using your db, I get the following: 
[image]

Which then prompts to export as a ply rather then an obj. It's quite strange and I'm thinking I'll do a wipe and re-install just to be sure. 

First, the only difference between a master reset and your DB settings is: 
[image]

I also got this in the log after attempting an export: 

[image]

matlabbe, 2016-11-01:
There is a bug in the release version (it is fixed in source) in which MLS is still running even if Organized Cloud reconstruction is selected. To fix this, select dense cloud reconstruction flavor, this will make visible the MLS group box, then uncheck it. Select back organized cloud reconstruction flavor.

Lhun, 2016-11-01:
Okay, so, success! That was indeed the issue - For anyone else experiencing the same - I did the following: 

Uninstall of rtab map using the provided uninstaller. Cleared .rtabmap in the users directory and dumped my documents/rtab-map directory. 

Uninstalled openNi2, reinstalled the latest from their website. 
Re-installed Rtabmap 0.11.11. Imported the included database, saved the optimizer strategy changes and exported, successfully, to an obj. 

I then followed this: 
"There is a bug in the release version (it is fixed in source) in which MLS is still running even if Organized Cloud reconstruction is selected. To fix this, select dense cloud reconstruction flavor, this will make visible the MLS group box, then uncheck it. Select back organized cloud reconstruction flavor." 

and 
re-organizing now also seems to work, as well as decimation. I'm testing cloud filtering as well, but so far things seem great.

Thanks again to @matlabbe , I would say it's safe to say this is no longer an issue.  Well done, I can't wait for the fixes to dense!

matlabbe, 2016-11-01:
Issue #137 is fixed, I'll try to do a re-release soon of 0.11.11 (I'll make it clear on the [release](https://github.com/introlab/rtabmap/releases/tag/0.11.11) page when the binaries will be updated)

closed 2016-11-01 by matlabbe