introlab/rtabmap #138: Error in 'make -j4' RTAB-Map

kriszhao, 2016-10-31:
hello, @matlabbe ,

I have followed the instruction about installing RTAB-Map in wiki. But i always get error in this process. 

After I typed ''make -j4'......
Could you suggest me how to fix it?

thanks a lot,
Kris

`[  8%] [  8%] [  9%] [  9%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/FlannIndex.cpp.o
Building C object corelib/src/CMakeFiles/rtabmap_core.dir/sqlite3/sqlite3.c.o
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationVis.cpp.o
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/discrete_depth_distortion_model_helpers.cpp.o
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/clams/discrete_depth_distortion_model.cpp.o
In file included from /home/kris/rtabmap/corelib/src/rtflann/util/matrix.h:35:0,
                 from /home/kris/rtabmap/corelib/src/rtflann/flann.hpp:41,
                 from /home/kris/rtabmap/corelib/src/FlannIndex.cpp:31:
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h: In instantiation of ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   required from ‘void rtflann::serialization::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:51:40:   required from ‘static void rtflann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = rtflann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:298:61:   required from ‘Archive& rtflann::serialization::InputArchive<Archive>::operator&(T&) [with T = std::unordered_map<unsigned int, std::vector<unsigned int> >; Archive = rtflann::serialization::LoadArchive]’
/home/kris/rtabmap/corelib/src/rtflann/util/lsh_table.h:319:10:   required from ‘void rtflann::lsh::LshTable<ElementType>::serialize(Archive&) [with Archive = rtflann::serialization::LoadArchive; ElementType = float]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9:   required from ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = rtflann::lsh::LshTable<float>]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   [ skipping 6 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9:   required from ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = rtflann::LshIndex<rtflann::L2<float> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   required from ‘void rtflann::serialization::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = rtflann::LshIndex<rtflann::L2<float> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:51:40:   required from ‘static void rtflann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = rtflann::serialization::LoadArchive; T = rtflann::LshIndex<rtflann::L2<float> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:298:61:   required from ‘Archive& rtflann::serialization::InputArchive<Archive>::operator&(T&) [with T = rtflann::LshIndex<rtflann::L2<float> >; Archive = rtflann::serialization::LoadArchive]’
/home/kris/rtabmap/corelib/src/rtflann/algorithms/lsh_index.h:204:9:   required from ‘void rtflann::LshIndex<Distance>::loadIndex(FILE*) [with Distance = rtflann::L2<float>; FILE = _IO_FILE]’
/home/kris/rtabmap/corelib/src/FlannIndex.cpp:519:1:   required from here
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
         type.serialize(ar);
         ^
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h: In instantiation of ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   required from ‘void rtflann::serialization::serialize(Archive&, T&) [with Archive = rtflann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:56:56:   required from ‘static void rtflann::serialization::Serializer<T>::save(OutputArchive&, const T&) [with OutputArchive = rtflann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:316:61:   required from ‘Archive& rtflann::serialization::OutputArchive<Archive>::operator&(const T&) [with T = std::unordered_map<unsigned int, std::vector<unsigned int> >; Archive = rtflann::serialization::SaveArchive]’
/home/kris/rtabmap/corelib/src/rtflann/util/lsh_table.h:319:10:   required from ‘void rtflann::lsh::LshTable<ElementType>::serialize(Archive&) [with Archive = rtflann::serialization::SaveArchive; ElementType = float]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9:   required from ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::SaveArchive; T = rtflann::lsh::LshTable<float>]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   [ skipping 6 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9:   required from ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::SaveArchive; T = rtflann::LshIndex<rtflann::L2<float> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   required from ‘void rtflann::serialization::serialize(Archive&, T&) [with Archive = rtflann::serialization::SaveArchive; T = rtflann::LshIndex<rtflann::L2<float> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:56:56:   required from ‘static void rtflann::serialization::Serializer<T>::save(OutputArchive&, const T&) [with OutputArchive = rtflann::serialization::SaveArchive; T = rtflann::LshIndex<rtflann::L2<float> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:316:61:   required from ‘Archive& rtflann::serialization::OutputArchive<Archive>::operator&(const T&) [with T = rtflann::LshIndex<rtflann::L2<float> >; Archive = rtflann::serialization::SaveArchive]’
/home/kris/rtabmap/corelib/src/rtflann/algorithms/lsh_index.h:198:9:   required from ‘void rtflann::LshIndex<Distance>::saveIndex(FILE*) [with Distance = rtflann::L2<float>; FILE = _IO_FILE]’
/home/kris/rtabmap/corelib/src/FlannIndex.cpp:519:1:   required from here
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/FlannIndex.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/kris/rtabmap/corelib/src/rtflann/util/matrix.h:35:0,
                 from /home/kris/rtabmap/corelib/src/rtflann/flann.hpp:41,
                 from /home/kris/rtabmap/corelib/src/RegistrationVis.cpp:43:
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h: In instantiation of ‘static void rtflann::serialization::access::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:42:30:   required from ‘void rtflann::serialization::serialize(Archive&, T&) [with Archive = rtflann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:51:40:   required from ‘static void rtflann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = rtflann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
[... cut ...]
/home/kris/rtabmap/corelib/src/rtflann/algorithms/lsh_index.h:198:9:   required from ‘void rtflann::LshIndex<Distance>::saveIndex(FILE*) [with Distance = rtflann::L2<float>; FILE = _IO_FILE]’
/home/kris/rtabmap/corelib/src/RegistrationVis.cpp:1191:1:   required from here
/home/kris/rtabmap/corelib/src/rtflann/util/serialization.h:34:9: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationVis.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2`

matlabbe, 2016-10-31:
What is your machine? gcc version?

kriszhao, 2016-11-01:
thanks for quick reply! :)
gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) 
ubuntu 14.04 x64

matlabbe, 2016-11-01:
Can you show the output of cmake ?

kriszhao, 2016-11-03:
sure !
kris@kris:~/rtabmap/build$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenMP
-- Found OpenCV: /usr/local/include;/usr/local/include/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/indigo/include/libfreenect
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found octomap: /opt/ros/indigo/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.11
--   CMAKE_INSTALL_PREFIX = /home/kris/catkin_ws/devel
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = NO (not found, License: BSD)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kris/rtabmap/build

kriszhao, 2016-11-03:
in face,  i have reinstalled the same versiong of system and ros, but i still got errors in the process of "make -j4".

kriszhao, 2016-11-03:
some part of the error for this time.

In file included from /home/kris/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:42:0,
                 from /home/kris/rtabmap/corelib/src/CameraThread.cpp:31:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘void openni_wrapper::Image::fillRaw(unsigned char*) const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:121:38: error: ‘class xn::ImageMetaData’ has no member named ‘Data’
       memcpy (rgb_buffer, image_md_->Data (), image_md_->DataSize ());
                                      ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:121:58: error: ‘class xn::ImageMetaData’ has no member named ‘DataSize’
       memcpy (rgb_buffer, image_md_->Data (), image_md_->DataSize ());
                                                          ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getWidth() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:181:23: error: ‘class xn::ImageMetaData’ has no member named ‘XRes’
     return image_md_->XRes ();
                       ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getHeight() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:187:23: error: ‘class xn::ImageMetaData’ has no member named ‘YRes’
     return image_md_->YRes ();
                       ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getFrameID() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:193:23: error: ‘class xn::ImageMetaData’ has no member named ‘FrameID’
     return image_md_->FrameID ();
                       ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘long unsigned int openni_wrapper::Image::getTimeStamp() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:199:51: error: ‘class xn::ImageMetaData’ has no member named ‘Timestamp’
     return static_cast<unsigned long> (image_md_->Timestamp ());
                                                   ^
[ 11%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OptimizerGTSAM.cpp.o
In file included from /home/kris/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:42:0,
                 from /home/kris/rtabmap/corelib/src/CameraRGBD.cpp:28:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getHeight() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:188:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h: In member function ‘unsigned int openni_wrapper::Image::getWidth() const’:
/usr/include/pcl-1.7/pcl/io/openni_camera/openni_image.h:182:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2016-11-03:
Note that on my computer with 14.04.5 and gcc 4.8.4, it builds without errors:
```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Make sure to have latest Ubuntu updates. Otherwise, you can try without `-std=c++11`, to do so you must disable `g2o` ad `GTSAM` ([see these Travis logs for example](https://travis-ci.org/introlab/rtabmap/jobs/172683268)):
```bash
$ cd rtabmap/build
$ cmake -DWITH_G2O=OFF -DWITH_GTSAM=OFF ..
# make sure there is no -std=c++11 in CMAKE_CXX_FLAGS at the end of cmake output
$ make
```

kriszhao, 2016-11-03:
Thanks for suggestions.
For my errors, it seems to be my openni issue. because I have installed some version of openni for source.
code:
cd OpenNI-Bin-Dev-Linux-x64-v1.5.7.10
sudo ./install.sh

Can I uninstalled this version of openni? please give me some commands to try. 
Thanks a lot!

matlabbe, 2016-11-03:
Note that if you install `ros-indigo-openni-launch`, it will install openni automatically (though I think the same version is installed from ros indigo `libpcl-1.7-all-dev` package).

kriszhao, 2016-11-03:
Alright, I will try again later. Thanks very much!

kriszhao, 2016-11-04:
I am sorry for new error...

when i  try "$ cmake -DWITH_G2O=OFF -DWITH_GTSAM=OFF .."
 
It did work ! However, when I try to 
 "kris@kris:~/catkin_ws$ catkin_make"

following:
Base path: /home/kris/catkin_ws
Source space: /home/kris/catkin_ws/src
Build space: /home/kris/catkin_ws/build
Devel space: /home/kris/catkin_ws/devel
Install space: /home/kris/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/kris/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/kris/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kris/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;genlisp;genpy
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/kris/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kris/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

kriszhao, 2016-11-04:
And then :
kris@kris:~/catkin_ws/build$ make cmake_check_build_system
-- Using CATKIN_DEVEL_PREFIX: /home/kris/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kris/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;genlisp;genpy
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/kris/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kris/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1

matlabbe, 2016-11-04:
Install move_base_msgs:
```bash
$ sudo apt-get install ros-indigo-move-base-msgs
```
Do the same for any other missing ros dependencies.
cheers

kriszhao, 2016-11-05:
Hi !
After  I installed these ros dependencies, and i follow the[http://official-rtab-map-forum.67519.x6.nabble.com/Rtabmap-error-in-process-rtabmap-rgbd-odometry-1-ubuntu-on-raspberry-pi-td444.html](url), the code success.

Then , when I tried to update, one error followed with the command:`make install`:

CMake Error at cmake_install.cmake:63 (FILE):
  file Internal CMake error when trying to open file:
  /home/kris/rtabmap/build/install_manifest.txt for writing.

make: *** [install] Error 1

kriszhao, 2016-11-05:
Thanks very much for your help!

closed 2016-11-09 by matlabbe