introlab/rtabmap #139: Possible vocabulary additions?

Lhun, 2016-11-01:
Hi again, last time for a while, I promise. :)

Due to my recent successes with RTAB and the excellent attitude and active team behind it, I've become very interested in integrating it as part of my workflow in personal projects. 

I know over at Introlab the focus is for autonomy in robot vision and localization, and not as much aesthetically pleasing visualization (as there are many other tools for that, but correct me if I'm wrong) however it seems clear from some of the issues on this git that people have become interested in using it for general 3d model visualization due to it's ease of use and broad compatibility with hardware and OS support. 

Are there any plans to add more recently developed rgb-d/stereo/monocular dense/robust slam approaches? In particular the work of T. Whelan (who now works for Oculus: 

https://github.com/mp3guy/ElasticFusion
https://github.com/mp3guy/Kintinuous
or:
https://github.com/raulmur/ORB_SLAM2 (apologies if your implementation of ORB is actually using v2, I'm not sure) 

I've been compiling these approaches manually in VM and bare metal installs to test them, but it would be fantastic to have various bag of words approaches based on the target use of the output of a dataset from within a central tool like RTAB. Perhaps I'm asking too much. :)

matlabbe, 2016-11-02:
Hi,

Indeed, my focus is more on online long-term and large-scale continuous SLAM for robots, but people are using RTAB-Map for 3D scanning too, so I did implemented some reconstruction approaches in the "Export Clouds" dialog. Implementing a TSDF in RTAB-Map could be great to have similar meshing results than ElasticFusion-like approaches (assuming having a powerful computer). 

Not sure if the ORB feature (from OpenCV) used in RTAB-Map is exactly the same as in ORB-SLAM2. The SLAM approach in RTAB-Map is different than ORB-SLAM2, to not confuse between them.

The structures of SLAM systems are rarely compatible between them, and rely on many dependencies. Binary ROS packages are the current best place to don't bother too much with installation issues.

cheers,
Mathieu

Lhun, 2016-11-02:
That makes a lot of sense, I really appreciate the input! The tech world is
clamoring for consumer accessible 3d scanning techniques - thank you for
being so open to all uses of what you've developed!

closed 2016-11-09 by matlabbe