introlab/rtabmap #143: Phab2 Pro Tango: coordinates seem backwards

matlabbe, 2016-11-09:
"coordinates on the Phab2 seem backwards and it isn't working"

Rafcin, 2016-11-14:
@matlabbe Yep

matlabbe, 2016-11-15:
Unfortunately, I don't have a Phab2 Pro to test with (I added this issue from a comment I received). The problem may be a wrong transformation when converting Tango pose to RTAB-Map coordinate system, e.g., this [one](https://github.com/introlab/rtabmap/blob/7fd3d9f331f7f3a5a60572ec6049a6e92439d443/app/android/jni/CameraTango.cpp#L308) is depending on the hardware (cameras/IMU position).

If someone can map with a Phab2 Pro (like moving 2-3 meters forward), then save the database and send it here I could investigate more the problem.

cheers,
Mathieu

cavercse, 2016-11-15:
Hi, I just specifically signed up to try to help get RTAB-Map working on the Lenovo Phab2 Pro. Here's a database of me walking forward and turning left 180 degrees. Hopefully this helps. If not than let me know what you need. 
[WalkFowardandTurnLeft.zip](https://github.com/introlab/rtabmap/files/593284/WalkFowardandTurnLeft.zip)

matlabbe, 2016-11-16:
Thx, it helped a lot to find a problem with the depth registration to color camera. The projected point cloud looks good but the textures are upside down (in reality it is the depth image that is upside down, top center view): 
[image]

I also did a cleanup of the transforms used, according to Tango, OpenGL and RTAB-Map different referential. The new version will be officially released on the Play Store.

matlabbe, 2016-11-16:
The update is now online. You should know that you have the latest version if in the "Mapping Options..." menu there is a "HD Mode" option instead of "720p Mode" option.

matlabbe, 2016-11-16:
If there is something wrong, capturing another similar database would be great. thx!

cavercse, 2016-11-16:
Sweet! Thanks! I just tried it out and it works good as far as the rendering/mapping goes now. The biggest issue I can see now is that the first and third person screen viewing orientation is still off 90 degrees though. The X and Y axis and grid needs to be rotated 90 degrees counter clockwise with respect to the screen if that makes any sense. I've attached a video that will probably show the issue better than I can explain it. 

Also, I just tried the "HD Mode" and it crashes the program. The "720p Mode" did the same thing in the past. 

[Google Tango Lenovo.zip](https://github.com/introlab/rtabmap/files/595293/Google.Tango.Lenovo.zip)

matlabbe, 2016-11-16:
The movements look ok **in** the application, but the app should be in landscape mode. That makes sense as the Phab 2 Pro is a phone (so it is in Portrait mode by default), not a tablet like the Google Tango development kit.

It seems to be an option in the app's manifest.xml:
```xml
android:screenOrientation="landscape"
```
I'll play with it on my tablet to see if I can reproduce the same bug as forcing my tablet to be in Portrait mode.

matlabbe, 2016-11-16:
New version with this fix is in prod, should be available through update in a moment.

cavercse, 2016-11-16:
Now that you point it out, the auto-rotate on the phone seems to be stuck in portrait mode for almost everything. That's got to be an issue with the phone or android os. I'll let you know how the RTAB-Map update works once I try it out.

matlabbe, 2016-11-16:
From the [Android documentation](https://developer.android.com/guide/topics/manifest/activity-element.html) for the `android:screenOrientation` tag, I think it is more app-dependent as in the manifest we can ignore the phone orientation "nosensor" so that it will be always in the same mode (portrait or landscape). Setting to "fullSensor" or "sensor" would flip between portrait and landscape.

cavercse, 2016-11-16:
It seems to be working good now. I'll keep using it and see if there are another issues that I can find.

matlabbe, 2016-11-16:
Thank you for your feedback! I'll close this issue. Note that I just opened an issue for the HD Mode issue.

cheers,
Mathieu

closed 2016-11-16 by matlabbe