introlab/rtabmap #144: Benchmark Gives out poor results. Did I get something wrong?

hanl09, 2016-11-10:
I just installed RTAB-MAP in Ubuntu 14.04, and tried to run [benchmark](https://github.com/introlab/rtabmap/wiki/Benchmark) using rtabmap-console. I tried the New-College dataset, and imageJoiner is used. My input is the same as suggested in wiki as below:

_./rtabmap-console -Rtabmap/StatisticLogged true -Rtabmap/StatisticLoggedHeaders false -Rtabmap/PublishStats false -Mem/RehearsalIdUpdatedToNewOne true -Mem/BadSignaturesIgnored true -Rtabmap/LoopRatio 0.0 -Bayes/FullPredictionUpdate true -Kp/TfIdfLikelihoodUsed false -Mem/STMSize 30 -Mem/RehearsalSimilarity 0.2 -Mem/RecentWmRatio 0.2 -SURF/Extended false -Kp/WordsPerImage 400 -Kp/BadSignRatio 0.25 -SURF/HessianThreshold 1000 -Kp/IncrementalFlann false -Rtabmap/TimeThr 1400 ~/Documents/data/Images_NewCollege/_joined._

It runs with a warning:
[image]

And the last result only detects 3 loop closures. And I noted that the average runtime is only 300ms per image.
[image]

Did I get something wrong?

matlabbe, 2016-11-10:
I just fixed a [bug](https://github.com/introlab/rtabmap/commit/9fbd02b06d79ecc1564e3148201b50755795c290) (introduced after 0.11.8 release) making the visual feature quantization not working on the [benchmark examples](https://github.com/introlab/rtabmap/wiki/Benchmark). I can now reproduce pretty much exactly the same results than on the [example output for NewCollege](https://github.com/introlab/rtabmap/wiki/Example-NewCollege). If you have rtabmap 0.11.8, it would not be your problem. If your using the latest 0.11.11 version, update the code or set `-Kp/IncrementalFlann=true` instead of `-Kp/IncrementalFlann=false` in your command line above.

cheers

matlabbe, 2016-11-11:
Note also I added `-Kp/DetectorStrategy 0` to [examples](https://github.com/introlab/rtabmap/wiki/Benchmark) to warn the user that SURF is required if rtabmap is not built with OpenCV nonfree (without SURF and SIFT support).

hanl09, 2016-11-14:
I update the new code and it works OK.
Thanks a lot!

closed 2016-11-14 by matlabbe