introlab/rtabmap #145: rtabmap make error

jje8910, 2016-11-15:
Hi, I want to run rtabmap in my Ubuntu.
I follow the guid,
cmake .. succeed, but when I try "make -j4" it throw an error like below.
/path/rtabmap/utilite/src/ULogger.cpp: In static member function ‘static void ULogger::unregisterCurrentThread()’:
/path/rtabmap/utilite/src/ULogger.cpp:235:9: error: no match for ‘operator=’ (operand types are ‘std::map<std::basic_string<char>, long unsigned int>::iterator {aka std::_Rb_tree_iterator<std::pair<const std::basic_string<char>, long unsigned int> >}’ and ‘void’)
    iter = registeredThreads_.erase(iter);
         ^
/path/rtabmap/utilite/src/ULogger.cpp:235:9: note: candidate is:
In file included from /usr/include/c++/4.8/map:60:0,
                 from /path/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:34,
                 from /path/rtabmap/utilite/src/ULogger.cpp:20:
/usr/include/c++/4.8/bits/stl_tree.h:157:12: note: std::_Rb_tree_iterator<std::pair<const std::basic_string<char>, long unsigned int> >& std::_Rb_tree_iterator<std::pair<const std::basic_string<char>, long unsigned int> >::operator=(const std::_Rb_tree_iterator<std::pair<const std::basic_string<char>, long unsigned int> >&)
     struct _Rb_tree_iterator
            ^
/usr/include/c++/4.8/bits/stl_tree.h:157:12: note:   no known conversion for argument 1 from ‘void’ to ‘const std::_Rb_tree_iterator<std::pair<const std::basic_string<char>, long unsigned int> >&’

Could you give me a guide?

matlabbe, 2016-11-15:
This bug was introduced by this [commit](https://github.com/introlab/rtabmap/commit/9276607920c8d593929a7e4c813e9ffc426b525b) yesterday (well on ma mchine it built correctly). The referenced [commit](https://github.com/introlab/rtabmap/commit/6d9f1828c694c110e01be9b9e3f5b790e618b03b) above would fix the problem.

cheers,
Mathieu

jje8910, 2016-11-16:
I pull the master git and I check the above issue is solved. But when I try "make -j4", the below error occurs.

Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: /usr/local/lib/libz.a(compress.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libz.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/librtabmap_core.so.0.11.11] error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] error 2

jje8910, 2016-11-16:
Now I succeed the compile for rtabmap 
but when I try "cmake .." in app/android, the below error occurs.

"CMake Error at src/CMakeLists.txt:11 (QT5_WRAP_CPP):
  Unknown CMake command "QT5_WRAP_CPP"."

matlabbe, 2016-11-16:
You cannot do that. The android app should be cross-compiled, so all the app/android branch is ignored in the standard build.

Building for android is not easy, I cannot help more than what is it written in this issue: https://github.com/introlab/rtabmap/issues/74

cheers,
Mathieu