introlab/rtabmap #148: rtabmap settings not saving

natejgardner, 2016-11-22:
Several rtabmap configuration settings do not save. Detection rate and buffer size are two that don't show any changes in the config.ini file.

I change them in the rtabmap preferences dialogue, I click apply, I save the config.ini. When I then launch from that config.ini, those settings are not there. When I examine the config.ini, it has not changed at all. Most other settings save successfully, but I have found that a number of them simply don't save at all. Hypothesis verification does this, too.

matlabbe, 2016-11-24:
Which version are you using? Windows or Linux?

I tested the default config file:
```bash
$ cat ~/.rtabmap/rtabmap.ini | grep Detection
Rtabmap\DetectionRate=1

# Here I opened Preferences, changed Detection rate 
# to 0, clicked ok to close Preferences, then "ctrl-s" to 
# save the config file.

$ cat ~/.rtabmap/rtabmap.ini | grep Detection
Rtabmap\DetectionRate=0
```

Same thing when using the "Save settings (*.ini)" button:
```bash
# First save a new config.ini in Preferences, then
$ cat ~/Documents/RTAB-Map/config.ini | grep Detection
Rtabmap\DetectionRate=1
# change Detection Rate and then save again:
$ cat ~/Documents/RTAB-Map/config.ini | grep Detection
Rtabmap\DetectionRate=0
```

cheers,
Mathieu

natejgardner, 2016-12-05:
Using Linux, 0.11.11

matlabbe, 2016-12-05:
Try 0.11.11, and still not able to reproduce the problem (parameters are correctly saved). When setting logging to "File", `LogRtabmap.txt` is created in working directory and correctly filled depending on the level selected. 

Maybe there is a problem with writing permissions on your computer. Make sure you didn't make rtabmap or launch rtabmap in sudo mode. Can you post a `config.ini` that you saved on this page?