introlab/rtabmap #149: standalone binary release for windows crashes when g2o graph match has been selected.

zjsprit, 2016-11-24:
Recently, I can sucessfuly run the standalone binary release (RTABMap-0.11.11-win64.exe) in the Windows 8.1, but if  I select the g2o in the Graph Optimization of the Perferences panels , the program crashes when close and exit from the program. Can you give me some suggestions? 
My Selection and the abnormal dialog are as follows pictures:
[image]

[image]

matlabbe, 2016-11-28:
Hi, 

thx for the observation. I can reproduce the error too. Hopefully, the crash happens after closing everything, so the database and config are correctly saved. The crash may happen when releasing g2o DLL. I'll update g2o version used in Windows binaries to see if the crash still happen.

cheers

matlabbe, 2016-11-28:
More info (this is with the latest code of g2o in Release mode):
[image]

matlabbe, 2016-11-30:
By building g2o in debug mode, we can see that it is crashing when unregistering action right [there](https://github.com/RainerKuemmerle/g2o/blob/aecec52e7a5eed9c8fa8ac9e60e2d26a8374a4b6/g2o/core/hyper_graph_action.h#L216).

If I set -DG2O_DEBUG_ACTIONLIB when building g2o:
```bash
...\rtabmap\bin>RTABMap.exe
g2o::RegisterActionProxy<class g2o::VertexSE2WriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::VertexSE2WriteGnuplotAction
g2o::HyperGraphActionLibrary::registerAction: creating collection for "writeGnuplot"
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::VertexSE2
g2o::RegisterActionProxy<class g2o::VertexPointXYWriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::VertexPointXYWriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::VertexPointXY
g2o::RegisterActionProxy<class g2o::EdgeSE2WriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE2WriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::EdgeSE2
g2o::RegisterActionProxy<class g2o::EdgeSE2PointXYWriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE2PointXYWriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::EdgeSE2PointXY
g2o::RegisterActionProxy<class g2o::EdgeSE2PointXYBearingWriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE2PointXYBearingWriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::EdgeSE2PointXYBearing
g2o::RegisterActionProxy<class g2o::VertexSE2DrawAction>::RegisterActionProxy: Registering action of type class g2o::VertexSE2DrawAction
g2o::HyperGraphActionLibrary::registerAction: creating collection for "draw"
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::VertexSE2
g2o::RegisterActionProxy<class g2o::VertexPointXYDrawAction>::RegisterActionProxy: Registering action of type class g2o::VertexPointXYDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::VertexPointXY
g2o::RegisterActionProxy<class g2o::EdgeSE2DrawAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE2DrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::EdgeSE2
g2o::RegisterActionProxy<class g2o::EdgeSE2PointXYDrawAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE2PointXYDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::EdgeSE2PointXY
g2o::RegisterActionProxy<class g2o::EdgeSE2PointXYBearingDrawAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE2PointXYBearingDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::EdgeSE2PointXYBearing
g2o::RegisterActionProxy<class g2o::VertexSE3WriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::VertexSE3WriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::VertexSE3
g2o::RegisterActionProxy<class g2o::VertexPointXYZWriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::VertexPointXYZWriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::VertexPointXYZ
g2o::RegisterActionProxy<class g2o::EdgeSE3WriteGnuplotAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE3WriteGnuplotAction
g2o::HyperGraphElementActionCollection::registerAction writeGnuplot class g2o::EdgeSE3
g2o::RegisterActionProxy<class g2o::VertexPointXYZDrawAction>::RegisterActionProxy: Registering action of type class g2o::VertexPointXYZDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::VertexPointXYZ
g2o::RegisterActionProxy<class g2o::VertexSE3DrawAction>::RegisterActionProxy: Registering action of type class g2o::VertexSE3DrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::VertexSE3
g2o::RegisterActionProxy<class g2o::EdgeSE3DrawAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE3DrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::EdgeSE3
g2o::RegisterActionProxy<class g2o::EdgeSE3PointXYZDrawAction>::RegisterActionProxy: Registering action of type class g2o::EdgeSE3PointXYZDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::EdgeSE3PointXYZ
g2o::RegisterActionProxy<class g2o::EdgeProjectDisparityDrawAction>::RegisterActionProxy: Registering action of type class g2o::EdgeProjectDisparityDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::EdgeSE3PointXYZDisparity
g2o::RegisterActionProxy<class g2o::CacheCameraDrawAction>::RegisterActionProxy: Registering action of type class g2o::CacheCameraDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::CacheCamera
g2o::RegisterActionProxy<class g2o::CacheSE3OffsetDrawAction>::RegisterActionProxy: Registering action of type class g2o::CacheSE3OffsetDrawAction
g2o::HyperGraphElementActionCollection::registerAction draw class g2o::CacheSE3Offset

***** VIDEOINPUT LIBRARY - 0.1995 - TFW07 *****

libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[... cut ...]
If I remove these [lines](https://github.com/introlab/rtabmap/blob/9276607920c8d593929a7e4c813e9ffc426b525b/corelib/src/OptimizerG2O.cpp#L526-L529) from OptimizerG2O.cpp, the crash is fixed:
```cpp
optimizer.clear();
g2o::Factory::destroy();
g2o::OptimizationAlgorithmFactory::destroy();
g2o::HyperGraphActionLibrary::destroy();
```
I think the problem is coming from g2o::HyperGraphActionLibrary::destroy();, which may delete all actions. So when the static objects are deleted on exit, the actions are already deleted, causing the crash. Anyway, I don't think destroying the instances each time we call g2o is required, so I removed this line in the referred commit above.

closed 2016-11-30 by matlabbe

zjsprit, 2016-12-01:
I really appreciate your help! Thank you!

commit d879491a (2016-11-30): Fixed #149