introlab/rtabmap #151: Cannot run LoopClosure algorithm for set of images

K0stIa, 2016-12-05:
Cannot run LoopClosure detection from the directory with images as it is stated in the tutorial.

https://github.com/introlab/rtabmap/wiki/Loop-closure-detection

matlabbe, 2016-12-05:
The tutorial is updated. The steps should match now the latest versions.

cheers,
Mathieu

K0stIa, 2016-12-06:
hey matlabbe,

thanks for updating tutorial, but there is still one more problem with it. 
After I selected Images and clicked "Start" button system wants me to do camera calibration
and camera calibration window pops up. If I refuse to use it, the system does not proceed further.

matlabbe, 2016-12-06:
Normally when loading [settingsAppearanceDemo.ini](https://raw.githubusercontent.com/wiki/introlab/rtabmap/doc/Tutorials/ProcessADataSet/settingsAppearanceDemo.ini), RGB-D SLAM should be disabled. If not, uncheck RGB-D SLAM here:
[image]

cheers

K0stIa, 2016-12-06:
Alright, RGB-D SLAM was not disabled. It works well now.

Thanks, Mathieu.

closed 2016-12-06 by matlabbe