introlab/rtabmap #154: Installation error ubuntu 14.04 rtabmap ros source install

aaravrav142, 2016-12-10:
Hi, 
I am installing rtabmap on a fresh ubuntu 14.04 install. Followed all the instructions for the rtabmap_ros install. I am doing the installation from source. I have installed all the optional dependencies including g2o, cvsba, gtsam and freenect. I have ros indigo installed. Below is the output from the terminal.

```
ankit@ankitd:~/rtabmap/build$ cmake ..
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/indigo/include/libfreenect
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /opt/ros/indigo/include;/usr/local/include/EXTERNAL/csparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found RealSense: /opt/ros/indigo/include
-- Found octomap: /opt/ros/indigo/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.12
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 2 nonfree module (SIFT/SURF) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ankit/rtabmap/build
```
```
ankit@ankitd:~/rtabmap/build$ make -j4
[  1%] [  1%] [  1%] [  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsManager.cpp.o
Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsHandler.cpp.o
Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UEventsSender.cpp.o
Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UFile.cpp.o
[  2%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UDirectory.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UConversion.cpp.o
[  3%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/ULogger.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UThread.cpp.o
[  4%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UTimer.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UProcessInfo.cpp.o
[  5%] Building CXX object utilite/src/CMakeFiles/rtabmap_utilite.dir/UVariant.cpp.o
Linking CXX shared library ../../../bin/librtabmap_utilite.so
[  5%] Built target rtabmap_utilite
[... 264 lines cut ...]
[ 83%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_OdometryViewer.cxx.o
[ 83%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_LoopClosureViewer.cxx.o
[ 84%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DataRecorder.cxx.o
[ 84%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CameraViewer.cxx.o
[ 85%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CalibrationDialog.cxx.o
[ 85%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ExportDialog.cxx.o
[ 86%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_PostProcessingDialog.cxx.o
[ 86%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ExportCloudsDialog.cxx.o
[ 87%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ExportScansDialog.cxx.o
[ 87%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_MapVisibilityWidget.cxx.o
[ 88%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_GraphViewer.cxx.o
[ 88%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_CreateSimpleCalibrationDialog.cxx.o
[ 89%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_ParametersToolBox.cxx.o
[ 89%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/moc_DepthCalibrationDialog.cxx.o
[ 89%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/3rdParty/moc_QMultiComboBox.cxx.o
[ 90%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/qrc_GuiLib.cxx.o
Linking CXX shared library ../../../bin/librtabmap_gui.so
[ 90%] Built target rtabmap_gui
make: *** [all] Error 2

```

I am not sure what is wrong here. Can you help. 

Thanks

matlabbe, 2016-12-10:
Hi,

If you make without g2o (`cmake -DWITH_G2O=OFF ..`), do you still have these errors? Which g2o version are you using?

cheers,
Mathieu

aaravrav142, 2016-12-12:
Hi,
Withoug g2o the build was successful. I cloned the latest version from https://github.com/RainerKuemmerle/g2o.

I will need g2o for my project. Is it possible to build rtabmap with g2o somehow. Or can you point me to the version of g2o that works. 

Thanks

matlabbe, 2016-12-13:
Yes it should build normally. The errors look like a missing library to link on. If you find in which library "`undefined reference to cs_pinv`" is defined for example, look at the file [FindG2O.cmake](https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindG2O.cmake) to see if it is correctly added to `G2O_LIBRARIES` variable.

When you do "`$ make VERBOSE=1`", you can see to which libraries the compiled library is linking to.

Just tried the latest g2o source code and I don't have these errors (I'm on Mac OS X).

cheers

aaravrav142, 2016-12-13:
OK,

I found the problem. From your suggestion, I was able to link the g2o libraries and was able to clean install the rtabmap with g2o. Many thanks for your help.

closed 2016-12-13 by aaravrav142

PeterPanxb, 2017-01-07:
Hi,
I encounter the same problem  with you.after read what you said , i still can't find out the question.Can you tell me how did you solve this problem.Thanks a lot.

hebahakim, 2019-02-07:
Hi,
I also encounter the same problem in Raspberry pi 3 B+. How can you able to link the g2o libraries? please advice me.

any help will be appreciated
regards,
Heba