introlab/rtabmap #162: error while build

Exception4U, 2017-01-21:
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_pinv'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_post'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_calloc'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_malloc'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_nfree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_spfree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_etree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_sparse const*, double*, cs_symbolic const*, double*, int*)'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_symperm'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_schol'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_sparse const*, bool)'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_cumsum'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_sfree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_amd'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_sparse const*, cs_symbolic const*, int*, double*)'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_free'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_counts'
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/rtabmap-camera] Error 1
make[1]: *** [tools/Camera/CMakeFiles/camera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 67%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/PdfPlot.cpp.o
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_pinv'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_post'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_calloc'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_malloc'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_nfree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_spfree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_etree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_sparse const*, double*, cs_symbolic const*, double*, int*)'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_symperm'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_schol'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_sparse const*, bool)'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_cumsum'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_sfree'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_amd'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_sparse const*, cs_symbolic const*, int*, double*)'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_free'
../../../bin/librtabmap_core.so.0.11.10: undefined reference to `cs_counts'
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/rtabmap-stereoEval] Error 1

Exception4U, 2017-01-22:
it seems it cannot bind csparse . I have installed libsuitesparse-dev allready. Any comments ?

matlabbe, 2017-01-27:
What is the output of (clean) cmake? 
```
$ cd rtabmap/build
$ rm CMakeCache.txt
$ cmake .. 
```

jemmazhou, 2017-03-16:
Hello,I encounter the same problem as you ,Can you give me some help.Thanks

Exception4U, 2017-03-16:
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/indigo/include/libfreenect
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /opt/ros/indigo/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found RealSense: /opt/ros/indigo/include
-- Found octomap: /opt/ros/indigo/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.13
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 2 nonfree module (SIFT/SURF) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tushar/codes/rtabmap/build

DavidHan008, 2017-03-17:
学姐你好

CelestialBoon, 2017-10-12:
I've got a similar error: 
```[ 83%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_pinv'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_post'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_calloc'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_malloc'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_nfree'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_spfree'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_etree'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_sparse const*, double*, cs_symbolic const*, double*, int*)'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_symperm'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_schol'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_sparse const*, bool)'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_cumsum'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_sfree'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_amd'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_sparse const*, cs_symbolic const*, int*, double*)'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_free'
../../../bin/librtabmap_core.so.0.14.2: undefined reference to `cs_counts'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:518: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2```

And the clean `cmake` output:
```-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL_KDTREE: /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so  
-- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
-- Found PCL_SEARCH: /usr/lib/x86_64-linux-gnu/libpcl_search.so  
-- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so  
-- Found PCL_FILTERS: /usr/lib/x86_64-linux-gnu/libpcl_filters.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_FEATURES: /usr/lib/x86_64-linux-gnu/libpcl_features.so  
-- Found PCL_SEGMENTATION: /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so  
[... 46 lines cut ...]
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = NO (libpointmatcher not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/celestial/ext/rtabmap/build```

matlabbe, 2017-10-12:
Cannot reproduce the problem on my system (Ubuntu 16.04, ROS Kinetic). My config is below, note that CSparse is not found in a local EXTERNAL directory like in your cmake.

```bash
$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'flann'
--   Found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Found qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- Found PCL_COMMON: /usr/lib/x86_64-linux-gnu/libpcl_common.so  
-- Found PCL_KDTREE: /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so  
-- Found PCL_OCTREE: /usr/lib/x86_64-linux-gnu/libpcl_octree.so  
-- Found PCL_SEARCH: /usr/lib/x86_64-linux-gnu/libpcl_search.so  
-- Found PCL_IO: /usr/lib/x86_64-linux-gnu/libpcl_io.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so  
-- Found PCL_FILTERS: /usr/lib/x86_64-linux-gnu/libpcl_filters.so  
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- Found PCL_FEATURES: /usr/lib/x86_64-linux-gnu/libpcl_features.so  
-- Found PCL_SEGMENTATION: /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so  
-- Found PCL_SURFACE: /usr/lib/x86_64-linux-gnu/libpcl_surface.so  
-- Found PCL_REGISTRATION: /usr/lib/x86_64-linux-gnu/libpcl_registration.so  
-- Found PCL_RECOGNITION: /usr/lib/x86_64-linux-gnu/libpcl_recognition.so  
-- Found PCL_KEYPOINTS: /usr/lib/x86_64-linux-gnu/libpcl_keypoints.so  
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so  
-- Found PCL_PEOPLE: /usr/lib/x86_64-linux-gnu/libpcl_people.so  
-- Found PCL_OUTOFCORE: /usr/lib/x86_64-linux-gnu/libpcl_outofcore.so  
-- Found PCL_TRACKING: /usr/lib/x86_64-linux-gnu/libpcl_tracking.so  
-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- Found PCL_MODELER: /usr/include/pcl-1.7  
[... 50 lines cut ...]
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = YES (License: BSD)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = YES (License: Apache-2)
--   With RealSenseSlam        = YES
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = YES (License: BSD)
--   With libfovis             = YES (License: GPLv2)
--   With libviso2             = YES (License: GPLv3)
--   With dvo_core             = YES (License: GPLv3)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/rtabmap/build
```

CelestialBoon, 2017-10-13:
Removing the EXTERNAL folder and forcing use of the /usr/include one has fixed it for me. Thanks a bunch!

closed 2017-10-13 by matlabbe

Pankaj-Baranwal, 2019-01-25:
Where is the "EXTERNAL" folder??
How did you force the use of /usr/include?

songyongkang, 2019-10-25:
where is the EXTERNAL folder

matlabbe, 2019-10-25:
```
-- Found CSPARSE: /usr/local/include/EXTERNAL/csparse
```
What is the path of csparse on your system?

songyongkang, 2019-10-25:
> ```
> -- Found CSPARSE: /usr/local/include/EXTERNAL/csparse
> ```
> 
> What is the path of csparse on your system?

/usr/local/include/g2o/solvers/csparse

songyongkang, 2019-10-25:
my g2o.BUILD
```
`package(default_visibility = ["//visibility:public"])

licenses(["notice"])

cc_library(
    name = "g2o",
    hdrs = glob([
	    "*.h",
        "**/*.h",
        "**/**/*.h",
        "**/**/**/*.h",
    ]),
    includes = [
        ".",
    ],
    linkopts = [ 
        "-L/usr/local/lib -lg2o_cli",
        "-L/usr/local/lib -lg2o_core",
        "-L/usr/local/lib -lg2o_csparse_extension",
        "-L/usr/local/lib -lg2o_ext_freeglut_minimal",
        "-L/usr/local/lib -lg2o_incremental",
        "-L/usr/local/lib -lg2o_interactive",
        "-L/usr/local/lib -lg2o_interface",
        "-L/usr/local/lib -lg2o_opengl_helper",
        "-L/usr/local/lib -lg2o_parser",
        "-L/usr/local/lib -lg2o_simulator",
        "-L/usr/local/lib -lg2o_solver_cholmod",
        "-L/usr/local/lib -lg2o_solver_csparse",
        "-L/usr/local/lib -lg2o_solver_dense",
        "-L/usr/local/lib -lg2o_solver_eigen",
        "-L/usr/local/lib -lg2o_solver_pcg",
        "-L/usr/local/lib -lg2o_solver_slam2d_linear",
        "-L/usr/local/lib -lg2o_solver_structure_only",
        "-L/usr/local/lib -lg2o_stuff",
        "-L/usr/local/lib -lg2o_types_data",
        "-L/usr/local/lib -lg2o_types_icp",
        "-L/usr/local/lib -lg2o_types_sba",
        "-L/usr/local/lib -lg2o_types_sclam2d",
        "-L/usr/local/lib -lg2o_types_sim3",
        "-L/usr/local/lib -lg2o_types_slam2d",
        "-L/usr/local/lib -lg2o_types_slam2d_addons",
        "-L/usr/local/lib -lg2o_types_slam3d",
        "-L/usr/local/lib -lg2o_types_slam3d_addons",

    ],
)
```

`
my WORKSPACE

#g2o
```
new_local_repository(
    name = "g2o",
    build_file = "third_party/g2o.BUILD",
    path = "/usr/local",
)

```

my g2o version g2o-20170730_git

part of my problem 

`bazel-out/local-dbg/bin/modules/prediction/predictor/optimal/liboptimal.a(optimal_optimizer.pic.o): In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&)':
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:208: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_sparse const*, cs_symbolic const*, int*, double*)'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:213: undefined reference to `cs_nfree'
bazel-out/local-dbg/bin/modules/prediction/predictor/optimal/liboptimal.a(optimal_optimizer.pic.o): In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::computeSymbolicDecomposition(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&)':
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:250: undefined reference to `cs_schol'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:265: undefined reference to `cs_amd'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:281: undefined reference to `cs_free'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:284: undefined reference to `cs_calloc'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:285: undefined reference to `cs_pinv'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:286: undefined reference to `cs_symperm'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:287: undefined reference to `cs_etree'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:288: undefined reference to `cs_post'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:289: undefined reference to `cs_counts'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:290: undefined reference to `cs_free'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:291: undefined reference to `cs_spfree'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:292: undefined reference to `cs_malloc'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:293: undefined reference to `cs_cumsum'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:294: undefined reference to `cs_free'
/usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:296: undefined reference to `cs_sfree'

`

songyongkang, 2019-10-25:
> ```
> -- Found CSPARSE: /usr/local/include/EXTERNAL/csparse
> ```
> 
> What is the path of csparse on your system?

maybe use wrong g2o version?

songyongkang, 2019-10-26:
.so: undefined symbol: _ZN3g2o17csparse_extension17cs_chol_workspaceEPK9cs_sparsePK11cs_symbolicPiPd

songyongkang, 2019-10-27:
> > ```
> > -- Found CSPARSE: /usr/local/include/EXTERNAL/csparse
> > ```
> > 
> > 
> > What is the path of csparse on your system?
> 
-- Found CSPARSE: /usr/include/suitesparse

matlabbe, 2019-11-03:
Is g2o building with its own csparse version? On my system, g2o installed from binaries seems built with system csparse. The cmake output of rtambap shows:
```
-- Found g2o: /opt/ros/melodic/include;/usr/include/suitesparse;/usr/include/suitesparse
``` 
Or to make sure rtabmap is finding the right csparse, you may modify this:
https://github.com/introlab/rtabmap/blob/179f17d2671f48472a9ee860165873ea06db16c3/cmake_modules/FindG2O.cmake#L1-L5

cheers,
Mathieu