introlab/rtabmap #163: Possibility of receiving lighthouse position/orientation data

eric-schleicher, 2017-01-25:
Not an issue but a few questions.  

I love working with RTAB-MAP, and have found it to be the most robust and stable way to capture room scale with the tango (lenovo phab 2 pro). 

I'm just now looking into compiling RTAB-MAP myself and to determine whether contributing to RTAB-MAP makes sense, or whether I should be trying to build upon one of the existing example tango apps.  

My immediate interests center around additions to the android application which leads to my first question...

I'll qualify that for the purposes of these questions I'm not (necessarily) interested in realtime meshing, just capturing/using additional data in a way that would benefit offline/downstream post processing.

- **Q1) Is source for the android application part of this repo?**  

I see some android related folders in the project, but is everything required to make a build of the android app present in this repo?

- **Q2) Is it inline with the approach for RTAB map to allow some external tracking information to be (either captured alongside) or directly to augment or replace device position/orientation in post processing?**

At first glance, it would seems that the db/capture model would allow the strage of external tracking data to be saves/used along with rest of the capture stream.  I've built a small harness to relay lighthouse tracked object data from a Vive to a connected machine over webRTC.  The data rate seems satisfactory and I would like to increase the precision of tracking by using this data.  One implementation (c/w)ould be to consume the webRTC data and associate it to the capture frame in RTAB-MAPs db file; resulting in a much more accurate (and self correcting) device position/orientation.

I know it wouldn't contribue to large scale captures, but would significantly increase the precision of object and room capture.

- **Q3) if it's possible/appropriate to add the ability to receive postition/pose information, are there any thoughts or recommendation on how where to store it, so that it might be used later.**

Interested in discussing if this isn't the right place to chat about it, please just let me know.

matlabbe, 2017-01-27:
Hi Eric,
Great that you find RTAB-Map Tango useful. Here some answers to your questions:

 * Q1) yes, everything is here to build the [Tango app](https://github.com/introlab/rtabmap/tree/master/app/android). However, if you are not familiar with cross-compiling, you may find very difficult to get all dependencies compiled correctly. See this long [post](https://github.com/introlab/rtabmap/issues/74) for example.

 * Q2) Yes, for example [here ](https://github.com/introlab/rtabmap/blob/ac1c5f81ae7d1df140f88f9c61e4bc283adee315/app/android/jni/CameraTango.cpp#L628) you can use an external pose (i.e. odom) instead of the one given by Tango. If you use a global localization system, you don't even need loop closure detection (this can be disabled like for Data Recorder mode).

 * Q3) Yes, the SensorData class has a ground truth field that can be used to store external global localization pose (we may even compare in RTAB-Map desktop "Tango pose" vs "Global localization pose"). For example, [right here](https://github.com/introlab/rtabmap/blob/ac1c5f81ae7d1df140f88f9c61e4bc283adee315/app/android/jni/CameraTango.cpp#L663) instead of null transform, we could set the pose of external localization (ground truth). On RTAB-Map Desktop, there is currently no way to export clouds accordingly to ground truth poses, but that would not be too hard to add the option if needed.

No problem to discuss this here, on github there are discussion/bug/question issues.

cheers,
Mathieu

eric-schleicher, 2017-01-28:
That's so awesome.  I see a clear path to achieving something truly great now!   that's a good amount to chew on, i will start digging in.  I'm sure i will have more questions : )

Thanks for responding.