introlab/rtabmap #165: Postprocessing Error [ OptimizerCVSBA.cpp:156::rtabmap::OptimizerCVSBA::optimizeBA()]

eric-schleicher, 2017-02-02:
A Capture with Phab2 Pro with RTAB-MAP 0.11.13

Choose Post process with the following options.
[image]

It runs but at the very end of the run i see the following error in the console.

Afterware it does appear as if changes have been made in the cloud

```
SBA: sba_motstr_levmar_x() cannot solve a problem with fewer measurements [8732] than unknowns [12245]
[ERROR] (2017-02-02 15:16:05) OptimizerCVSBA.cpp:156::rtabmap::OptimizerCVSBA::optimizeBA() Running SBA... error! C:\workspace\cvsba-1.0.0\src\cvsba.cpp:233: error: (8888) Error occured during Sba optimization in function Sba::run
```

Is this a legitimate error?  this behavior is consistent with other phab 2 captures i've made.

matlabbe, 2017-02-03:
cvsba requires that there are visual constraints (visual feature correspondences) between all images linked in the map. If not, there will be more unknowns than measurements. g2o's SBA is more robust to this, as it can use pure odometric links (without visual correspondences) of the map's graph in its optimization.

eric-schleicher, 2017-02-04:
Thank you!  I will try this.

closed 2017-02-04 by eric-schleicher