introlab/rtabmap #166: Consider adding a short meshlab tutorial

eric-schleicher, 2017-02-04:
I created a [FAQ](https://github.com/introlab/rtabmap/wiki/FAQ) page, and thought It would be nice to link to a short tutorial on reconstructing meshes & re-projecting texture from meshlab.

@matlabbe  any issue with that?  I didn't see one on the page or in the wiki..

matlabbe, 2017-02-04:
Yeah, you could link your [youtube video](https://www.youtube.com/watch?v=N3YlncWbMto&feature=youtu.be) with the bridge somewhere. 

Well, the features appear faster than the documentation. In 0.11.14, it is possible to generate a similar mesh+texture directly from Export dialog of RTAB-Map. See this post for example: http://official-rtab-map-forum.67519.x6.nabble.com/Scan-app-gt-import-optimize-PC-gt-export-td2634.html#a2637

For the FAQ page, I'll try to answer some of these questions soon. 

thx,
Mathieu

matlabbe, 2017-02-08:
I updated the [FAQ](https://github.com/introlab/rtabmap/wiki/FAQ) page.

fishcu, 2017-02-23:
@eric-schleicher, just as a hint, you can quickly get rid of weird surface meshing faces by using Filters->Selection->Select faces with edges longer than..., then deleting the selected faces. That filter will automatically set a threshold that is based on the average edge length. Use "preview" to verify your selection beforehand.