introlab/rtabmap #167: Setting not saved from session to session in android application?

eric-schleicher, 2017-02-04:
Two things i immediately do upon starting RTAB-MAP on android (everytime ) is go into the rendering settings  and disable meshing and set it to display point clouds only.  Not sure if it's just in my head, but it *seems* to perform better this way during mapping (phab 2 pro) ...  or at least the display is more informative without the (sort of janky looking) mesh bits in the way of the points

I've also started setting specific mapping refresh rates which has had a positive impact on the point density and accordingly on the resulting quality of the clouds. 

I've observed that these options have to be set **each** time starting the app.  It just slows down getting started and would be a better experience if the options persisted from session to session.

matlabbe, 2017-02-04:
All settings are now saved in RTAB-Map Tango 0.11.14. I will release it this weekend. 

cheers

eric-schleicher, 2017-02-08:
Confirmed save from session to sessions.  Thanks!

closed 2017-02-08 by eric-schleicher