introlab/rtabmap #169: rtabmap compile fail

QinZiwen, 2017-02-13:
I have install opencv3.2.0 and cuda8.0, but when I make rtabmap, some errors show:
```
turtlebot@turtlebot-OMEN-by-HP-Laptop:~/ThridPKG/rtabmap/build$ cmake ..
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenMP
-- Found OpenCV: /usr/include/opencv;/usr/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /opt/ros/indigo/include/libfreenect
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found RealSense: /opt/ros/indigo/include
-- Found octomap: /opt/ros/indigo/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.11.14
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
--   With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = NO (g2o not found)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = YES (License: Apache-2)
--   With OCTOMAP              = YES (License: BSD)
--   With Qt4                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/turtlebot/ThridPKG/rtabmap/build
turtlebot@turtlebot-OMEN-by-HP-Laptop:~/ThridPKG/rtabmap/build$ make -j8
[  6%] Built target rtabmap_utilite
[  7%] [  7%] Built target res_tool
Built target imagesJoiner
[  7%] Built target extractObject
Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: cannot find -lopencv_nonfree
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/librtabmap_core.so.0.11.14] Error 1
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [all] Error 2
```
And I find you said that " If you want OpenCV3, you should build ros vision-opencv package yourself (and all ros packages depending on it) so it can link on OpenCV3.". I dont known how to use vision-opencv to ros. 

Thanks for your help!

matlabbe, 2017-02-13:
Hi,

You may want to upgrade to 16.04 to get [vision-opencv](https://github.com/ros-perception/vision_opencv) package built already with OpenCV3.

Otherwise, the error above tells that rtabmap is building against OpenCV2:
```
With OpenCV 2 nonfree module (SIFT/SURF) = NO (not found, License: BSD)
```
 Make sure to clean the build directory to reset the CMake cache (path of OpenCV from previous builds).
```bash
$ rm rtabmap/build/*
```

I suggest to remove completely all OpenCV2 installations from your computer. This may uninstall ros packages depending on it, if so, just catkin_make from git source again all these packages so they link on your OpenCV3 installation. Example:

```bash
$ cd ~/catkin_ws/src
$ git clone https://github.com/ros-perception/vision_opencv.git
$ cd ..
$ catkin_make
```

cheers

QinZiwen, 2017-02-14:
Thanks, Opencv2 is a good choice!