introlab/rtabmap #170: rtabmap_ros and point cloude

aseyfi, 2017-02-16:
Hello, I am trying to use [this app](https://github.com/Intermodalics/tango_ros) on Lenovo Pahb2 Tango phone to stream raw data from sensors in ROS and run "rtabmap_ros" on my laptop to process the data. The app is publishing data from fisheye camera, rgb camera and point cloud and also pose of camera in different frames (rgb camera, depth camera, device, start of service). I wasn't able to find any rtabmap_ros tutorial that uses point cloud data. Would you please let me know if the current implementation of rtabmap_ros supports point cloud input?

Thank You,
Ahmad

matlabbe, 2017-02-16:
Hi,

The last time I tried, the released version (Play Store) didn't have the camera_info published from the cameras and also missing a registered depth image. A registered depth image is required by rtabmap, there is no point cloud input (beside 3D laser scan input /scan_cloud). To get a registered depth image to color camera, we could create a node projecting the cloud into color camera. The best would be that ROS tango streamer publishes the registered depth image directly. The project is currently on heavy developments, maybe they will add this soon.

cheers,
Mathieu

aseyfi, 2017-02-20:
Hi Mathieu

Thank you for your response. Following you suggestion, I modified the tango_ros_streamer app to publish the depth image. I can see the depth image in RVIZ, so I would assume it should be correct. Here is the list of the topics that I expect rtabmap_ros to use:

`/tango/camera/color_1/camera_info
/tango/camera/color_1/image_raw/compressed
/tango/depth/compressedDepth
/tf
/tf_static`

Then I modified "demo_robot_mapping.launch" launch file based on the instructions in [roswiki page](http://wiki.ros.org/rtabmap_ros). When I launch the file, I see the following warnings and nothing can be seen in rtabmapviz window:

`[ WARN] [1487375729.028781909]: Could not get transform from camera_color to start_of_service after 5.000000 seconds (for stamp=1487370759.994714)! Error="Lookup would require extrapolation into the future.  Requested time 1487370759.994714022 but the latest data is at time 1487370759.913281918, when looking up transform from frame [start_of_service] to frame [camera_color]. canTransform returned after 5.00636 timeout was 5.".
[ WARN] [1487375733.194010925]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=500).`
The output is based on  "wait_for_transform=5.0" and "queue_size=500". 

Using "rostopic hz" it was confirmed that the topics were publishing data.
Would you please let me know the series of steps to be taken to find the root cause of the issue?

Thank You,
Ahmad

aseyfi, 2017-02-20:
@matlabbe, looking at rtabmap_ros source code, it seems that it is using /tf topic when odom_frame_id is not empty. tango_ros_streamer is publishing pose data using tf2, so it is publishing /tf and /tf_static topics. If that's the case, do you have any suggestions on how this problem can be fixed?

Thank You,
Ahmad

matlabbe, 2017-02-21:
Hi, 

You may use [rtabmap.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/rtabmap.launch) directly instead of using `demo_robot_mapping.launch`, following this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping):

`
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" visual_odometry:=false odom_frame_id:="start_of_service" frame_id:="device" rgb_topic:=/tango/camera/color_1/image_raw depth_topic:=/tango/depth camera_info_topic:=/tango/camera/color_1/camera_info compressed:=true
`

Assuming that `device` is the base frame of the camera and `start_of_service` is the odom frame.

cheers,
Mathieu

fishcu, 2017-02-22:
@matlabbe, thank you for the awesome support!
@aseyfi, could you push your additions that enable publishing of the depth image? I want to try the exact same application as you. Thank you!

aseyfi, 2017-02-22:
@matlabbe , thank you very much for the prompt response.  I was looking at the source code of the rtabmap android app and in [this line](https://github.com/introlab/rtabmap/blob/af1cdadabd97e179d30421666b2974f973b6e898/app/android/jni/CameraTango.cpp#L643) there is a conversion from tango device coordinate system to rtabmap coordinate system:

`Transform odom = rtabmap_world_T_tango_world * poseDevice * tango_device_T_rtabmap_device;`

Do I need to perform similar transformation before publishing the data on /tf topic?

@fishcu, I am still working on publishing depth images and there are some bugs to be fixed. Please give me a few days to push the code to a public repo.

Thank You,
Ahmad

matlabbe, 2017-02-23:
Hi,

rtabmap's coordinate system is the same as ROS. Tango ROS streamer already publishes TF correctly. If you have to transform "transforms" in ROS coordinate system, that could be done similar to that. I don't think you need to publish more TF though. 

**I updated rtabmap and rtabmap_ros to support the current released version of Tango ROS Streamer**. 

 * Launch roscore on the computer:
  `
$ roscore
`
 * On the Tango device, launch Tango ROS Streamer and enable Device pose, Point cloud and Color image. Back on the computer, the following does registration of the point cloud to color camera to create the required registered depth image. As the point cloud has a lot less point than pixels in the RGB image (1920x1080), we set the depth image as 8 times smaller (240x135) than RGB and fill the remaining holes:
  `
$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/tango/point_cloud camera_info:=/tango/camera/color_1/cameranfo _fixed_frame_id:=start_of_service _decimation:=8 _fill_holes_size:=5
`
* Launch rtabmap. RGB image is decimated by 2 to save computation time (960x540), for feature extraction (`Mem/ImagePreDecimation`) and saving in database (`Mem/ImagePostDecimation`). We don't use visual odometry as it is already computed by Tango through TF, so we set `odom_frame_id` to use it. We use compressed topic only for the RGB image, as the depth is computed on the computer. Exact synchronization is used to get depth image corresponding exactly to RGB image that we registered earlier:
  `
$ roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start --Mem/ImagePreDecimation 2 --Mem/ImagePostDecimation 2" visual_odometry:=false odom_frame_id:="start_of_service" frame_id:="device" rgb_topic:=/tango/camera/color_1/image_raw depth_topic:=/image camera_info_topic:=/tango/camera/color_1/camera_info compressed:=true depth_image_transport:=raw approx_sync:=false
`

Note that to not screw up the TF tree, we should set `odom_frame_id:="area_description"` because there is a competition between `/area_description` -> `/start_of_service` and `/map` -> `/start_of_service`. So we would have instead `/map` -> `/area_description` -> `/start_of_service`, which seems not correct as two modules correct odometry. I didn't see the option to remove that frame (`/area_description`), which seems related to enabled drift correction of Tango by default. 

In term of performance, it would be better to decimate the RGB image (x2) on the phone, so we could save a lot of bandwidth (I am on WiFi). Make sure to adjust camera_info too if so. While the depth registration looks okay with `pointcloud_to_depthimage`, doing it on the phone may be more convenient, as you seem to do already. We could subscribe to `compressedDepth` to save even more bandwidth (in contrast to not compressed point cloud).

cheers,
Mathieu

fishcu, 2017-02-23:
@matlabbe, Great! I will try to support this project (in this particular configuration). Will test and report back any issues by the end of the week. @aseyfi I can also help with testing and bug fixing once you publish your code! Just want to avoid doing work that you have done already so far. :)

fishcu, 2017-02-24:
There is a small typo in the command used to run the depth point cloud to depth image converter. "cameranfo" should be "camera_info", like so:
`$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/tango/point_cloud camera_info:=/tango/camera/color_1/camera_info _fixed_frame_id:=start_of_service _decimation:=8 _fill_holes_size:=5`

Other than that, it works! The biggest problem right now is the bandwidth, as you mentioned. This causes big jumps in the pose graph (and the map, subsequently).

[image]

aseyfi, 2017-03-03:
@matlabbe , Thank you for the support and the recent improvements. Are you aware of any ros package similar to image transport that can compress point cloud? If there is no such a package, maybe we can create one. PCL has an example [here](http://pointclouds.org/documentation/tutorials/compression.php).

@fishcu , I uploaded my code to [this repo](https://github.com/aseyfi/tango_ros). Unfortunately I didn't have time to make any fixes and haven't tried it with rtabmap_ros yet. It publishes depth image in raw and compressedDepth format. You can build the app in master branch and then switch to add_depth_image and make it again. I am using tango_support_api and wasn't able to get a successful build from "catkin build" while tango_support_api.h was included. Also we need to apply something similar to [this function](https://github.com/introlab/rtabmap/blob/a9b03bb39e8756359a4deed4918eeb7080dc38e0/app/android/jni/CameraTango.cpp#L656) to fill the holes in the depth image.

Thank You,
Ahmad

matlabbe, 2017-03-03:
@aseyfi There is an old project (not updated since groovy) from ethz asl called [ethzasl_message_transport](http://wiki.ros.org/ethzasl_message_transport) that seems to be working for point cloud messages. However, it is better the transfer depth images if you can already publish them compressed (e.g., [image_transport's compressedDepth plugin](http://wiki.ros.org/compressed_depth_image_transport)). Depth images are either one 32bits (float meters) or one 16bits (unsigned short mm) per point. Point clouds are three 32bits per point. Well, for point cloud sources that cannot be converted to depth images (e.g., LiDAR scans), a node using the PCL code could be nice. There is a [post here](http://www.pcl-users.org/PCL-ROS-Compression-trouble-td3157088.html) from someone trying to do the same, it is dated from 2011, maybe that node is already somewhere on Github.

cheers,
Mathieu

fishcu, 2017-03-06:
Thanks @aseyfi!! I will look into it soon. What exactly needs to be fixed?
I can probably find out why by digging through the code, but why do we need to fill holes in the depth image?

matlabbe, 2017-03-08:
@fishcu The holes are the results of the pixel quantization (float value to integer value) during registration. Here is an example if I comment this [line](https://github.com/introlab/rtabmap/blob/c12980ee9168ec4dab30f7ab469fb327f686c3ff/app/android/jni/CameraTango.cpp#L656):
[image]
The right image is a close-up of the raw Tango point cloud (yellow) and the projection of the registered depth image back to 3D (gray).

aseyfi, 2017-03-17:
@matlabbe, in your comment
`Note that to not screw up the TF tree, we should set odom_frame_id:="area_description" because there is a competition between /area_description -> /start_of_service and /map -> /start_of_service. So we would have instead /map -> /area_description -> /start_of_service, which seems not correct as two modules correct odometry. I didn't see the option to remove that frame (/area_description), which seems related to enabled drift correction of Tango by default.`

Would you please explain what you mean by there is a competition? Who is publishing /map -> /start_of_service? Which modules are you talking about when you say " as two modules correct odometry"?

Thank You,
Ahmad

matlabbe, 2017-03-18:
Hi @aseyfi,

`/map -> /start_of_service` is published by rtabmap, it is the drift error estimated by RTAB-Map's loop closure detection. `/area_description -> /start_of_service` is the drift error estimated by Tango's loop closure detection (when Tango drift correction is enabled).

The TF error is that we cannot have a frame (`/start_of_service`) with two parents (`/map` and `/area_description`). Tango with drift correction does the same job as rtabmap node: correcting odometry drift on loop closure. A solution would be to disable Drift correction from Tango ROS Streamer, so that `/area_description` is not published.

[Just saw in Tango ROS Streamer repo](https://github.com/Intermodalics/tango_ros/blob/4027bb35e77824d16be374b3a20a9d74f9937f5b/tango_ros_common/tango_ros_native/src/tango_ros_node.cpp#L432-L444) that there would be an option to use Tango in Odometry mode only, which is the mode we would want with rtabmap (`config_enable_drift_correction=false`):
```cpp
bool enable_drift_correction = false;
  int localization_mode;
  node_handle_.param(publisher_config_.localization_mode_param, localization_mode, LocalizationMode::ODOMETRY);
  if (localization_mode == LocalizationMode::ONLINE_SLAM) {
    enable_drift_correction = true;
  }
  const char* config_enable_drift_correction = "config_enable_drift_correction";
  result = TangoConfig_setBool(tango_config_, config_enable_drift_correction, enable_drift_correction);
  if(result != TANGO_SUCCESS) {
    LOG(ERROR) << function_name << ", TangoConfig_setBool "
        << config_enable_drift_correction << " error: " << result;
    return result;
  }
```

cheers,
Mathieu

fishcu, 2017-03-22:
I tried out aseyfi's code today. It builds fine with Android studio. When trying to build with catkin, the error I get is that jni.h is not found -- I haven't looked into that, but it should be fixable by telling cmake where to look for the required java headers, which Android studio seems to provide automatically.

I cannot currently get it to run with RTAB-map because the resolution of the published depth image is not compatible with the rgb image resolution (rgb_h % depth_h != 0, i.e., RTAB-map expects the RGB size to be a multiple of the depth image size).
I checked the header of the published images. Indeed, the code currently publishes a 240*135 depth image instead of the native 320*180 resolution. @aseyfi is there a reason for this?
I'm going to look into the code now to see if there's any downscaling or similar going on, but I also asked here because aseyfi probably knows why!

@matlabbe: I see that the holes are a result of quantization. Why is it a problem, though? The behavior that I would expect is that having invalid measurements in the depth image are simply skipped (not incorporated into the map).

On another note, what do you use to visualize the depth image topic on the host computer? I tried RVIZ and the image_view (disparity_view) packages with different settings, but I cannot get an image to show up. I know that I'm getting data by outputting it raw with "rostopic echo".

aseyfi, 2017-03-23:
@fishcu , you can visualize the depth image as a regular image:
[image]

As you noticed, I am using Lenovo Phab 2 and the depth image size is (rgb_row/8)x(rgb_col/8) . My code is experimental and as you mentioned, I have hard-coded the rgb image dimensions. We definitely can merge your changes to the repo.

Did you get to use the app with RTABMAP?

Thanks,
Ahmad

aseyfi, 2017-03-23:
@fishcu, you should be able to run rtabmap_ros with the app using this command:
`roslaunch rtabmap_ros rtabmap.launch visual_odometry:=false odom_frame_id:="start_of_service" frame_id:="device" rgb_topic:=/tango/camera/color_1/image_raw depth_topic:=/tango/depth camera_info_topic:=/tango/camera/color_1/camera_info compressed:=true depth_image_transport:=compressedDepth approx_sync:=false rtabmap_args:="--delete_db_on_start --Mem/ImagePreDecimation 2 --Mem/ImagePostDecimation 2 --Rtabmap/MaxRetrieved 20"`

[image]

Cheers,

matlabbe, 2017-03-23:
Forgot to mention it, but I moved my [post](https://github.com/introlab/rtabmap/issues/170#issuecomment-281902859) above on ROS wiki with a video: http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer

@fishcu The holes are not so annoying when showing the point cloud, but when showing the mesh it can create big holes (in particular if the mesh is created from a lower resolution of the depth image).

@aseyfi great to see some results with your modifications!

cheers

fishcu, 2017-03-23:
My code is [here.](https://github.com/fishcu/tango_ros)

Thanks for your help. At the moment I'm getting a lot of crashes and I don't know why. I'm new to Android programming. Here's a paste of the logcat, cropped to before I launch the app, and after the crash (it crashes when trying to launch Tango / connecting to ROS -- no crash if I disconnect the device from the network): http://pastebin.com/iCjU0ETw
I will try to figure out what's wrong, then I'll do tests in combination with RTAB-map!
When it doesn't crash, I can get depth images out of it -- just like you have shown.

aseyfi, 2017-03-28:
Hi @matlabbe, What should I set the following parameter if I am using Lenovo Phab2? Can I use any arbitrary number? Or the number is calculated based on the spatial location of the sensors on the phone?

`Param: g2o/Baseline = "0.075"                              [When doing bundle adjustment with RGB-D data, we can set a fake baseline (m) to do stereo bundle adjustment (if 0, mono bundle adjustment is done). For stereo data, the baseline in the calibration is used directly.]`

Thanks,
Ahmad

hungbv11, 2017-03-29:
Hi @aseyfi ,
I am trying to publish the depth image directly from ROS_tango_streamer for Tango device. If you don't mind, could you give me some ideas how and where to modify the app?

Thanks a lot,
Hung

matlabbe, 2017-03-31:
@aseyfi The value is not important, it is just to be able to use stereo bundle adjustment with rgb-d cameras, for which we can keep the scale constraints on optimization in comparison to mono bundle adjustment. You may don't see a lot of difference between 5 cm and 10 cm baseline for example. On RTAB-Map Tango, the default value is used for Yellowstone tablet and Phab2Pro.

fishcu, 2017-05-08:
To follow-up on an earlier question, why is a depth image used instead of using /tango/point_cloud as input to /scan_cloud directly? I am not sure I understand the purpose of /scan_cloud.
I guess the reason is that a registered depth image is needed for feature extraction (features only extracted at valid depth), which requires a projection to image plane anyways? Still, it seems a bit wasteful to project, then re-project the depth image to 3d.

matlabbe, 2017-05-09:
Yes, to know where are the 2D features in 3D we need a **registered** depth for efficiency. Without a registered depth image, we should have to reproject the cloud anyway as we cannot know which point in the cloud corresponds to a pixel (multiple points with different depth can correspond to same pixel in RGB frame). 

To reconstruct the cloud we have two choices: project again the depth image (Grid/FromDepth=true) or use tango point cloud directly with `scan_cloud` input topic (Grid/FromDepth=false). 

In term of memory efficiency, it is better to subscribe only to a depth image than depth image + point cloud. For example, for a depth image of 160x120, we have `19200 x2`(unsigned short)=38400 bytes  to save instead of `19200 x3`(x,y,z)`x4`(float) = 230400 bytes to save just for the corresponding point cloud.

cheers,
Mathieu

matlabbe, 2017-08-28:
See http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer for tutorial.

closed 2017-08-28 by matlabbe