introlab/rtabmap #171: kinectv2

yalan2017, 2017-02-20:
Hi
how can i use it for kinectv2? is it possible?
Regards

fishcu, 2017-02-23:
Using ROS, it should be possible. A quick google reveals this forum thread:
[http://official-rtab-map-forum.67519.x6.nabble.com/rtabmap-for-kinect-v2-on-ROS-Indigo-tp61p260.html](http://official-rtab-map-forum.67519.x6.nabble.com/rtabmap-for-kinect-v2-on-ROS-Indigo-tp61p260.html)
The tutorial website for RTAB-MAP ROS also indicates that a special launch file is included for Kinect V2:
[http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping)
More specifically:
`roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true`

yalan2017, 2017-02-23:
thank you, i get it

matlabbe, 2017-02-23:
The Windows binaries have also Kinect v2 support out-of-the-box. See this [page](https://github.com/introlab/rtabmap/wiki/Installation#windows) for driver installations:

> Driver for the Kinect v2: UsbDK is now used. See libfreenect2 page for more info.

cheers,
Mathieu

eric-schleicher, 2017-03-15:
@yalan2017  were you able to get the kinect2 with RTAB in windows?  or ROS on linux?

Looking to do the same.

yalan2017, 2017-03-17:
@eric-schleicher  i have installed on ubuntu 14.04 (ROS-indigo).
Good luck

proepkes, 2017-04-05:
I tried rtabmap on Ubuntu 16.06 with kinectv2 but the texture always shows some black bars/pixels
My guess is that this is due to how the kinectv2 works, take a look at the images at https://github.com/code-iai/iai_kinect2#screenshots
The rgb-d image shows some black pixels.. and when you're recording and map these pixels to your "scene", your texture will get some black spots in it

The problem does not occur with kinect1

matlabbe, 2017-04-05:
Hi, 

yes, good observation! I'm thinking about three approaches:

- Maybe there is a parameter in `iai_kinect2` package to "zoom" or "crop" the rgb image after rectification (to avoid black contours, similar to `alpha` parameter of [cv::stereoRectify](http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#stereorectify)). The registered depth image and camera_info would be also modified to match the new rgb image size or focal.

- Another way would be to change how texturing is done in rtabmapviz, like adding a region of interest on the projected cameras. I put this issue as enhancement for this point.

- I may need to reverify, but kinect v2 calibration with rtabmap's calibration tool would not produce black areas after rectification, `alpha=0` is used with `cv::stereoRectify` [here](https://github.com/introlab/rtabmap/blob/d21ae76fff23a21582b77af6c6f2da5fe5301aed/guilib/src/CalibrationDialog.cpp#L908-L911). The output calibration of rtabmap may be compatible with the one used by `kinect2_bridge`, so a copy/paste may fix the problem.

On Kinect v1, the RGB image is not rectified (it is assumed taken from a very low distortion len), so RGB fills all the image.

cheers,
Mathieu

eric-schleicher, 2017-04-14:
I think this need a wiki page. when i come around to calibrating the kinect2 i'll take this on.

matlabbe, 2017-04-20:
[image]

Commit above should fix the problem when exporting with texture. For example, setting  "`0.1 0.0 0.05 0.0`" will ignore 10% of the right and 5% of the top of image.

For black points in the 3D view while mapping, the same parameter can be set under Preferences->3D Rendering (ratio should give final size divisible by decimation factor).

cheers

closed 2017-04-20 by matlabbe