introlab/rtabmap #173: V0.12.1 Octomap issues

SherylZhou, 2017-03-06:
Hi,

I installed the rtabmap version 0.12.1 and tried it with our own databases and online sample RGBD datasets with the octomap options enabled, but after a session is ended, the "export octomap" option is always disabled. It worked in v0.11.14. Another issue is that I have trouble viewing the octomap in rviz. It complained that no octotree is created. I am using virtualbox ubuntu 14.04. Could you please provide examples/steps to get them work?

Thanks,
Sheryl

matlabbe, 2017-03-08:
For the standalone, see this [post](http://official-rtab-map-forum.67519.x6.nabble.com/Can-t-export-octomap-file-tp2743p2750.html).

For ROS Indigo, if you have this error "Failed to create octree structure":
[image]

It is because we need the ColorOctree rviz plugin. Remove binaries `ros-indigo-octomap-rviz-plugins` and build the indigo-devel branch of the [github project](https://github.com/OctoMap/octomap_rviz_plugins):
```bash
$ sudo apt-get remove ros-indigo-octomap-rviz-plugins
$ cd ~/catkin_ws/src
$ git clone -b indigo-devel https://github.com/OctoMap/octomap_rviz_plugins.git
$ cd ..
$ catkin_make
```
Now in RVIZ, select "ColorOccupancyGrid":
[image]

One last note: On ROS, if you are subscribed to a laser scan (like the demo robot mapping example), please see this [post](http://official-rtab-map-forum.67519.x6.nabble.com/Build-octomap-with-rtabmap-tp68p2694.html).

cheers,
Mathieu

SherylZhou, 2017-03-14:
It works now, thanks!

closed 2017-03-14 by SherylZhou