introlab/rtabmap #176: Android: Ability to set save location in preferences settings.

eric-schleicher, 2017-03-13:
It would be nice to have the ability to specify a location where we would live to save the MAPs in the android applications settings.

I would like to start saving captured maps directly to another location (other than the default location; external storage connected via an OTG usb reader).

If there is a way to do this now (even if it's editing a file) just let me know.

matlabbe, 2017-03-14:
Hi,

The export directory is set on this [line](https://github.com/introlab/rtabmap/blob/1bd9d411648ab631b0526311e3c9df239f67c6f1/app/android/src/com/introlab/rtabmap/RTABMapActivity.java#L1734):
```cpp
File exportDir = new File(mWorkingDirectory + RTABMAP_EXPORT_DIR);
```
Where `RTABMAP_EXPORT_DIR` is just sub directory "Export/" and `mWorkingDirectory` is defined [here](https://github.com/introlab/rtabmap/blob/1bd9d411648ab631b0526311e3c9df239f67c6f1/app/android/src/com/introlab/rtabmap/RTABMapActivity.java#L312-L313):
```cpp
if(Environment.getExternalStorageState().compareTo(Environment.MEDIA_MOUNTED)==0)
{
	File extStore = Environment.getExternalStorageDirectory();
	mWorkingDirectory = extStore.getAbsolutePath() + "/" + getString(R.string.app_name) + "/";
```
If [seems](https://developer.android.com/reference/android/os/Environment.html#getExternalStorageState(java.io.File)) that we can give a path to `Environment.getExternalStorageState()` to get another storage drive. Not sure how to use it though. By default it gets the path to SD card.

cheers,
Mathieu

eric-schleicher, 2017-03-14:
This StackExchange thread has some related and potentaily helpful information:

[Android - API 23 Unable to get USB/SD Card Directory
](http://stackoverflow.com/questions/41012320/android-api-23-unable-to-get-usb-sd-card-directory)

I'll see what I can do.