introlab/rtabmap #180: Exporting cloud to disk operatoin exceeds available storage fails silently

eric-schleicher, 2017-03-22:
just noting that if you export a cloud and run a disk out of space, RTAB map doesn't provide any type of error notification and just completes successfully.

Not a blocking issue or high priority.  but i'm guessing the cloud size can be projected before export?

matlabbe, 2017-03-22:
Is that for RTAB-Map Tango or RTAB-Map Desktop? On Desktop, there should be [error messages in the progress dialog](https://github.com/introlab/rtabmap/blob/e003fb8d3533b41942aa8d45df68a886553e6d0c/guilib/src/ExportCloudsDialog.cpp#L2415) or a [message box](https://github.com/introlab/rtabmap/blob/e003fb8d3533b41942aa8d45df68a886553e6d0c/guilib/src/ExportCloudsDialog.cpp#L2362) telling that saving failed. On Tango, it would [say](https://github.com/introlab/rtabmap/blob/5b08a69607f7693392a4eeb9a7abc294d1b060fd/app/android/src/com/introlab/rtabmap/RTABMapActivity.java#L1826) that export has failed. On Tango, there is indeed no information about what caused the error (e.g., saving to disk failed, out of memory). 

cheers,
Mathieu

eric-schleicher, 2017-03-22:
This was on desktop.

Also ***not*** on the primary (boot) disk.

matlabbe, 2017-03-22:
I reproduced the bug. I tried saving a 450 MB cloud on a USB drive have 140 MB left. It outputs a ply file of 140 MB, but it cannot be opened in MeshLab (unexpected EOF).

It seems to be a PCL bug: https://github.com/PointCloudLibrary/pcl/blob/05e04426804bf00641c194d660b268ed87f5b06b/io/src/ply_io.cpp#L1443-L1444

After `close()`, it should check if the "bad" bit is set like in this [post](http://stackoverflow.com/questions/28342660/error-handling-in-stdofstream-while-writing-data):
```cpp
fpout.close();
if ( ! fpout ) {
    PCL_ERROR ("[pcl::PLYWriter::writeBinary] Error after closing (%s)!\n", file_name.c_str ());
    return (-1);
}
```

I'll try to check this soon (rebuilding PCL with this fix) and make a pull request to PCL library if it is the problem.

The only way now to know if export has failed for this reason is to check the drive size after export: if 0 MB space left, it is likely that the exported file is not complete.

cheers,
Mathieu