introlab/rtabmap #183: Is it possible to integrate rtabmap with Unity?

god2g, 2017-04-14:
I want to use rtabmap to reconstruct the environment, but use Unity as render engine, is it possible?How to do that?Thanks.

matlabbe, 2017-04-19:
I've never coded in Unity, though I think we can do c++ plugins. So for your question, no, Unity is not supported. Google Tango API has some examples using their mapping pipeline, we may inspire from this. Hope rtabmap will be integrated to Unity in the future to get game level rendering/physics with rtabmap's maps.

cheers,
Mathieu

TheBricktop, 2019-10-21:
Coming from unity background im currently trying to get better intel realsense integration for slam than intel itself can provide ( when one guy complained that old realsense sdk integrated slam, handtracking, facetracking and object recognition into one package, they pointed him to Your repo )
how one could start to port Your system to unity in some kind of plugin that would simply provide odometry for some of the sensors ( realsense, orbec, or generic stereo camera) ?

matlabbe, 2019-10-22:
To simply provide odometry, this could be done relatively easily for someone used to unity and c++ (I think!). Using RTAB-Map API for realsense2 cameras (inspired by this C++ [example](https://github.com/introlab/rtabmap/blob/master/examples/NoEventsExample/main.cpp)):

```cpp
// Includes
#include <rtabmap/core/camera/CameraRealSense2.h>
#include <rtabmap/core/Odometry.h>
using namespace rtabmap;

// Initialize objects
CameraRealSense2 camera;
Odometry * odom = Odometry::create();
SensorData data = camera.takeImage();

// Loop
while(data.isValid())
{
  OdometryInfo info;
  Transform pose = odom->process(data, &info);
  if(!pose.isNull())
  {
    // Send pose here to Unity?
    // The pose is in camera coordinate system: 
    //    is x->right, y->down, z->forward
  }
  else
  {
    // Camera is lost!
  }
  data = camera.takeImage();
}
delete odom;
```

For any other camera drivers, look at the ones available here: https://github.com/introlab/rtabmap/tree/master/corelib/include/rtabmap/core/camera, RTAB-Map should be built with their support though.

cheers,
Mathieu

TheBricktop, 2019-10-24:
Im very mych used to unity but still dabbling at c++ so it would be a slight challange to me