introlab/rtabmap #185: Pose Saved in DB

vidit09, 2017-05-02:
Hi.
Thanks for excellent application.I'm using the android application and saving the database of the map. I've a question regarding the pose saved in Node table of the database. I can see on the device that the trajectory is being corrected according to the loop closing but when I access the pose from the  saved database on PC, I get back the un-corrected version of the trajectory. 
Am I accessing the correct pose matrix ? Do I need to do other operation before saving the db?

matlabbe, 2017-05-02:
Hi,

I assume your are referring to this "[pose](https://github.com/introlab/rtabmap/blob/595ac4955208155634cf518a2125602a2040542f/corelib/src/resources/DatabaseSchema.sql.in#L21)" field in the database. The format of this blob field is 12 floats (3x4 transformation matrix). As you observed, it is indeed the odometry pose, not the optimized one.

To get the corrected pose, the whole graph must be loaded and optimized. You can use rtabmap standalone to do it for you ([binaries](https://github.com/introlab/rtabmap/releases) available). After opening the database in rtabmap, do File->Export poses...:
[image]

You can also export odometry poses by re-downloading the graph and select not optimized:
[image]
[image]

cheers,
Mathieu

vidit09, 2017-05-02:
Thanks Mathieu for the prompt reply. I'm on CentOS and was trying to build rtabmap for it but had to abandon it due to some library conflict on my system. Is there a way to directly get corrected pose from android app? If not then I'll resume building the binaries.

matlabbe, 2017-05-02:
It is not possible from the android app. A virtual box with ubuntu 16.04 could be a trade-off to build rtabmap. What are the conflicts on CentOS? If you can get PCL, OpenCV and Qt installed, rtabmap should be able to build correctly.

vidit09, 2017-05-02:
I was able to build the rtabmap but during run there was conflict with the runtime library path. I'll try to resolve it. thanks for your inputs.

closed 2017-05-02 by vidit09