introlab/rtabmap #187: Seg Fault on CentOS when using database viewer

vidit09, 2017-05-03:
Hi,
I was able to successfully create the binaries for the CentOS, but when I use database-viewer executable it crashes after loading the db. Here is terminal log
`[ WARN] (2017-05-03 09:08:47.330) Parameters.cpp:771::readINI() Section "Core" in /home/viditv/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when this node will close.`
`[ INFO] (2017-05-03 09:11:14.194) DBDriverSqlite3.cpp:343::connectDatabaseQuery() Using database "/local/rtab_dataset/5MappingSessions_OnlineConstraint700ms.db" from the hard drive.`
`[ INFO] (2017-05-03 09:11:14.195) DBDriverSqlite3.cpp:382::connectDatabaseQuery() Database version = 0.0.0`
`[ INFO] (2017-05-03 09:11:14.196) DatabaseViewer.cpp:1285::updateIds() Loading all IDs...`
`[ INFO] (2017-05-03 09:11:15.007) DatabaseViewer.cpp:1400::updateIds() Loaded 2256 ids, 2074 poses and 2515 links`
`[ INFO] (2017-05-03 09:11:15.007) DatabaseViewer.cpp:1408::updateIds() Update database info...
Segmentation fault (core dumped)
` 
Any idea what could have gone wrong? I was testing with example db provided [here](https://github.com/introlab/rtabmap/wiki/Tools).
Thanks 
Vidit

vidit09, 2017-05-03:
dmesg shows this in the log
`rtabmap[15826]: segfault at 38 ip 00007f41490e7126 sp 00007fff1ab09c20 error 4 in libpcl_visualization.so.1.7.2[7f414902e000+161000]`

matlabbe, 2017-05-05:
Looks like a bug with QVTKWidget and PCLVisualizer on that platform. Try replacing line 232 by 231 [here](https://github.com/introlab/rtabmap/blob/f1d96a30a40432ac9be741b1870af32c6c21ae5a/guilib/src/CloudViewer.cpp#L229-L232) in CloudViewer:
```cpp
// Replaced by the second line, to avoid a crash in Mac OS X on close, as well as
// the "Invalid drawable" warning when the view is not visible.
//_visualizer->setupInteractor(this->GetInteractor(), this->GetRenderWindow());
this->GetInteractor()->SetInteractorStyle (_visualizer->getInteractorStyle());
```

If it still crash, you may disable the visualizer to avoid the crash, if you don't need to visualize 3D constraints. Comment these [lines](https://github.com/introlab/rtabmap/blob/f1d96a30a40432ac9be741b1870af32c6c21ae5a/guilib/src/CloudViewer.cpp#L227-L232):
```cpp
this->SetRenderWindow(_visualizer->getRenderWindow());
this->GetInteractor()->SetInteractorStyle (_visualizer->getInteractorStyle());
```

cheers

vidit09, 2017-05-09:
Crash persists with the first change and has same dmesg log as the one before.
When commenting both the mentioned lines crash happens but with different dmesg log 
`rtabmap[14094]: segfault at 0 ip 00007ff29270e451 sp 00007ffdb97a5220 error 4 in librtabmap_gui.so.0.12.5[7ff2924ff000+3a7000]`

matlabbe, 2017-05-10:
I installed CentOS 7 in a virtual box and got rtabmap-databaseViewer working without your error:
[image]

I installed these default binaries available on CentOS: opencv2, qt5, boost
CMake3 is required for VTK: `wget -nv https://cmake.org/files/v3.7/cmake-3.7.2-Linux-x86_64.tar.gz`
I built from source: eigen3, flann, VTK latest, PCL latest:
```
$ wget -nv http://bitbucket.org/eigen/eigen/get/3.2.7.tar.gz
$ wget -nv http://www.cs.ubc.ca/research/flann/uploads/FLANN/flann-1.8.4-src.zip
$ git clone https://github.com/Kitware/VTK.git
$ git clone https://github.com/PointCloudLibrary/pcl.git
```
Set VTK cmake config to build with Qt5. PCL should build with visualization module. The only issue is that 3D view shows nothing and there are VTK errors complaining about opengl2, though this may be related to use of the Virtual machine (opengl not correctly installed).

cheers,
Mathieu

vidit09, 2017-05-10:
Right now I've moved to MacOS , where things are working fine for me. I'll try to re-install the dependencies again on CentOS. 
Thanks
Vidit

closed 2018-10-17 by vidit09