introlab/rtabmap #192: ZED camera SVO file input with standard sensing mode fatal error

mtee, 2017-05-09:
Whenever I select a ZED SVO file as a source and leave the sensing mode as Standard the program crashes upon start. The error I see in the console is

> "CameraStereo.cpp:824::CameraStereoZed() Condition (sensingMode_ >= sl::SENSING_MODE_FILL && sensingMode_ <sl::SENSING_MODE_LAST) not met!"

Which makes me wondering if we only can use FILL mode with the SVO input, since the error does not occur when the sensing mode is set to FILL. ZED sdk version is 2.0, OS is linux mint 18.

matlabbe, 2017-05-09:
I updated the code, there was a missing change for the SVO constructor interface (ZED SDK 2.0 has changed order of the sensing enum).

closed 2017-05-09 by matlabbe

matlabbe, 2017-05-09:
You may want this fix too for SVO input: https://github.com/introlab/rtabmap/commit/474011033b9742dcc797d5b39ace07ccd2812b18
otherwise the scale is wrong!

commit 9d18f7b1 (2017-05-09): Update CameraStereo.cpp

fixed #192