introlab/rtabmap #198: Segmentation Fault when exporting calibration data with freenect2

mtee, 2017-05-19:
Hello again, this time I am facing a segfault when exporting calibration data acquired from kinect 2. It worked once for me and never since then, and sadly I am not sure, whether anything changed on my side. It is quite possible that I have updated or downgraded the VTK lib since then.
Nevertheless here is all information I could gather:
The program crashes when I click on "Export Calibration Data" after the calibration is performed. 
I tried to debug it in Eclipse, it crashes around line 994 in CalibrationDialog.cpp, the console message is 
```
Thread 1 "rtabmap" received signal SIGSEGV, Segmentation fault.
__GI___pthread_mutex_lock (mutex=0x0) at ../nptl/pthread_mutex_lock.c:67
```

dmesg reads:
`[11685.504295] traps: rtabmap[9651] general protection ip:7f690e525d44 sp:7ffc4de71870 error:0 in libpthread-2.23.so[7f690e51c000+18000]`
...
`[13279.894210] pool[15098]: segfault at a ip 00007fd403f51240 sp 00007fd394ba17e0 error 4 in libc-2.23.so[7fd403ed1000+1bf000]`
This is the output of cmake: 

```
Checking for module 'openni-dev'
  No package 'openni-dev' found
Checking for module 'openni2-dev'
  No package 'openni2-dev' found
Checking for module 'openni-dev'
  No package 'openni-dev' found
Checking for module 'openni2-dev'
  No package 'openni2-dev' found
Checking for module 'openni-dev'
  No package 'openni-dev' found
Found OpenMP
Found OpenCV: /usr/local/include;/usr/local/include/opencv
Found PCL: /usr/local/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-5.10
Found ZLIB: /usr/local/include
Found freenect2: /home/username/freenect2/include
Found OpenNI2: /usr/include/openni2
Found DC1394: /usr/include/dc1394
Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
Found ZED sdk: /usr/local/zed/include
Found CUDA: /usr/local/cuda-8.0/include
Found Pthreads
--------------------------------------------
Info :
  Version : 0.12.5
  CMAKE_INSTALL_PREFIX = /usr/local
  CMAKE_BUILD_TYPE =     DEBUG
  CMAKE_INSTALL_LIBDIR = lib
  BUILD_APP =            ON
  BUILD_TOOLS =          ON
  BUILD_EXAMPLES =       ON
  BUILD_SHARED_LIBS =    ON
  CMAKE_CXX_FLAGS =  -fmessage-length=0  -Wno-deprecated -fopenmp -std=c++11
  With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = NO (not found, License: BSD)
  With Freenect             = NO (WITH_FREENECT=OFF)
  With OpenNI2              = YES (License: Apache v2)
  With Freenect2            = YES (License: Apache v2 and/or GPLv2)
  With dc1394               = YES (License: LGPL)
  With FlyCapture2/Triclops = NO (Point Grey SDK not found)
  With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
  With g2o                  = YES (License: BSD)
  With GTSAM                = NO (GTSAM not found)
  With VERTIGO              = YES (License: GPLv3)
  With cvsba                = NO (cvsba not found)
  With ZED                  = YES (With CUDA)
  With RealSense            = NO (librealsense not found)
  With OCTOMAP              = NO (octomap not found)
  With CPUTSDF              = NO (CPUTSDF not found)
  With Qt4                  = YES (License: Open Source or Commercial)
--------------------------------------------
```
libQVTK version is 5.10 and QT is 4.8.7
let me know if there is any other information I need to provide

mtee, 2017-05-19:
The reason seems to be that the `savingDirectory_` is null in the CalibrationDialog::save function. I am now investigating why is it null, the rest of the post can be ignored I guess.

matlabbe, 2017-05-21:
Yes normally `savingDirectory_` should not be "null". Could it be similar to this [bug](http://official-rtab-map-forum.67519.x6.nabble.com/RTab-Map-closes-unexpectedly-when-I-m-trying-to-save-database-tp1149p1448.html) where Qt crashes when opening a native dialog?