introlab/rtabmap #199: Using Different Sensors for Mapping and Relocalization

aseyfi, 2017-05-22:
Hello,

I would like to use two different sensors for mapping and re-localization. For example I want to use Tango for the mapping and use another sensor with similar information (odometry, depth, rgb), but with different sensor properties (e.g. different focal length of RGB camera). 
Would you please let me know how tightly is the map coupled with the sensor technology and data?
Also, Can I use Tango for the mapping, but only use a different single RGB camera (no depth and odometry) for re-localization?

Thank You,
Ahmad

matlabbe, 2017-05-23:
Hi Ahmad,

The two approaches are possible. See this [post](http://official-rtab-map-forum.67519.x6.nabble.com/Localize-in-a-pre-loaded-local-map-created-by-RGB-D-mapping-tp2383p2399.html) for RGB-only localization case with a previously built map. **However**, a different RGB image size will affect how 2D features are extracted. Some cameras have also different auto white-balance or auto-exposure settings that can generate different visual features for the same scene. I tested with a Tango database and a Kinect camera (using only RGB image) but localization fails. The only way to get some localizations is to reprocess the Tango database with SURF features instead of the default GFTT/BRIEF for the vocabulary, then try again to localize the kinect against this new database.

Some parameters I changed on Tango in Settings->Mapping...: 
 * Update Rate=Max, 
 * Max Features Extracted (voc/loop)=1000, 
 * Optimize from Graph End=false

cheers,
Mathieu

aseyfi, 2017-07-14:
@matlabbe , considering both mapping and re-localization accuracy, if you have the option of changing the RGB camera on Lenovo Phab2 Tango phone, what specifications would you recommend for image resolution, frame rate, shutter type, field of view, time stamp accuracy, physical location of RGB camera respect to ToF, ...?

Thanks,
Ahmad

matlabbe, 2017-07-24:
I would start by using a motion blur-less camera (fast global shutter) similar to what is in the [BumbleBee2](https://www.ptgrey.com/bumblebee2-firewire-stereo-vision-camera-systems) stereo camera, and maybe better physical optical stabilization. Resolution/frame rate depend more on processing power available. Time stamp accuracy with motion module of Tango is required.

cheers,
Mathieu