introlab/rtabmap #200: change Sift to BRIEF or Orb

forthealllight, 2017-05-25:
I am a Green hand in SLAM , thank you for your souce code , I have a idea that in order to speed up to form  dictionary ，may I use BRIEF or Orb descriptor (like DBW2 or  ORB-SLAM)? if so , recall will be influenced or not  ?  thank you~

matlabbe, 2017-05-25:
Yes you can choose any other detector (vocabulary or odometry). Loop closure recall performance could be indeed affected. However, I'm using often GFTT/BRIEF (or ORB) and all loop closures (seen from same point of view) are detected most of the time. I didn't compare with datasets of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf) (using SURF).

cheers

forthealllight, 2017-05-26:
thank you very much

forthealllight, 2017-05-30:
as you know

closed 2017-05-30 by forthealllight