introlab/rtabmap #202: why STM size is always  set to 30

forthealllight, 2017-05-29:
from your source code， I see that the STM size is always set to 30 regardless  the database size。
I don not know why STM size is always set to this，In my opnion , for big scene ， the STM size is set to small， but  for big one ,it is set to bigger?  Can  you give some tip ,Why STM size is 30 , thank you very much

matlabbe, 2017-05-29:
From this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf):

> STM is used to observe similarities through time between consecutive images for weight updates, while the role of the WM is to detect loop closures between locations in space. Similar to [17], RTAB-Map does not use locations in STM to avoid loop closures on locations that have just been visited (because most of the time, the last location frequently looks similar to the most recent ones). The STM size is set based on the robot velocity and the rate at which the locations are acquired.

Therefore STM size is independent of the database size, but dependent on the robot speed and detection rate of rtabmap. Another way to see that is if the environment changes fast while the robot move, STM can be smaller. If environment changes slowly, STM should be higher. How fast the environment changes is dependent on robot velocity, detection rate and how far are the objects (e.g., environment would change faster indoor than outdoor for the same speed and same detection rate).

If processing time is not an issue, the working memory (WM) will increase in size as the database is getting bigger.

cheers,
Mathieu

forthealllight, 2017-05-30:
thank you for your answer~

closed 2017-05-30 by forthealllight