introlab/rtabmap #204: why the hyp is  lower than the loopThr  even if  the two frames is truely positive

forthealllight, 2017-06-06:
Hi, why the hyp is  lower than the loopThr  even if  the two frames is truely positive,of course the compared frames is in WM. In my opnion , the loopThr between them should be high.

                                                                                        sincerely

matlabbe, 2017-06-06:
The hypothesis will be high if the location is very different than all other locations. A location with a lot of common visual features will score very low. Note also that generally a minimum of 20/30 locations in WM is required to start getting meaningful hypotheses.

You can look at the likelihood curve (View->Show View->Likelihood) to see if the score is high. You can compare it to "A Posteriori PDF" curve too, which are the loop closure hypotheses. A sequence of matching images is often required so that hypothesis increases over the threshold.

cheers

forthealllight, 2017-06-07:
thank you , I  think that I get your idear

closed 2017-06-07 by forthealllight