introlab/rtabmap #205: RTABMAP Windows C++ example

ahsteven, 2017-06-07:
I installed RTABMAP with the windows binaries and am trying to run Loop closure C++ example at the bottom of the tutorials page. I am able to build the project successfully in Visual Studio 2015 however when I run it with the sample images provided I get an exception at the line:

rtabmap::CameraImages camera(path);

Unhandled exception at 0x00007FFCADE09E08 in example.exe: Microsoft C++ exception: std::bad_alloc at memory location 0x00000041986FBF60.

Then I saw that camera is just a wrapper for cv::imread so I then tried reading the images in by the filenames manually.

Then I get a read access violation exception while executing:

parameters.insert(rtabmap::ParametersPair(rtabmap::Parameters::kRGBDEnabled(), "false"));

Is it because I am missing a .lib file?

When I generated the VS project with cmake the library linker additional dependencies did not turn out good, see screenshot below:

[image]

I did not think that this loop closure example used g20 and some other dependencies anyway so I removed them. Any idea on why these .lib linker dependencies got loaded like this by cmake?

So after removing these I just have the rtabmap .lib files and teh opencv lib files as shown below
[image]

It is with these settings that I get the exceptions I mention above. the 32.lib linker files look suspicious.

What is the easiest way to run a c++ rtabmap example on Windows? Thanks for any help.

matlabbe, 2017-06-07:
Make sure you executed cmake in a Win64 terminal. Also, what is the output after the original cmake (when build folder is empty)?
```
$ cmake -G"Visual Studio 14 2015 Win64" ..
```
OpenCV should also been built the same way in a Win64 terminal, if you don't use the binaries.

cheers,
Mathieu

ahsteven, 2017-06-07:
Hi Mathieu,

I build this RTABMAP project as well as my opencv in cmake gui
selecting  Visual
Studio 14 2015 Win64 as the compiler.

Shouln't that work?

When I run the cmake command you posted above from say a bash shell or the
powershell the results are the same.

The command line output is from the powershell build is:

-- The C compiler identification is MSVC 19.0.24213.1
-- The CXX compiler identification is MSVC 19.0.24213.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual
Studio 14.0/VC/bin/x86_amd64/cl.
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual
Studio 14.0/VC/bin/x86_amd64/cl.
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual
Studio 14.0/VC/bin/x86_amd64/c
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual
Studio 14.0/VC/bin/x86_amd64/c
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: C:/Users/Adam/opencv/build (found version "3.2.0")
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Adam/Documents/Visual Studio
2015/Projects/RTABMAP/build3

matlabbe, 2017-06-08:
Hi,

there is my cmake status on Windows 10 / Visual Studio 2015:
```bash
C:\workspace\rtabmap\build>cmake -G"Visual Studio 14 2015 Win64" ..
-- The C compiler identification is MSVC 19.0.23918.0
-- The CXX compiler identification is MSVC 19.0.23918.0
-- Check for working C compiler using: Visual Studio 14 2015 Win64
-- Check for working C compiler using: Visual Studio 14 2015 Win64 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Visual Studio 14 2015 Win64
-- Check for working CXX compiler using: Visual Studio 14 2015 Win64 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found eigen: C:/Program Files/Eigen/include/eigen3
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS)
-- Found Flann: C:/Program Files/flann/lib/flann_cpp_s.lib
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS)
-- Found qhull: C:/Program Files/qhull/lib/qhullstatic.lib
-- Found PCL_COMMON: C:/Program Files/PCL/lib/pcl_common_release.lib
-- Found PCL_OCTREE: C:/Program Files/PCL/lib/pcl_octree_release.lib
-- Found PCL_IO: C:/Program Files/PCL/lib/pcl_io_release.lib
-- Found PCL_KDTREE: C:/Program Files/PCL/lib/pcl_kdtree_release.lib
-- Found PCL_SEARCH: C:/Program Files/PCL/lib/pcl_search_release.lib
-- Found PCL_SAMPLE_CONSENSUS: C:/Program Files/PCL/lib/pcl_sample_consensus_release.lib
-- Found PCL_FILTERS: C:/Program Files/PCL/lib/pcl_filters_release.lib
-- Found PCL_2D: C:/Program Files/PCL/include/pcl-1.8
-- Found PCL_GEOMETRY: C:/Program Files/PCL/include/pcl-1.8
-- Found PCL_FEATURES: C:/Program Files/PCL/lib/pcl_features_release.lib
-- Found PCL_ML: C:/Program Files/PCL/lib/pcl_ml_release.lib
-- Found PCL_SEGMENTATION: C:/Program Files/PCL/lib/pcl_segmentation_release.lib
-- Found PCL_VISUALIZATION: C:/Program Files/PCL/lib/pcl_visualization_release.lib
-- Found PCL_SURFACE: C:/Program Files/PCL/lib/pcl_surface_release.lib
-- Found PCL_REGISTRATION: C:/Program Files/PCL/lib/pcl_registration_release.lib
-- Found PCL_KEYPOINTS: C:/Program Files/PCL/lib/pcl_keypoints_release.lib
-- Found PCL_TRACKING: C:/Program Files/PCL/lib/pcl_tracking_release.lib
-- Found PCL_RECOGNITION: C:/Program Files/PCL/lib/pcl_recognition_release.lib
-- Found PCL_STEREO: C:/Program Files/PCL/lib/pcl_stereo_release.lib
-- Found PCL_OUTOFCORE: C:/Program Files/PCL/lib/pcl_outofcore_release.lib
-- Found PCL_PEOPLE: C:/Program Files/PCL/lib/pcl_people_release.lib
-- Try OpenMP C flag = [/openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [/openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: /openmp
-- Found OpenMP
-- Found OpenCV: C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.1.0/include;C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.1.0/include/opencv
-- Found PCL: C:/Program Files/PCL/include/pcl-1.8;C:/Program Files/Eigen/include/eigen3;C:/boost_1_61_0;C:/Program Files/OpenNI2/Include;C:/Program Files (x86)/Intel/RSSDK/include;C:/Program Files/VTK/include/vtk-6.3;C:/Program Files/flann/include;C:/Program Files/qhull/include
-- Found ZLIB: C:/Program Files/zlib/include
-- Found freenect2: C:/workspace/libfreenect2/build/install/include
-- Found OpenNI2: C:/Program Files/OpenNI2/Include
-- Found CSPARSE: C:/Program Files/g2o/include/EXTERNAL/csparse
-- Found g2o: C:/Program Files/g2o/include;C:/Program Files/g2o/include/EXTERNAL/csparse
-- GTSAM include directory:  C:/Program Files/GTSAM/CMake/../include;C:/Program Files/GTSAM/include/gtsam/3rdparty/Eigen/
-- Found FlyCapture2: C:/Program Files/Point Grey Research/FlyCapture2/include;C:/Program Files/Point Grey Research/Triclops Stereo Vision SDK/include
-- Found cvsba: C:/Program Files/cvsba/include
-- Found ZED sdk: C:\Program Files (x86)\ZED SDK\include
-- Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v8.0 (found version "8.0")
-- Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v8.0/include
-- Found RealSense: C:/workspace/librealsense/include
-- Found octomap: C:/Program Files/octomap-distribution/include
-- Found CPUTSDF: C:/Program Files/cpu_tsdf/include
-- --------------------------------------------
-- Info :
--   Version : 0.12.5
[... 108 lines cut ...]
 #69 ptime(0.038375s) STM(10) WM(59) hyp(30) value(0.49) *LOOP 69->30*
 #70 ptime(0.038834s) STM(10) WM(60) hyp(31) value(0.47) *LOOP 70->31*
 #71 ptime(0.037154s) STM(10) WM(61) hyp(32) value(0.45) *LOOP 71->32*
 #72 ptime(0.042587s) STM(10) WM(62) hyp(33) value(0.48) *LOOP 72->33*
 #73 ptime(0.047081s) STM(10) WM(63) hyp(34) value(0.44) *LOOP 73->34*
 #74 ptime(0.048294s) STM(10) WM(64) hyp(35) value(0.45) *LOOP 74->35*
 #75 ptime(0.034725s) STM(10) WM(65) hyp(36) value(0.52) *LOOP 75->36*
 #76 ptime(0.042964s) STM(10) WM(66) hyp(37) value(0.68) *LOOP 76->37*
 #77 ptime(0.037979s) STM(10) WM(67) hyp(38) value(0.79) *LOOP 77->38*
 #78 ptime(0.054263s) STM(10) WM(68) hyp(39) value(0.83) *LOOP 78->39*
 #79 ptime(0.027939s) STM(10) WM(69) hyp(39) value(0.82) *LOOP 79->39*
 #80 ptime(0.023227s) STM(10) WM(70) hyp(0) value(0.00)
 #81 ptime(0.035671s) STM(10) WM(71) hyp(41) value(0.78) *LOOP 81->41*
 #82 ptime(0.032199s) STM(10) WM(72) hyp(42) value(0.77) *LOOP 82->42*
 #83 ptime(0.037945s) STM(10) WM(73) hyp(3) value(0.79) *LOOP 83->3*
 #84 ptime(0.044736s) STM(10) WM(74) hyp(4) value(0.77) *LOOP 84->4*
Processing images completed. Loop closures found = 37
Generated graph "Graph.dot", viewable with Graphiz using "neato -Tpdf Graph.dot -o out.pdf"
Saving Long-Term Memory to "rtabmap.db"...
```

The OpenCV version is the one from ZED sdk, but it should be okay with the standard OpenCV binaries.

cheers,
Mathieu

ahsteven, 2017-06-08:
Thank you Mathieu. I noticed that the cmake file used for this bow example
is quite different from the cmake file from the Cplusplus Loop Closure
Detection webpage from 2015 here:

https://github.com/introlab/rtabmap/wiki/Cplusplus-Loop-Closure-Detection

I will try to work with this BOW example you have referred to here in your
response. I will also install the other dependencies you have which I do
not yet have.

Thank you.

ahsteven, 2017-06-08:
Mathieu,

I was able to compile and run with the cmake file from the 2016 BOW example
and produce the same output that you have.

However, I had to delete some of the library linker input dependencies that
it tried to include even though I did not have them installed.

It seems there is something wrong with the cmake when the dependencies are
not found. You can see what I mean below by taking a look at the additional
dependencies added to the visual studio project. Starting with zlib.

Thank you for your help.

kernel32.lib
user32.lib
gdi32.lib
winspool.lib
shell32.lib
ole32.lib
oleaut32.lib
uuid.lib
comdlg32.lib
advapi32.lib
C:\Program Files\RTABMap\lib\rtabmap_core.lib
C:\Program Files\RTABMap\lib\rtabmap_utilite.lib
C:\Program Files\RTABMap\lib\rtabmap_gui.lib
C:\Program
Files/zlib/lib/zlib.lib.lib
Files/g2o/lib/g2o_core_d.lib.lib
Files/g2o/lib/g2o_core.lib.lib
Files/g2o/lib/g2o_types_slam2d_d.lib.lib
Files/g2o/lib/g2o_types_slam2d.lib.lib
Files/g2o/lib/g2o_types_slam3d_d.lib.lib
Files/g2o/lib/g2o_types_slam3d.lib.lib
Files/g2o/lib/g2o_types_sba_d.lib.lib
Files/g2o/lib/g2o_types_sba.lib.lib
Files/g2o/lib/g2o_stuff_d.lib.lib
Files/g2o/lib/g2o_stuff.lib.lib
Files/g2o/lib/g2o_solver_csparse_d.lib.lib
Files/g2o/lib/g2o_solver_csparse.lib.lib
Files/g2o/lib/g2o_csparse_extension_d.lib.lib
Files/g2o/lib/g2o_csparse_extension.lib.lib
Files/g2o/lib/g2o_ext_csparse.lib.lib
Files/NVIDIA.lib
GPU.lib
Computing.lib
Toolkit/CUDA/v8.0/lib/x64/OpenCL.lib.lib
C:\workspace\libfreenect2\build\install\lib\freenect2.lib
Files/OpenNI2/Lib/OpenNI2.lib.lib
Files/Point.lib
Grey.lib
Research/FlyCapture2/lib64/FlyCapture2_v100.lib.lib
Research/Triclops.lib
Stereo.lib
Vision.lib
SDK/lib64/triclops_v100.lib.lib
SDK/lib64/flycapture2bridge_v100.lib.lib
SDK/lib64/pnmutils_v100.lib.lib
C:\workspace\librealsense\bin\x64\realsense.lib
Files/octomap-distribution/lib/octomap.lib.lib
Files/octomap-distribution/lib/octomath.lib.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_shape320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_stitching320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_superres320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_videostab320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_viz320d.lib
C:\Program Files\PCL
1.8.0\3rdParty\Boost\lib\libboost_system-vc140-mt-gd-1_61.lib
C:\Program Files\PCL
[... 147 lines cut ...]
C:\Program Files\PCL 1.8.0\lib\pcl_keypoints_debug.lib
C:\Program Files\PCL 1.8.0\lib\pcl_tracking_debug.lib
C:\Program Files\PCL 1.8.0\lib\pcl_recognition_debug.lib
C:\Program Files\PCL 1.8.0\lib\pcl_stereo_debug.lib
C:\Program Files\PCL 1.8.0\lib\pcl_outofcore_debug.lib
C:\Program Files\PCL 1.8.0\lib\pcl_people_debug.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_objdetect320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_calib3d320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_features2d320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_flann320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_highgui320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_ml320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_photo320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_video320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_videoio320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_imgcodecs320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_imgproc320d.lib
C:\Users\Adam\opencv\build\lib\Debug\opencv_core320d.lib
vfw32.lib
glu32.lib
opengl32.lib
comctl32.lib
wsock32.lib
ws2_32.lib
Psapi.lib

closed 2017-07-04 by matlabbe

frischzenger, 2023-05-12:
onecore\net\netprofiles\service\src\nsp\dll\namespaceserviceprovider.cpp(550)\nlansp_c.dll!00007FFB15FCD93D: (caller: 00007FFB3F24ACF6) LogHr(72) tid(793c) 8007277C 此服务不存在。在指定的命名空间中找不这个服务。
'RTABMap.exe' (Win32): Unloaded 'C:\Windows\System32\msftedit.dll'
The thread 0x8574 has exited with code 0 (0x0).
The thread 0x87dc has exited with code 0 (0x0).
The thread 0x79e8 has exited with code 0 (0x0).
The thread 0x7cc8 has exited with code 0 (0x0).
The thread 0x5f48 has exited with code 0 (0x0).
Exception thrown at 0x00007FFB3E7EFDEC in RTABMap.exe: Microsoft C++ exception: cv::Exception at memory location 0x000000483CE7E898.
Unhandled exception at 0x00007FFB3E7EFDEC in RTABMap.exe: Microsoft C++ exception: cv::Exception at memory location 0x000000483CE7E898.
----------------------------------------------------------------------------------------------------
i also meet this problem, have you try it on vs2019?

frischzenger, 2023-05-12:
onecore\net\netprofiles\service\src\nsp\dll\namespaceserviceprovider.cpp(550)\nlansp_c.dll!00007FFB15FCD93D: (caller: 00007FFB3F24ACF6) LogHr(72) tid(793c) 8007277C 
'RTABMap.exe' (Win32): Unloaded 'C:\Windows\System32\msftedit.dll'
The thread 0x8574 has exited with code 0 (0x0).
The thread 0x87dc has exited with code 0 (0x0).
The thread 0x79e8 has exited with code 0 (0x0).
The thread 0x7cc8 has exited with code 0 (0x0).
The thread 0x5f48 has exited with code 0 (0x0).
Exception thrown at 0x00007FFB3E7EFDEC in RTABMap.exe: Microsoft C++ exception: cv::Exception at memory location 0x000000483CE7E898.
Unhandled exception at 0x00007FFB3E7EFDEC in RTABMap.exe: Microsoft C++ exception: cv::Exception at memory location 0x000000483CE7E898.
------------------------------------------------------------------------
i 've also meet this problem, i have try it on vs2019, but crashed

frischzenger, 2023-05-12:
after read 3 pair of rgb/depth images, it crashes