introlab/rtabmap #210: Rtabmap Compilation Error

ziqB, 2017-06-22:
I've been trying to compile the source code but had some problems with it. I'm at the last step but apparently i got an error saying MSBUILD: error MSB1009: Project file does not exist.

I did some research and it says that i might have a spelling error for my PATH.

The following are the variables i put in my PATH:
C:\Users\qlin\Downloads\pcl-pcl-1.7.2\build\bin;
C:\Program Files\CMake\bin;C:\Program Files\zlib;
C:\Users\qlin\Downloads\rtabmap-0.12.5\build;
C:\Program Files\RTABMap\lib;C:\Users\qlin\Downloads\rtabmap-0.12.5\utilite/src;

am I missing something in the PATH?
Hope someone could help. Thanks

matlabbe, 2017-06-26:
Hi,

The PATH should contain only bin directories. You don't have to modify it for rtabmap folders, so you can remove "C:\Users\qlin\Downloads\rtabmap-0.12.5\build;
C:\Program Files\RTABMap\lib;C:\Users\qlin\Downloads\rtabmap-0.12.5\utilite/src;"

Did you successfully build PCL? Note that as PCL binaries are old (visual studio 2010), I prefer to rebuild all dependencies myself for Visual Studio 2015 for example.

Then which rtabmap installation steps are you referring from [here](https://github.com/introlab/rtabmap/wiki/Installation#source-1)?

ziqB, 2017-06-27:
I think I built it succesfully but i used the cmake GUI instead of the visual studio comand line. The comand line showed me that it had 1 warning and 12 errors. I'm currently on step 8. I placed the source file in this location: C:\Program Files\RTABmap. Im trying to use the command line but there will a lot of files that can not be found like open ni and a lot of PCL_....

matlabbe, 2017-06-27:
Copy/paste the cmake output here. Tohelp cmake finding PCL, you can add environment variable PCL_DIR:="directory/of/PCLConfig.cmake". PCLConfig.cmake is a file generated during PCL compilation.

ziqB, 2017-06-28:
**This is my output:**

Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Found OpenMP
Boost version: 1.50.0
Found the following Boost libraries:
system
filesystem
thread
date_time
iostreams
chrono
regex
Eigen found (include: C:/Program Files (x86)/Eigen/include, version: 3.0.5)
FLANN found (include: C:/Program Files/flann/include, lib: optimized;C:/Program Files/flann/lib/flann_cpp_s.lib;debug;C:/Program Files/flann/lib/flann_cpp_s-gd.lib)
Could NOT find LIBUSB_1 (missing: LIBUSB_1_LIBRARY LIBUSB_1_INCLUDE_DIR)
Could NOT find OpenNI (missing: OPENNI_LIBRARY OPENNI_INCLUDE_DIR)
Could NOT find OpenNI2 (missing: OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS)
Could NOT find FZAPI (missing: FZAPI_LIBS FZAPI_INCLUDE_DIR)
Could NOT find PXCAPI (missing: PXCAPI_LIBS PXCAPI_INCLUDE_DIRS)
Could NOT find PNG (missing: PNG_LIBRARY PNG_PNG_INCLUDE_DIR)
QHULL found (include: C:/Program Files/qhull/include, lib: optimized;C:/Program Files/qhull/lib/qhullstatic.lib;debug;C:/Program Files/qhull/lib/qhullstatic_d.lib)
Found unsuitable Qt version "" from NOTFOUND
VTK_MAJOR_VERSION 5
Could NOT find QVTK (missing: QVTK_LIBRARY QVTK_INCLUDE_DIR)
VTK found (include: C:/Program Files/VTK 5.8.0/include/vtk-5.8, lib: C:/Program Files/VTK 5.8.0/lib/vtk-5.8)
Could NOT find PCAP (missing: PCAP_LIBRARIES PCAP_INCLUDE_DIRS)
CMake Warning (dev) at cmake/pcl_targets.cmake:113 (if):
Policy CMP0054 is not set: Only interpret if() arguments as variables or
keywords when unquoted. Run "cmake --help-policy CMP0054" for policy
details. Use the cmake_policy command to set the policy and suppress this
warning.

Quoted variables like "EIGEN_FOUND" will no longer be dereferenced when the
policy is set to NEW. Since the policy is not set the OLD behavior will be
used.
Call Stack (most recent call first):
common/CMakeLists.txt:7 (PCL_SUBSYS_DEPEND)
This warning is for project developers. Use -Wno-dev to suppress it.
CMake Warning (dev) at cmake/pcl_targets.cmake:113 (if):
Policy CMP0054 is not set: Only interpret if() arguments as variables or
keywords when unquoted. Run "cmake --help-policy CMP0054" for policy
details. Use the cmake_policy command to set the policy and suppress this
warning.

Quoted variables like "FLANN_FOUND" will no longer be dereferenced when the
policy is set to NEW. Since the policy is not set the OLD behavior will be
used.
Call Stack (most recent call first):
kdtree/CMakeLists.txt:7 (PCL_SUBSYS_DEPEND)
This warning is for project developers. Use -Wno-dev to suppress it.
CMake Warning (dev) at cmake/pcl_targets.cmake:113 (if):
Policy CMP0054 is not set: Only interpret if() arguments as variables or
keywords when unquoted. Run "cmake --help-policy CMP0054" for policy
details. Use the cmake_policy command to set the policy and suppress this
warning.

Quoted variables like "FLANN_FOUND" will no longer be dereferenced when the
policy is set to NEW. Since the policy is not set the OLD behavior will be
used.
Call Stack (most recent call first):
search/CMakeLists.txt:7 (PCL_SUBSYS_DEPEND)
This warning is for project developers. Use -Wno-dev to suppress it.
CMake Warning (dev) at cmake/pcl_targets.cmake:113 (if):
Policy CMP0054 is not set: Only interpret if() arguments as variables or
keywords when unquoted. Run "cmake --help-policy CMP0054" for policy
details. Use the cmake_policy command to set the policy and suppress this
warning.

Quoted variables like "VTK_FOUND" will no longer be dereferenced when the
policy is set to NEW. Since the policy is not set the OLD behavior will be
used.
Call Stack (most recent call first):
visualization/CMakeLists.txt:17 (PCL_SUBSYS_DEPEND)
This warning is for project developers. Use -Wno-dev to suppress it.
DOXYGEN_FOUND
HTML_HELP_COMPILER
Found CPack generators: NSIS
The following subsystems will be built:
common
geometry
octree
io
kdtree
search
sample_consensus
filters
features
segmentation
visualization
surface
registration
keypoints
tracking
recognition
outofcore
people
tools
The following subsystems will not be built:
apps: OpenNI was not found.
examples: Code examples are disabled by default.
global_tests: No reason
Configuring done
Generating done

matlabbe, 2017-06-28:
This is the output of PCL cmake. VTK doesn't seem built with QT ("Could NOT find QVTK"). VTK should be built with Qt support before building PCL and rtabmap, so that in rtabmap you can have the UI. You can use cmake-gui to help seeing all options of cmake when building a project. So rebuild VTK with Qt, then PCL, then rtabmap.

cheers,
Mathieu

ziqB, 2017-06-29:
Hi Mathieu 
I did what you told me but when i build my PCL i got 1 warning and 11 errors most are LNK2019, is the error caused by PATH.
I did some research on the error LNK2019 and it says that the linker can't find a matching definition for an external symbol in any of the link files. 

The following is part of the output that says the error:

"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\install.vcxproj" (default target
) (1) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\ALL_BUILD.vcxproj" (default targ
et) (3) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\tools\pcl_add_gaussian_noise.vcx
proj" (default target) (4) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\common\pcl_common.vcxproj" (defa
ult target) (5) ->
(Link target) ->
  time_trigger.obj : error LNK2019: unresolved external symbol "bool __cdecl bo
ost::this_thread::interruptible_wait(void *,struct boost::detail::timeout)" (?i
nterruptible_wait@this_thread@boost@@YA_NPAXUtimeout@detail@2@@Z) referenced in
 function "public: bool __thiscall boost::detail::basic_cv_list_entry::wait(str
uct boost::detail::timeout)" (?wait@basic_cv_list_entry@detail@boost@@QAE_NUtim
eout@23@@Z) [C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\common\pcl_common.vc
xproj]
  time_trigger.obj : error LNK2019: unresolved external symbol "public: void __
thiscall boost::thread::detach(void)" (?detach@thread@boost@@QAEXXZ) referenced
 in function "public: __thiscall boost::thread::~thread(void)" (??1thread@boost
@@QAE@XZ) [C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\common\pcl_common.vcxp
roj]
  time_trigger.obj : error LNK2019: unresolved external symbol "class boost::sy
stem::error_category const & __cdecl boost::system::system_category(void)" (?sy
stem_category@system@boost@@YAABVerror_category@12@XZ) referenced in function "
public: __thiscall boost::thread_exception::thread_exception(int,char const *)"
 (??0thread_exception@boost@@QAE@HPBD@Z) [C:\Users\takrit\Downloads\pcl-pcl-1.7
.2\build\common\pcl_common.vcxproj]
  time_trigger.obj : error LNK2019: unresolved external symbol "public: void __
thiscall boost::thread::join(void)" (?join@thread@boost@@QAEXXZ) referenced in
function "public: __thiscall pcl::TimeTrigger::~TimeTrigger(void)" (??1TimeTrig
ger@pcl@@QAE@XZ) [C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\common\pcl_comm
on.vcxproj]
  time_trigger.obj : error LNK2019: unresolved external symbol "private: void _
_thiscall boost::thread::start_thread(void)" (?start_thread@thread@boost@@AAEXX
Z) referenced in function "public: __thiscall boost::thread::thread<class boost
::_bi::bind_t<void,class boost::_mfi::mf0<void,class pcl::TimeTrigger>,class bo
ost::_bi::list1<class boost::_bi::value<class pcl::TimeTrigger *> > > >(class b
oost::_bi::bind_t<void,class boost::_mfi::mf0<void,class pcl::TimeTrigger>,clas
s boost::_bi::list1<class boost::_bi::value<class pcl::TimeTrigger *> > > &&,st
ruct boost::thread::dummy *)" (??$?0V?$bind_t@XV?$mf0@XVTimeTrigger@pcl@@@_mfi@
boost@@V?$list1@V?$value@PAVTimeTrigger@pcl@@@_bi@boost@@@_bi@3@@_bi@boost@@@th
read@boost@@QAE@$$QAV?$bind_t@XV?$mf0@XVTimeTrigger@pcl@@@_mfi@boost@@V?$list1@
V?$value@PAVTimeTrigger@pcl@@@_bi@boost@@@_bi@3@@_bi@1@PAUdummy@01@@Z) [C:\User
s\takrit\Downloads\pcl-pcl-1.7.2\build\common\pcl_common.vcxproj]
  time_trigger.obj : error LNK2019: unresolved external symbol "public: __thisc
all boost::thread::thread(void)" (??0thread@boost@@QAE@XZ) referenced in functi
on "public: __thiscall pcl::TimeTrigger::TimeTrigger(double,class boost::functi
on<void __cdecl(void)> const &)" (??0TimeTrigger@pcl@@QAE@NABV?$function@$$A6AX
XZ@boost@@@Z) [C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\common\pcl_common.
vcxproj]
  time_trigger.obj : error LNK2019: unresolved external symbol "class boost::sy
stem::error_category const & __cdecl boost::system::generic_category(void)" (?g
eneric_category@system@boost@@YAABVerror_category@12@XZ) referenced in function
 "void __cdecl boost::system::`dynamic initializer for 'posix_category''(void)"
 (??__Eposix_category@system@boost@@YAXXZ) [C:\Users\takrit\Downloads\pcl-pcl-1
.7.2\build\common\pcl_common.vcxproj]
  C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\bin\pcl_common_release.dll : fa
tal error LNK1120: 7 unresolved externals [C:\Users\takrit\Downloads\pcl-pcl-1.
7.2\build\common\pcl_common.vcxproj]

"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\install.vcxproj" (default target
) (1) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\ALL_BUILD.vcxproj" (default targ
et) (3) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\tools\pcl_add_gaussian_noise.vcx
proj" (default target) (4) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\io\pcl_io.vcxproj" (default targ
et) (6) ->
"C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\io\pcl_io_ply.vcxproj" (default
target) (7) ->
  ply_parser.obj : error LNK2019: unresolved external symbol "class boost::syst
em::error_category const & __cdecl boost::system::system_category(void)" (?syst
em_category@system@boost@@YAABVerror_category@12@XZ) referenced in function "pu
blic: __thiscall boost::thread_exception::thread_exception(int,char const *)" (
??0thread_exception@boost@@QAE@HPBD@Z) [C:\Users\takrit\Downloads\pcl-pcl-1.7.2
\build\io\pcl_io_ply.vcxproj]
  ply_parser.obj : error LNK2019: unresolved external symbol "class boost::syst
em::error_category const & __cdecl boost::system::generic_category(void)" (?gen
eric_category@system@boost@@YAABVerror_category@12@XZ) referenced in function "
void __cdecl boost::system::`dynamic initializer for 'posix_category''(void)" (
??__Eposix_category@system@boost@@YAXXZ) [C:\Users\takrit\Downloads\pcl-pcl-1.7
.2\build\io\pcl_io_ply.vcxproj]
  C:\Users\takrit\Downloads\pcl-pcl-1.7.2\build\bin\pcl_io_ply_release.dll : fa
tal error LNK1120: 2 unresolved externals [C:\Users\takrit\Downloads\pcl-pcl-1.
7.2\build\io\pcl_io_ply.vcxproj]

matlabbe, 2017-06-29:
You can try latest code from PCL github, or look at the PCL forum for this kind of errors. It seems that some boost libraries are missing.

hildaaw, 2017-07-07:
Hi Mathieu
Is it possible to use the all in one pcl installer instead of building the pcl from source

ziqB, 2017-07-07:
Hi Mathieu
Is it possible to use the all in one PCL installer instead of building it from source, if yes do i just add the installed bin directory to the Path

matlabbe, 2017-07-08:
Yes, if you use Visual Studio 2010.

ziqB, 2017-07-11:
Thanks I've gotten PCL to work with no errors and proceeded with the RTAB source code.
Unluckily i've still got 2 errors in my RTAB compared to the previous 11 errors:

UProcessInfo.obj : error LNK2019: unresolved external symbol GetProcessMemoryIn
fo referenced in function "public: static long __cdecl UProcessInfo::getMemoryU
sage(void)" (?getMemoryUsage@UProcessInfo@@SAJXZ) [C:\Users\qlin\Desktop\RTABma
p compilation\rtabmap-0.12.5\build\utilite\src\rtabmap_utilite.vcxproj]
C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\bin\rtabmap_utilite.dl
l : fatal error LNK1120: 1 unresolved externals [C:\Users\qlin\Desktop\RTABmap
compilation\rtabmap-0.12.5\build\utilite\src\rtabmap_utilite.vcxproj]
Done Building Project "C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5
\build\utilite\src\rtabmap_utilite.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5
\build\utilite\resource_generator\res_tool.vcxproj" (default targets) -- FAILED
.

Done Building Project "C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5
\build\corelib\src\rtabmap_core.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5
\build\examples\BOWMapping\bow_mapping.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5
\build\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5
\build\install.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\install.vcxproj
" (default target) (1) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\ALL_BUILD.vcxpr
oj" (default target) (3) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\examples\BOWMap
ping\bow_mapping.vcxproj" (default target) (4) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\corelib\src\rta
bmap_core.vcxproj" (default target) (5) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\utilite\resourc
e_generator\res_tool.vcxproj" (default target) (6) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\utilite\src\rta
bmap_utilite.vcxproj" (default target) (7) ->
(Link target) ->
  UProcessInfo.obj : error LNK2019: unresolved external symbol GetProcessMemory
Info referenced in function "public: static long __cdecl UProcessInfo::getMemor
yUsage(void)" (?getMemoryUsage@UProcessInfo@@SAJXZ) [C:\Users\qlin\Desktop\RTAB
map compilation\rtabmap-0.12.5\build\utilite\src\rtabmap_utilite.vcxproj]
  C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\bin\rtabmap_utilite.
dll : fatal error LNK1120: 1 unresolved externals [C:\Users\qlin\Desktop\RTABma
p compilation\rtabmap-0.12.5\build\utilite\src\rtabmap_utilite.vcxproj]

    0 Warning(s)
    2 Error(s)
Could it be because I placed the source file seperate from the RTAB application folder? 
Thanks in advance

ziqB, 2017-07-11:
Hi 
could it be because i did not do the following :
cmake -G"NMake Makefiles" -DCMAKE_BUILD_TYPE=Release ..
$ nmake
$ nmake install

I'm a bit confuse with the (cmake 2) above the command above
I interpret it as a different version of cmake so i skipped the step for the PCL build.
Will that affect my build?

matlabbe, 2017-07-11:
`GetProcessMemoryInfo` is a function from the windows libraries. The cmake may not have correctly set all dependencies. Which Windows are you using, and which Visual Studio? Make sure you used cmake in a "**Visual Studio command prompt**", like stated [here](https://github.com/introlab/rtabmap/wiki/Installation#source-1).

I recommend to use CMake 3 directives, as you have the choice to build from terminal or from the generated Visual Studio project.

ziqB, 2017-07-13:
Hi
I'm using windows 7 and visual studio 2010. I used the cmake from the command prompt. How do i set the dependencies for cmake?

matlabbe, 2017-07-13:
cmake will find them automatically if it can find them. Did you install all libraries as 64 bits? For cmake, you can do this from a visual studio command (x64 if you build for 64 bits) and copy paste all the output (assuming you have cmake 3):
```
$ cd rtabmap/build
$ del CMakeCache.txt
$ cmake -G"Visual Studio 10 2010 Win64" ..
$ cmake --build . --config Release --target install
```

cheers

ziqB, 2017-07-14:
I checked and I did install all libraries as 64bits. I tried the visual studio command but it still gives the same 2 errors. However i tried using the cmake gui and opened the project, i went on to build it and had a different set of errors:

Error LNK1158	cannot run 'cvtres.exe'	rtabmap_utilite	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\utilite\src LINK file	line:1	

Error LNK1158	cannot run 'cvtres.exe'	res_tool	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\utilite\resource_generator LINK file	line:1	

Error LNK1104	cannot open file 'opencv_videostab.lib'	imagesJoiner	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\ImagesJoiner LINK file	line:1	

Error MSB6006	"cmd.exe" exited with code 9009.	rtabmap_core	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0 Microsoft.CppCommon.targets file	line:151	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	bow_mapping	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\examples\BOWMapping LINK file	line:1		

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	kitti_dataset	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\KittiDataset LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	rgbd_camera	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\CameraRGBD LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	camera	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\Camera LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	stereoEval	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\StereoEval LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	vocabularyComparison	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\VocabularyComparison LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	rtabmap_gui	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\guilib\src LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	consoleApp	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\ConsoleApp LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	odometryViewer	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\OdometryViewer LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	dataRecorder	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\DataRecorder LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	noEventsExample	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\examples\NoEventsExample LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	calibration	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\Calibration LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	epipolar_geometry	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\EpipolarGeometry LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	databaseViewer	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\DatabaseViewer LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	rgbd_mapping	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\examples\RGBDMapping LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	rtabmap	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\app\src LINK file	line:1	

Error LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	wifi_mapping	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\examples\WifiMapping LINK file	line:1	

Error LNK1158	cannot run 'cvtres.exe'	extractObject	C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.5\build\tools\ExtractObject LINK file	line:1

ziqB, 2017-07-18:
Hi 
I think i might know the problem. When using the cmake gui to build another version of rtabmap i came across a problem. It could not find the PSAPI and OpenNI. The rtabmap that i could open in visual studio was because i manually changed the directory. Therefore the source of my problem as matlabbe has said before was due to cmake not setting the dependencies correctly.
 
However i'm not sure how to troubleshoot the problem. I tried finding the files but was unable to find them. Thanks in advance : )

ziqB, 2017-07-20:
Hi I solved the previous problems by following your Mathieu's suggestion to delete the CMakeCache.txt but along the way i was left with one error:

Build FAILED.

"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.4\build\install.vcxproj
" (default target) (1) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.4\build\ALL_BUILD.vcxpr
oj" (default target) (3) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.4\build\examples\BOWMap
ping\bow_mapping.vcxproj" (default target) (4) ->
"C:\Users\qlin\Desktop\RTABmap compilation\rtabmap-0.12.4\build\corelib\src\rta
bmap_core.vcxproj" (default target) (5) ->
(ClCompile target) ->
  ..\..\..\corelib\src\CameraStereo.cpp(52): fatal error C1083: Cannot open inc
lude file: 'zed/Camera.hpp': No such file or directory [C:\Users\qlin\Desktop\R
TABmap compilation\rtabmap-0.12.4\build\corelib\src\rtabmap_core.vcxproj]

    0 Warning(s)
    1 Error(s)
Something to do with the zed/Camera.hpp
I tried reinstalling zed sdk but it did not seem to work in fact it caused way more errors by changing the opencv directory. I hope someone could help me.

ziqB, 2017-07-21:
I've solved the problem by using the cmake gui to untick WITH_ZED. Doing this i was able to succeed in the build. 
Thank you vey much

closed 2017-07-21 by ziqB