introlab/rtabmap #212: build

yyfBlog, 2017-06-28:
Under windows, how to compile the NativeRTABMap so file?

matlabbe, 2017-06-28:
The best option would be to use Docker on Windows, as we never built it on Windows.

yyfBlog, 2017-06-29:
How do I use the code on androidStudio?

matlabbe, 2017-06-29:
To use with Android Studio, you would need to cross compile everything (all third party libraries) on Windows. I would suggest to create a Virtual machine with Ubuntu 16.04 on Windows, so you can follow directly the installation scripts used for Docker (you may change all "`/root/`" by "`~/`" for convenience) as some dependencies require custom patches: https://github.com/introlab/rtabmap/tree/master/docker/xenial/android

cheers,
Mathieu

yyfBlog, 2017-06-30:
Can you submit a tutorial for me?    
thank you very much

matlabbe, 2017-07-01:
I don't develop the Tango app on Windows or with Android Studio, so I changed the title of the issue for an enhancement.

yyfBlog, 2017-07-03:
[image]

yyfBlog, 2017-07-03:
why?

yyfBlog, 2017-07-03:
or How do I get the file?（NativeRTABMap.so）

matlabbe, 2017-07-03:
Follow instructions from https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-tango-apk

heartzwh, 2018-04-24:
@yyfBlog do u find the way develop with android studio? or on windows? thx