introlab/rtabmap #213: RTABMAP's Estimated trajectory is noisy

aseyfi, 2017-06-28:
Hello,

We have been using rtapmap quite a bit to process data (odometry data, depth data and rgb image data) collected using the TangoRosStreamer (with some modifications to get a higher RGB frame rate) for mapping and localization. We are using rtabmap and rtabmap_ros Tag 0.12.2. In our experiments with localization, rtabmap is able to detect the loop closures and produce map corrections to transfer the odometry pose to the map frame. However, we often find that the map corrections are noisy and lead to pose trajectories with spikes even when the ground truth trajectories are smooth. We wonder if you have observed the similar issue in your experiments. It will be a great help to us if you can share with us some sample data you have recorded using Tango devices with good mapping and localization results as well as  the corresponding rtabmap settings used to process such data. Such data and settings will be a good reference to us. 

Thank you so much for the help!
Ahmad

matlabbe, 2017-06-29:
Hi,

There are indeed a "jumps" when localizations occur. See [REP 105](http://www.ros.org/reps/rep-0105.html#map) for reference, under `map` frame description. 

In rtabmap, we have two choices of handling localization effects:
 * `RGBD/OptimizeFromGraphEnd=false` (default): TF `/map -> /odom` updated to match current odometry pose in the map. The effect is "jumping" in /map frame as described in REP 105. This is also the standard configuration for navigation stack.
 * `RGBD/OptimizeFromGraphEnd=true`: Instead of updating only `/map -> /odom` to match odometry pose to map, it is instead the whole map that is transformed to match the current odometry pose (`/map -> /odom` is always identity in this mode). The advantage is that there are no "jumps", but the poses have no meaning against the initial map. For example, we cannot plan to a saved position in the initial map, as the map will move over time around the current odometry pose. Hopefully, rtabmap planning capability can handle this situation, and re-correct the goal while the map moves around the current pose. 

More info:
```
$ rtabmap --params | grep FromGr
Param: RGBD/OptimizeFromGraphEnd = "false"                 [Optimize graph from the newest node.
 If false, the graph is optimized from the oldest node of the current graph (this adds an overhead 
computation to detect to oldest node of the current graph, but it can be useful to preserve the 
map referential from the oldest node). Warning when set to false: when some nodes are 
transferred, the first referential of the local map may change, resulting in momentary changes in 
robot/map position (which are annoying in teleoperation).]
```

Try both modes to see differences on what your are observing in localization mode.

cheers,
Mathieu

aseyfi, 2017-06-30:
@matlabbe Thank you for the prompt reply. We have been using the first mode (RGBD/OptimizeFromGraphEnd=false) in our localization experiment. Would you please let us know the normal range of localization errors you have seen in your experiments using data from Tango devices?

Thanks,
Ahmad

matlabbe, 2017-07-03:
I didn't do a lot of mapping experiments with Tango ROS Streamer + rtabmap_ros. Do you have a small video capture of what you are seeing? or how do you compute the noise so I can try to reproduce a similar experiment? I could tell if it looks normal or not.

cheers,
Mathieu

aseyfi, 2017-07-14:
Hi @matlabbe , thanks again for being so helpful. I have shared two bag files that I have used for mapping and re-localization [here](https://www.dropbox.com/sh/3k4491hseuearcg/AABzl9hpdr0YQ8KY7oK_Ky-Ya?dl=0). Also, there is a word doc that explains the topic names and the parameters I have used and a screenshot of the re-localization results. Please let me know if you need further information.

Thanks,
Ahmad

matlabbe, 2017-07-26:
Hi Ahmad,

Thanks for the ros bags. Commands I used (creating depth image like in this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer)):

```bash
$ roscore
$ rosparam set use_sim_time true
$ rosrun rtabmap_ros pointcloud_to_depthimage cloud:=/tango/point_cloud camera_info:=/tango/camera/rgb/camera_info _fixed_frame_id:=start_of_service _decimation:=8 _fill_holes_size:=5
```

* Mapping
```bash
$ roslaunch rtabmap_ros rtabmap.launch \
visual_odometry:=false \
odom_frame_id:="start_of_service" \
frame_id:="device" \
rgb_topic:=/tango/camera/rgb/image_raw \
depth_topic:=/image \
camera_info_topic:=/tango/camera/rgb/camera_info \
compressed:=true \
depth_image_transport:=raw \
approx_sync:=false \
rtabmap_args:="\
--delete_db_on_start \
--Rtabmap/DetectionRate 0 \
--Mem/ImagePreDecimation 2 \
--Mem/ImagePostDecimation 2 \
--Kp/DetectorStrategy 6 \
--Vis/EstimationType 1"
```

* Localization
```bash
$ roslaunch rtabmap_ros rtabmap.launch \
localization:=true \
visual_odometry:=false \
odom_frame_id:="start_of_service" \
frame_id:="device" \
rgb_topic:=/tango/camera/rgb/image_raw \
depth_topic:=/image \
camera_info_topic:=/tango/camera/rgb/camera_info \
compressed:=true \
depth_image_transport:=raw \
approx_sync:=false \
rtabmap_args:="\
--Rtabmap/DetectionRate 0 \
--Mem/ImagePreDecimation 2 \
--Mem/ImagePostDecimation 2 \
--Kp/DetectorStrategy 6 \
--Vis/EstimationType 1"
```

* My trajectory:
[image]

Grid has cell size of 10 cm. I put here yours for comparison:
[image]

* Observations
RTAB-Map localizes by matching the best image in the map against the current one. If images in the map are taken far apart, localization with point of views between two nodes will be less accurate. In my example I added as most nodes as I can (`Rtabmap/DetectionRate=0`) during mapping. So in localization, the best image matched would be more close to current point of view, so less error. In the same way, we can see at the end of the localization bag that the camera is off the map, so the localization is a lot less precise. Also (refer to screenshot below), it seems that Tango drifted during the record of the Mapping.bag because the wall is not straight and correctly aligned between scans. We can also see that there are more than 10 cm depth error on the wall (wall appears up to 30 cm of width at some places!). In this case, as features have more than 10 cm depth error, the localization will varies up to 10 cm too or more (what we see in your screenshot).

[image]

So in general, the noise of the localization will be dependent on the depth noise at distance where the features are taken and if the best matching image of the map is close to current real position of the camera.

cheers,
Mathieu