introlab/rtabmap #214: Multicamera calibration for RTAB

hildaaw, 2017-06-29:
Hi, I'd like to combine multiple Xtion PRO Live cameras into a single output. So it's like 3 camera, which are placed side by side to scan a wider area at one go. 
May I know what ways are there to support multicamera calibration for RTAB? 

Thank you.

matlabbe, 2017-06-29:
Yes it is, though not with rtabmap standalone. I recommend to use [rtabmap_ros](http://wiki.ros.org/rtabmap_ros), which supports it. If you are not familiar with ROS, please see http://wiki.ros.org/ROS/Tutorials. There is an example here using two kinects at 90 degrees: https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch

cheers,
Mathieu

hildaaw, 2017-06-30:
Thank you for your recommendation, I will work on it.
May i also asked that do you have other recommendations for window 7 as well? 

Thank you once again.

matlabbe, 2017-07-01:
For multi-camera on windows7? no. It could be possible to do by hard coding some stuff, but it would be re-doing what ROS already does (flexibilty of the sensors, synchronization, transformations between the sensors...).

cheers,
Mathieu

closed 2017-07-14 by hildaaw