introlab/rtabmap #217: License

arogowiec, 2017-07-05:
Hi! 
I have some concerns about license, since I see that you have recently included odometry from ORB_SLAM2 which is licensed under GPLv3. As far as I know, any use of GPL licensed software forces such project to be licensed also under GPL. However I see that RtabMap is under BSD license, which is kind for commercial use, as opposed to GPL. 

Since I'm interested in use of part of RtabMap software in a commercial project, can you tell me if it is still licensed under BSD (i.e. don't force source disclosure)?

arogowiec, 2017-07-05:
Ok, just found message in CMakeLists.txt that when compiling with ORB_SLAM2 then RtabMap is under GPLv3.

closed 2017-07-05 by arogowiec

matlabbe, 2017-07-05:
Yes, cmake will tell you which licenses are used depending of the dependencies detected and used. Doing "`cmake -DWITH_ORBSLAM2=OFF ..`" disables ORB_SLAM2 support (you can also uncheck it with cmake-gui). Note that we can release commercial application with GPLv3 as long as the code is using also GPLv3 (code should be open source). In recent releases, for convenience, the licenses are also stated in About of RTAB-Map:
[image]

It is a similar problem to OpenCV, which is BSD too, unless you build it with SIFT/SURF support.

cheers,
Mathieu