introlab/rtabmap #218: RTABMAP-0.13.0 Unresolved external symbol errors

ahsteven, 2017-07-08:
Hello Mathieu,
I had built an ran the RGBD-Mapping example with RTAB-MAP 12.4. However, I needed to update to RTABMAP with Zed 20 and Cuda 8 support so I installed RTAB-MAP-0.13.0-win64-ZED20-CUDA80.exe. I was able to run an example with the GUI. However after regeneratingthe RGBD-Mapping example VS project with cmake  I am getting some linking errors when I build the project. Below are the errors:

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: __cdecl rtabmap::CloudViewer::CloudViewer(class QWidget *)" (__imp_??0CloudViewer@rtabmap@@QEAA@PEAVQWidget@@@Z) referenced in function "public: __cdecl MapBuilder::MapBuilder(class rtabmap::CameraThread *)" (??0MapBuilder@@QEAA@PEAVCameraThread@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: virtual __cdecl rtabmap::CloudViewer::~CloudViewer(void)" (__imp_??1CloudViewer@rtabmap@@UEAA@XZ) referenced in function "public: virtual void * __cdecl rtabmap::CloudViewer::`scalar deleting destructor'(unsigned int)" (??_GCloudViewer@rtabmap@@UEAAPEAXI@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: bool __cdecl rtabmap::CloudViewer::updateCloudPose(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class rtabmap::Transform const &)" (__imp_?updateCloudPose@CloudViewer@rtabmap@@QEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBVTransform@2@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processStatistics(class rtabmap::Statistics const &)" (?processStatistics@MapBuilder@@MEAAXAEBVStatistics@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: bool __cdecl rtabmap::CloudViewer::addCloud(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::shared_ptr<class pcl::PointCloud<struct pcl::PointXYZRGB> > const &,class rtabmap::Transform const &,class QColor const &)" (__imp_?addCloud@CloudViewer@rtabmap@@QEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$shared_ptr@V?$PointCloud@UPointXYZRGB@pcl@@@pcl@@@boost@@AEBVTransform@2@AEBVQColor@@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processOdometry(class rtabmap::OdometryEvent const &)" (?processOdometry@MapBuilder@@MEAAXAEBVOdometryEvent@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: bool __cdecl rtabmap::CloudViewer::addCloud(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::shared_ptr<class pcl::PointCloud<struct pcl::PointXYZ> > const &,class rtabmap::Transform const &,class QColor const &)" (__imp_?addCloud@CloudViewer@rtabmap@@QEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$shared_ptr@V?$PointCloud@UPointXYZ@pcl@@@pcl@@@boost@@AEBVTransform@2@AEBVQColor@@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processStatistics(class rtabmap::Statistics const &)" (?processStatistics@MapBuilder@@MEAAXAEBVStatistics@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl rtabmap::CloudViewer::updateCameraTargetPosition(class rtabmap::Transform const &)" (__imp_?updateCameraTargetPosition@CloudViewer@rtabmap@@QEAAXAEBVTransform@2@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processOdometry(class rtabmap::OdometryEvent const &)" (?processOdometry@MapBuilder@@MEAAXAEBVOdometryEvent@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl rtabmap::CloudViewer::addOrUpdateGraph(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::shared_ptr<class pcl::PointCloud<struct pcl::PointXYZ> > const &,class QColor const &)" (__imp_?addOrUpdateGraph@CloudViewer@rtabmap@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV?$shared_ptr@V?$PointCloud@UPointXYZ@pcl@@@pcl@@@boost@@AEBVQColor@@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processStatistics(class rtabmap::Statistics const &)" (?processStatistics@MapBuilder@@MEAAXAEBVStatistics@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl rtabmap::CloudViewer::removeAllGraphs(void)" (__imp_?removeAllGraphs@CloudViewer@rtabmap@@QEAAXXZ) referenced in function "protected: virtual void __cdecl MapBuilder::processStatistics(class rtabmap::Statistics const &)" (?processStatistics@MapBuilder@@MEAAXAEBVStatistics@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: bool __cdecl rtabmap::CloudViewer::removeCloud(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (__imp_?removeCloud@CloudViewer@rtabmap@@QEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processStatistics(class rtabmap::Statistics const &)" (?processStatistics@MapBuilder@@MEAAXAEBVStatistics@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: bool __cdecl rtabmap::CloudViewer::getPose(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class rtabmap::Transform &)" (__imp_?getPose@CloudViewer@rtabmap@@QEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEAVTransform@2@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processStatistics(class rtabmap::Statistics const &)" (?processStatistics@MapBuilder@@MEAAXAEBVStatistics@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: class QColor const & __cdecl rtabmap::CloudViewer::getDefaultBackgroundColor(void)const " (__imp_?getDefaultBackgroundColor@CloudViewer@rtabmap@@QEBAAEBVQColor@@XZ) referenced in function "protected: virtual void __cdecl MapBuilder::processOdometry(class rtabmap::OdometryEvent const &)" (?processOdometry@MapBuilder@@MEAAXAEBVOdometryEvent@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl rtabmap::CloudViewer::setBackgroundColor(class QColor const &)" (__imp_?setBackgroundColor@CloudViewer@rtabmap@@QEAAXAEBVQColor@@@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processOdometry(class rtabmap::OdometryEvent const &)" (?processOdometry@MapBuilder@@MEAAXAEBVOdometryEvent@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: void __cdecl rtabmap::CloudViewer::setCloudVisibility(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,bool)" (__imp_?setCloudVisibility@CloudViewer@rtabmap@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_N@Z) referenced in function "protected: virtual void __cdecl MapBuilder::processOdometry(class rtabmap::OdometryEvent const &)" (?processOdometry@MapBuilder@@MEAAXAEBVOdometryEvent@rtabmap@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBD [... cut ...] ts\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK2001	unresolved external symbol "protected: virtual void __cdecl rtabmap::CloudViewer::wheelEvent(class QWheelEvent *)" (?wheelEvent@CloudViewer@rtabmap@@MEAAXPEAVQWheelEvent@@@Z)	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\main.obj	1	
Error	LNK1120	25 unresolved externals	rgbd_mapping	C:\Users\Adam\Documents\Visual Studio 2015\Projects\RTABMAP_RGBDMapping\build9_13\Release\rgbd_mapping.exe	1

matlabbe, 2017-07-10:
This is a bug with the binary release, it has been fixed in this [commit](https://github.com/introlab/rtabmap/commit/5646751f885a716820eb1a96f7e8651b1fa5874b). Change installed RTABMapConfig.cmake file like the changes in the commit, or explicitly set this in your code:
```cmake
# Instead of 
find_package(RTABMap 0.13 REQUIRED)
# set 
find_package(RTABMap 0.13 REQUIRED core gui)
```

cheers,
Mathieu

ahsteven, 2017-07-11:
Yes Mathieu, following your suggestions I was able to solve this issue. Thank you

closed 2017-07-11 by ahsteven