introlab/rtabmap #221: USB issues with Kinect v2

EmilyCatJack, 2017-07-20:
So I'm trying to use a Kinect v2 with RTAB.

On the Kinect configuration verifier, it sees the Kinect, I get picture and depth, but I get a yellow usb warning. When I run RTAB,  I get a camera initialization failed error.

Any thoughts???

matlabbe, 2017-07-26:
You may set RTAB-Map in Debug mode: Preferences->Logging->Logger Level. Try to start the camera then copy/paste the log of the terminal here. Are you using the Windows binaries or are you on Ubuntu?

eric-schleicher, 2017-08-06:
it sounds like you're using windows (the kinect configuration verifier).   the yellow USB warning is that your USB chipset *may* not fully support the requirements based on who made the chipset (intel, nec, etc)  this is just a warning.  most proper usb3 host adapters *do* work.

if you are in fact on windows, can you confirm that you've install usbdk?

eric-schleicher, 2018-06-23:
what version of usbdk are you using?
you need to use this version:
https://github.com/daynix/UsbDk/releases/tag/v1.00-12
see 
https://github.com/OpenKinect/libfreenect2#windows--visual-studio

does that help?

GeorgeTower, 2018-06-25:
I used version v1.00-19! which can get linked with the kinect v2, can not get the pic and depth.  when i use the version v1.00-12, i can not initialize the device, and the demo may dying!  trying many times! my desktop computer with win7 sys is also the same appearance

eric-schleicher, 2018-06-25:
I can't help you much beyond that...  if the configuration verifier can't get a good state. it's possible it either wont work (in windows, or possibly at all)

is it possible to run linux?  
I've seen systems that refused to work with kinect2 on windows work just fine on ubuntu (16.04 LTS).

matlabbe, 2018-07-02:
Are you using the binaries or are you rebuilding rtabmap from source? Is Protonect.exe of Freenect2 project working? If you build from source and if Microsoft Kinect for Windows SDK 2 is installed (to use kinect v2), the corresponding driver in rtabmap will be available instead of Freenect2.

cheers,
Mathieu