introlab/rtabmap #223: Building RTABMAP on Windows

ahsteven, 2017-07-22:
Hello, I have been trying to build RTABMAP on Windows and am having some linking issues. 
I open a Developer Command Prompt for VS2015 in administrator mode and run the following commands:

`cmake -G"Visual Studio 14 2015 Win64" ..
`
This command is able to configure and generate the files fine. I have pasted the cmake output I recieve when running this command.

The next command is where I have trouble:

`cmake --build . --config Release --target install`

Here I get the linking errors:
"C:\Users\Adam\rtabmap\build\install.vcxproj" (default target) (1) ->
"C:\Users\Adam\rtabmap\build\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Users\Adam\rtabmap\build\examples\BOWMapping\bow_mapping.vcxproj" (default target) (4) ->
"C:\Users\Adam\rtabmap\build\corelib\src\rtabmap_core.vcxproj" (default target) (5) ->
(Link target) ->
  LINK : fatal error LNK1181: cannot open input file 'vtkGUISupportQt.lib' [C:\Users\Adam\rtabmap\build\corelib\src\rta
bmap_core.vcxproj]

"C:\Users\Adam\rtabmap\build\install.vcxproj" (default target) (1) ->
"C:\Users\Adam\rtabmap\build\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Users\Adam\rtabmap\build\tools\ExtractObject\extractObject.vcxproj" (default target) (15) ->
  LINK : fatal error LNK1181: cannot open input file 'vtkGUISupportQt.lib' [C:\Users\Adam\rtabmap\build\tools\ExtractOb
ject\extractObject.vcxproj]

    281 Warning(s)
    2 Error(s)

I have built vtk with gui support and have a library vtkGUISupportQt-7.0.lib in the following directory:

C:\Users\Adam\Documents\VTK-7.0.0\Build\lib\Release

I am not sure how to get cmake to find this library properly.

Also if I open the RTABMAP.sln file and build in visual studio 2015 itself I get the following list of errors:

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	dataRecorder	C:\Users\Adam\rtabmap\build\tools\DataRecorder\LINK	1	
Error	LNK1104	cannot open file 'vtkGUISupportQt.lib'	extractObject	C:\Users\Adam\rtabmap\build\tools\ExtractObject\LINK	1	
Error	LNK1104	cannot open file 'vtkGUISupportQt.lib'	rtabmap_core	C:\Users\Adam\rtabmap\build\corelib\src\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	stereoEval	C:\Users\Adam\rtabmap\build\tools\StereoEval\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	bow_mapping	C:\Users\Adam\rtabmap\build\examples\BOWMapping\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	consoleApp	C:\Users\Adam\rtabmap\build\tools\ConsoleApp\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	rgbd_dataset	C:\Users\Adam\rtabmap\build\tools\RgbdDataset\LINK	1	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	kitti_dataset	C:\Users\Adam\rtabmap\build\tools\KittiDataset\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	camera	C:\Users\Adam\rtabmap\build\tools\Camera\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_cored.lib'	rgbd_camera	C:\Users\Adam\rtabmap\build\tools\CameraRGBD\LINK	1	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rtabmap_gui	c:\users\adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	noEventsExample	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	calibration	C:\Users\Adam\rtabmap\build\tools\Calibration\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	databaseViewer	C:\Users\Adam\rtabmap\build\tools\DatabaseViewer\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	epipolar_geometry	C:\Users\Adam\rtabmap\build\tools\EpipolarGeometry\LINK	1	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	noEventsExample	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	odometryViewer	C:\Users\Adam\rtabmap\build\tools\OdometryViewer\LINK	1	
Error	LNK1104	cannot open file '..\..\..\lib\rtabmap_guid.lib'	rtabmap	C:\Users\Adam\rtabmap\build\app\src\LINK	1	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rgbd_mapping	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	rgbd_mapping	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	wifi_mapping	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	wifi_mapping	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	
Error	C1083	Cannot open include file: 'QVTKWidget.h': No such file or directory	wifi_mapping	C:\Users\Adam\rtabmap\guilib\include\rtabmap\gui\CloudViewer.h	36	

Can anyone help me understand what I am doing wrong and the best way to fix it?

ahsteven, 2017-07-26:
I was able to get the build by moving the files manually and changing the requested library name from vtkGUISupportQt.lib to vtkGUISupportQt-7.0.lib. If there is an easy fix that would allow this to build automatically please let me know, otherwise we can close the case. Thank you

matlabbe, 2017-07-28:
vtkGUISupportQt.lib would be a dependency set from `VTKConfig.cmake` file. I didn't try VTK 7 yet on Windows, so I will keep this issue opened.