introlab/rtabmap #225: Feature request: ability to filter bundle exports based on rotational and translational speed threshold(s)

eric-schleicher, 2017-08-06:
Just putting this on the list. 

Expect that this would improve the texturing workflow /result, specifically with tango (but possibly others) 

When makings larger scans, invariably there are images that the auto-exposure cannot capture clearly (motion blurred images).  This is particularly noticeable on the phab 2 pro where the camera is of criminally substandard quality.  

It's common that  session with 700-900 images get created, a significant number of them are of poor enough composition that they should not be used.  generally i find this is because the pose estimates and the depth capture can support faster movement.   

I propose that rather than coming up with a fancy algorithm for determining whether the composition of the image is bad, that a simpler scheme of classifying/using the linear/rotation speed to disqualify frames could be used to limit the number of blurred images that would be candidates for texturization.

Perhaps a speed threshold cutoff, or some scalar value.   Potentially this value could be used to filter down the number of images exported (for example in the .bundle export)  this certainly would simplify manual review work done in meshlab to texture meshes, and might also be of some benefit to the inbuilt texturization scheme.

Interested if other have contemplated other approaches.

matlabbe, 2017-08-07:
Hi Eric,

Thx for the detailed comment. I did a fast implementation when exporting clouds with texture or exporting in bundler format:
[image]

[image]

In the commit I also fixed a bug for which velocity was not saved in the database. I'll try to publish the Tango app as soon as docker has finished the image so that velocity filtering can be used with tango databases. The [Laplacian filtering approach](http://www.pyimagesearch.com/2015/09/07/blur-detection-with-opencv/) can work with any old database versions.

Here some results where I did a fast motion followed by a slow one for the same area: 
 1. All cameras are used (the red squares show some sections with blurred texture), 
 2. Filtering fast motions (0.1 m/s and 0.4 rad/s), 
 3. Laplacian threshold at 75

[image]

I think this commit can close this issue. However, the approach above will make some parts of the map disappearing (see bottom of second image) if they were only textured by blurred images. The best would be that at the texturing step, we use the camera with less motion blur, to keep all the map (e.g., blurred images could still texture some polygons if they are the only ones looking at them).

cheers,
Mathieu

eric-schleicher, 2017-08-30:
That's an amazing "fast" implementation.  i can't wait to play with it.

closed 2017-08-30 by eric-schleicher