introlab/rtabmap #226: Add GPS field to database

matlabbe, 2017-08-22:
Can be useful for outdoor robots and Tango phone for comparison purpose, limit loop closure detection area or to use in map optimization (#188).

Format (inspired from [Location](https://developer.android.com/reference/android/location/Location.html) android page): 
 * Minimum: latitude, longitude, and timestamp
 * Optional: bearing, altitude and velocity

`std::vector` of 6 doubles?

murmurSK, 2017-09-20:
Hi Mathieu,
if this isn't dead after all the situation around Tango, it would be nice if the time would be visible e.g. in Debug view. This will be good to identify the poses when the device was on particular point.

matlabbe, 2017-09-21:
Thx for the update, I'm supposed to check tomorrow how to add GPS to database (I was on a major update the past weeks). 

For the time, you can tap on the screen to show up the status bar. The idea to keep it **always** visible in Debug view is a good idea, I may let the whole status bar shown in Debug view (so we can also see battery level). Note also that when opening the database in database viewer afterward, we can see the timestamp for each pose.

cheers,
Mathieu

murmurSK, 2017-09-21:
Thanks for reply. I didn't realize that the absolute timestamps can be viewed in DB viewer. That's half of my problem solved :) However, it would help greatly if these could be exported with poses instead of relative timestamps.
The second half is the ability to note the time in field (during scanning). E.g. "now, at 15:25:15, the device is at Point 1 (which coordinates are known)". Of course, I can check the time using e.g. watches, but there could be a sync problem. The ability to note the time, which is subsequently included in database, will help for georeferencing or even accuracy assessment. According to the noted time I could identify the particular pose and relate it to reference coordinates. Therefore, time in the Status bar is not sufficient, seconds are needed for this (with acceptable level of uncertainty for me). So, this is why it would help me greatly if the time information would be shown somehow :)
Best regards,
Julian

matlabbe, 2017-09-22:
Another precision about the timestamp saved in the databases, they are real date/time stamps in EPOCH format. They can be easily converted to human-readable format using online tool like this [one](https://www.epochconverter.com/). For convenience, I added a tooltip in database viewer so we can see the real date/time like in the screenshot below (we don't see the mouse pointer but it is over the stamp). 

[image]

The commit above implements the GPS support in Tango. With "RTAB-Map Tango -> Settings-> Mapping -> Save GPS in database", GPS values are saved while scanning as shown in the screenshot above (last line). The time (up to milliseconds) and GPS are also shown in Debug view.
[image]

I just tested with Phab2Pro yet, I'll try with Zenfone and yellowstone tomorrow before doing an official release. Being able to export GPS coordinates from database viewer like the pose may be nice to have too.

murmurSK, 2017-09-22:
You are a magician :) Looking forward for the update.

matlabbe, 2017-09-26:
After one day trying to integrate the GPS into map optimizations (g2o, gtsam), I abandoned the project as it is less trivial than I thought (may try again in the future, other stuff to do right now). I updated the DatabaseViewer to export GPS values in raw format and in KML format that can be easily opened in Google Earth for fast comparison (we can export GPS values but also the optimized poses transformed into GPS values). Here are some screenshots:
  
ENU orientation option on right-click to get North-East orientation. GPS values are in green (color can be changed in the right-clock menu too). Align to GPS option should be also checked under the graph view to align poses in GPS referential.
[image]

[image]

[image]

As the minor version changed, I need to release new RTAB-Map desktop binaries to be compatible with the new RTAB-Map Tango version. When everything will be released, I'll close this issue. 

cheers

murmurSK, 2017-09-26:
Oh, I didn't expect that you'll go so far.
From my point of view, there are some complications with GPS/Tango integration (I mean internal smartphone GPS):
- first fix - is not very reliable and is refined for a while. Therefore, when measuring, I use to force GPS to be active long enough before I actually get to my object of interest. I don't know how it works in your app yet; however, if the GPS is active immediately when the switch is clicked, this is solved by resetting the scanning after a reliable period of time (or after checking stability of the position using e.g. GPS Status app)
- delay - especially under not ideal conditions (forest etc.) it can take quite a long time until the device "realizes" that the position has changed. In my point of view, this can influence synchronization quite significantly (i.e. Tango could be already far away from a point where GPS momentally arrived)
- accuracy - the best accuracy that we have achieved with smartphone internal GNSS was between two and three meters under ideal conditions. I don't know if this would be sufficient for some kind of integration. But this is perhaps something that all GNSS/IMU systems try to solve - IMU measurement is stable, but has drift and GNSS is unstable, but without drift.
Enough theory :), just two practical questions:
1. Would there be an option to export GNSS coordinates (and perhaps time) together with poses? Like just adding more columns in existing pose-export format?
2. Any idea when the Tango version could be online?

Thanks for your efforts.
Julian

matlabbe, 2017-09-27:
* first fix: The GPS is intialialized (when "Save GPS" option is checked) along with tango. You can do reset after a couple of seconds if you believe that the GPS values are not accurate enough.
* delay: In my tests, the GPS is updated at least 1 time / sec, then the most recent value is saved in the node. 

Not sure what contains GNSS coordinates, but the raw export format for GPS in DatabaseViewer is all data that android can give me: "stamp longitude latitude altitude error bearing". The bearing is actually the compass in the phone.

RTAB-Map Tango has been just rolled out to production. It should appear as an update in some time on play store.

cheers,
Mathieu

closed 2017-09-27 by matlabbe

fangxu622, 2019-01-02:
@matlabbe hi~~now i'm  get the pose data(rgbd-slam format) from tango(save as *.db) , like this 
[image]

[image]

,but now~~how can i get the real world scale from pose data? 
In other words，How many distances（unit:cm/mm/m）between these two points represent?

matlabbe, 2019-01-03:
Please in the future open a new issue as this question is not related to this issue. The RGBD-SLAM format is STAMP, X, Y, Z, QX, QY, QZ, QW where X, Y, Z are in meters. The Q# parameters are the quaternion's parameters.