introlab/rtabmap #235: ZED sdk is not available

smallfly, 2017-09-19:
Hello,

I'm trying to use the ZED camera with RTAB-Map. The ZED sdk is  installed on my machine but when I run  RTAB-Map the ZED sdk in sources is disabled. I looked in the 'preferences' but I do not see anything about that. What am I missing?

Thanks!

matlabbe, 2017-09-19:
What is your cmake status?
```bash
$ cd rtabmap/build
$ cmake ..
```

smallfly, 2017-09-19:
I'm using the latest windows 64 binary release - v0.13.0
I did not compile the application myself.

matlabbe, 2017-09-19:
Make sure you download the Windows binaries with CUDA/ZED support: https://github.com/introlab/rtabmap/releases

I tried with ZED sdk 2.1 installed on Windows 10 and it works.

smallfly, 2017-09-20:
So sorry. I did not see this version (I don't know how). Downloading it right now. Thanks a lot!

closed 2017-09-20 by smallfly