introlab/rtabmap #239: ZED SDK 2.1 compile error

bmterry, 2017-10-11:
I cannot compile rtabmap with ZED SDK 2.1. 

I recieve the error

[image]

So far as I know, the ZED 2.1 SDK does not include libcudpp anymore

(Compiling on an Nvidia Jetson TX1)

matlabbe, 2017-10-11:
Did you update ZED SDK after building rtabmap, then try to rebuild rtabmap? It seems that some CMake variables are still set to old values. Try cleanup the build directory of rtabmap and rebuild:
```bash
$ rm -r rtabmap/build/*
$ cd rtabmap/build
$ cmake ..
$ make
```

bmterry, 2017-10-11:
I have updated the ZED SDK, but I deleted the whole rtabmap folder before starting again.

I tried your steps and got the same result

matlabbe, 2017-10-11:
Can you show output of
```bash
$ ldd /usr/local/zed/lib/libsl_zed.so
```
and also
```bash
$ cat /usr/local/zed/zed-config.cmake
```
?

cheers,
Mathieu

bmterry, 2017-10-11:
**ldd /usr/local/zed/lib/libsl_zed.so**

	linux-vdso.so.1 =>  (0x0000007f95f48000)
	libcudart.so.8.0 => /usr/local/cuda/lib64/libcudart.so.8.0 (0x0000007f95b2c000)
	libsl_calibration.so => /usr/local/zed/lib/libsl_calibration.so (0x0000007f95a5d000)
	libsl_depthcore.so => /usr/local/zed/lib/libsl_depthcore.so (0x0000007f95376000)
	libsl_tracking.so => /usr/local/zed/lib/libsl_tracking.so (0x0000007f95125000)
	libsl_scanning.so => /usr/local/zed/lib/libsl_scanning.so (0x0000007f94f45000)
	libsl_svorw.so => /usr/local/zed/lib/libsl_svorw.so (0x0000007f94f21000)
	libsl_disparityFusion.so => /usr/local/zed/lib/libsl_disparityFusion.so (0x0000007f94df5000)
	libsl_core.so => /usr/local/zed/lib/libsl_core.so (0x0000007f94777000)
	libcuda.so.1 => /usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1 (0x0000007f938a7000)
	libnppc.so.8.0 => /usr/local/cuda/lib64/libnppc.so.8.0 (0x0000007f93830000)
	libnppi.so.8.0 => /usr/local/cuda/lib64/libnppi.so.8.0 (0x0000007f8ee7a000)
	libnpps.so.8.0 => /usr/local/cuda/lib64/libnpps.so.8.0 (0x0000007f8e882000)
	libusb-0.1.so.4 => /lib/aarch64-linux-gnu/libusb-0.1.so.4 (0x0000007f8e86a000)
	libX11.so.6 => /usr/lib/aarch64-linux-gnu/libX11.so.6 (0x0000007f8e741000)
	libstdc++.so.6 => /usr/lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f8e5b2000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f8e505000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f8e4e3000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f8e39c000)
	/lib/ld-linux-aarch64.so.1 (0x0000005562d93000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f8e389000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f8e35c000)
	librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000007f8e345000)
	libgomp.so.1 => /usr/lib/aarch64-linux-gnu/libgomp.so.1 (0x0000007f8e316000)
	libpng12.so.0 => /lib/aarch64-linux-gnu/libpng12.so.0 (0x0000007f8e2e7000)
	libtbb.so => /usr/local/zed/lib/libtbb.so (0x0000007f8e2a6000)
	libturbojpeg.so.0 => /usr/lib/aarch64-linux-gnu/libturbojpeg.so.0 (0x0000007f8e259000)
	libzstd.1.1.1024.so => /usr/local/lib/libzstd.1.1.1024.so (0x0000007f8e1fc000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f8e1d5000)
	libnvrm_gpu.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_gpu.so (0x0000007f8e1a9000)
	libnvrm.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm.so (0x0000007f8e172000)
	libxcb.so.1 => /usr/lib/aarch64-linux-gnu/libxcb.so.1 (0x0000007f8e146000)
	libnvos.so => /usr/lib/aarch64-linux-gnu/tegra/libnvos.so (0x0000007f8e129000)
	libXau.so.6 => /usr/lib/aarch64-linux-gnu/libXau.so.6 (0x0000007f8e115000)
	libXdmcp.so.6 => /usr/lib/aarch64-linux-gnu/libXdmcp.so.6 (0x0000007f8e100000)

**cat /usr/local/zed/zed-config.cmake**

===================================================================================
  The ZED CMake configuration file

  Usage from an external project:
    In your CMakeLists.txt, add these lines:

    SET(ZED_DIR "/usr/local/zed/")
    FIND_PACKAGE(zed REQUIRED )
    INCLUDE_DIRECTORIES(${ZED_INCLUDE_DIRS})
    link_directories(${ZED_LIBRARY_DIR})
    TARGET_LINK_LIBRARIES(MY_TARGET_NAME ${ZED_LIBRARIES})

    This file will define the following variables:
      - ZED_LIBRARIES      : The list of libraries to links against.
      - ZED_LIBRARY_DIR    : The directory where lib files are.
      - ZED_INCLUDE_DIRS   : The ZED include directories.

 ===================================================================================

 
set(ZED_INCLUDE_DIRS "/usr/local/zed/include")
set(ZED_LIBRARY_DIR "/usr/local/zed/lib")
 
 
set(ZED_LIBRARIES  
	"/usr/local/zed/lib/libsl_zed.so"
	"/usr/local/zed/lib/libsl_depthcore.so"
	"/usr/local/zed/lib/libsl_calibration.so"
	"/usr/local/zed/lib/libsl_tracking.so"
	"/usr/local/zed/lib/libsl_disparityFusion.so"
	"/usr/local/zed/lib/libsl_svorw.so"
	"/usr/local/zed/lib/libsl_scanning.so"
	"/usr/local/zed/lib/libsl_core.so"
)

(Took out the pound signs for formatting's sake)

matlabbe, 2017-10-11:
Strange... I thought that `libcudpp.so` could come from `ZED_LIBRARIES`. I don't see any other dependencies adding `/usr/local/zed/lib/libcudpp.so` to linker, unless CMake finds another `zed-config.cmake` somewhere with the old libraries including `libcudpp.so`. If you want to debug, you may try showing what `ZED_LIBRARIES` contains in rtabmap build: just after this [line](https://github.com/introlab/rtabmap/blob/49f9a1e8d7f7fc4fda64c037f6117c288a95ee2f/CMakeLists.txt#L325), add this:
```
find_package(ZED 1 QUIET)
MESSAGE(STATUS "ZED_LIBRARIES =${ZED_LIBRARIES }")
```
and see on the terminal when you do "`cmake ..`" what are the actual values.

Update: While writing this, I noticed that the requirement version of ZED was still to 1. Please try updating with this https://github.com/introlab/rtabmap/commit/ab0aad87ec567a29db7ed5c721c59b75a73c44d5, this is maybe the cause why some ZED version v1 files still on the computer were taken instead of ZED v2.

cheers,
Mathieu

bmterry, 2017-10-11:
Still didnt compile, same issue.

Here are the libraries it wants to include

-- ZED_LIBRARIES =/usr/local/zed/lib/libsl_zed.so;/usr/local/zed/lib/libsl_depthcore.so;/usr/local/zed/lib/libsl_calibration.so;/usr/local/zed/lib/libsl_tracking.so;/usr/local/zed/lib/libsl_disparityFusion.so;/usr/local/zed/lib/libsl_svorw.so;/usr/local/zed/lib/libsl_scanning.so;/usr/local/zed/lib/libsl_core.so;/usr/local/zed/lib/libcudpp.so;/usr/local/zed/lib/libcudpp_hash.so

Sorry it takes so long to get back, compilation takes a while on the TX1, not to mention that im currently SSH ing through an SSH

matlabbe, 2017-10-11:
You don't need to build, we can now see the problem just after the "`$ cmake ..`":

_ZED_LIBRARIES =/usr/local/zed/lib/libsl_zed.so;/usr/local/zed/lib/libsl_depthcore.so;/usr/local/zed/lib/libsl_calibration.so;/usr/local/zed/lib/libsl_tracking.so;/usr/local/zed/lib/libsl_disparityFusion.so;/usr/local/zed/lib/libsl_svorw.so;/usr/local/zed/lib/libsl_scanning.so;/usr/local/zed/lib/libsl_core.so;**/usr/local/zed/lib/libcudpp.so**;/usr/local/zed/lib/libcudpp_hash.so_

In bold is the problem. It is related to `find_package(ZED 2 QUIET)`. These values are from the file called "**zed-config.cmake**" installed by ZED SDK. You may try to find all "zed-config.cmake" installed on the TX1 (if there are more than one), and delete/edit the one with **libcudpp.so**. After that, cleanup again the rtabmap build directory (at least CMakeCache.txt) to clear all cmake variables, then redo cmake.

cheers,
Mathieu

bmterry, 2017-10-11:
I had several that ZED decided to keep in backups when I upgraded, I removed the offending files and ran cmake again, this time **libcudpp.so** was not included.

Im making again right now.

bmterry, 2017-10-12:
It built correctly, now im just working on installing the ROS part (cant find -lopencv_dep_cudart), but that is out of the scope of this issue.

Thank you for your help!

closed 2018-02-02 by matlabbe