introlab/rtabmap #240: Errors Linking on Ubuntu 16.04

anilprasadmn, 2017-10-17:
rtabmap_core compiles but fails to link with VTK modules (I have version 7.1.1 installed and linked) My platform is Ubuntu  16.04

```

>> make rtabmap_core -j6
[ 13%] Built target rtabmap_utilite
[ 15%] Built target res_tool
[ 18%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: cannot find -lvtklz4
/usr/bin/ld: cannot find -lvtklibharu
/usr/bin/ld: cannot find -lvtknetcdfcpp
/usr/bin/ld: cannot find -lvtkFiltersParallelVerdict
/usr/bin/ld: cannot find -lvtkFiltersTopology
collect2: error: ld returned 1 exit status
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:2430: recipe for target '../bin/librtabmap_core.so.0.14.2' failed
make[3]: *** [../bin/librtabmap_core.so.0.14.2] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
CMakeFiles/Makefile2:279: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/rule' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/rule] Error 2
Makefile:223: recipe for target 'rtabmap_core' failed
make: *** [rtabmap_core] Error 2

```

matlabbe, 2017-10-23:
Did you try with default vtk version installed with pcl ?

anilprasadmn, 2017-10-27:
Thanks. Linking it with the default VTK version solved the issue.

closed 2017-10-27 by anilprasadmn