introlab/rtabmap #241: Segmentation fault on addNewWords

CelestialBoon, 2017-10-19:
rtabmap (standalone) has begun crashing on me after I've compiled and ran the RGB-D Mapping c++ example in the examples section. When the sample program is ran, it exits immediately after the first frame shows, citing a segmentation fault. From there onwards, rtabmap also crashes, when mapping (also crashes right after starting to map, as soon as 2 frames get registered and mapped out on screen) and also when postprocessing (just after pressing Ok on the menu). Below are the details of the installation, and the conditions surrounding a postprocessing crash (they are very similar to the mapping one from what I've seen).

The Cmake output:
```
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- Found OpenMP
-- Found OpenCV: /opt/ros/kinetic/include/opencv-3.2.0-dev;/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/x86_64-linux-gnu;/usr/include/python2.7;/usr/include/jsoncpp;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/hdf5/openmpi;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /usr/include
-- Found Freenect: /usr/include
-- Found freenect2: /usr/local/include
-- Found OpenNI2: /usr/include/openni2
-- Found g2o: /opt/ros/kinetic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found octomap: /opt/ros/kinetic/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   Version : 0.14.2
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = YES (License: Apache v2 and/or GPLv2)
--   With dc1394               = NO (dc1394 not found)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--   With libpointmatcher      = NO (libpointmatcher not found)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = NO (librealsense not found)
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/celestial/ext/rtabmap/build
[... 213 lines cut ...]
7fffaf8e8000-7fffaf8ec000 r--p 0005e000 08:01 44569173                   /lib/x86_64-linux-gnu/libssl.so.1.0.0
7fffaf8ec000-7fffaf8f3000 rw-p 00062000 08:01 44569173                   /lib/x86_64-linux-gnu/libssl.so.1.0.0
7fffaf8f4000-7fffaf934000 rw-p 00000000 00:00 0 
7fffaf934000-7fffaf949000 r-xp 00000000 08:01 16123877                   /usr/lib/python2.7/lib-dynload/_ssl.x86_64-linux-gnu.so
7fffaf949000-7fffafb48000 ---p 00015000 08:01 16123877                   /usr/lib/python2.7/lib-dynload/_ssl.x86_64-linux-gnu.so
7fffafb48000-7fffafb49000 r--p 00014000 08:01 16123877                   /usr/lib/python2.7/lib-dynload/_ssl.x86_64-linux-gnu.so
7fffafb49000-7fffafb4d000 rw-p 00015000 08:01 16123877                   /usr/lib/python2.7/lib-dynload/_ssl.x86_64-linux-gnu.so
7fffafb4d000-7fffafd4d000 rw-p 00000000 00:00 0 
7fffafd4d000-7fffafdc3000 r-xp 00000000 08:01 20847645                   /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.so
7fffafdc3000-7fffaffc3000 ---p 00076000 08:01 20847645                   /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.so
7fffaffc3000-7fffaffc4000 r--p 00076000 08:01 20847645                   /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.so
7fffaffc4000-7fffaffff000 rw-p 00077000 08:01 20847645                   /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.so
7fffaffff000-7fffb0000000 rw-p 00000000 00:00 0 
7fffb0000000-7fffb0021000 rw-p 00000000 00:00 0 
7fffb0021000-7fffb4000000 ---p 00000000 00:00 0 
7fffb4004000-7fffb4013000 rw-s 00000000 00:05 20971590                   /SYSV00000000 (deleted)
7fffb4013000-7fffb401b000 rw-s 00000000 00:05 718310                     /drm mm object (deleted)
7fffb401b000-7fffb4023000 rw-s 00000000 00:05 718302                     /drm mm object (deleted)
7fffb4023000-7fffb402b000 rw-s 00000000 00:05 718297                     /drm mm object (deleted)
7fffb402b000-7fffb406b000 rw-p 00000000 00:00 0 
7fffb406b000-7fffb4074000 r-xp 00000000 08:01 20847825                   /usr/lib/python2.7/dist-packages/numpy/fft/fftpack_lite.x86_64-linux-gnu.so
Thread 17 "rtabmap" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff7e7fc700 (LWP 26642)]
0x00007ffff4ad5428 in raise () from /lib/x86_64-linux-gnu/libc.so.6
```

matlabbe, 2017-10-19:
Is it with the latest code from source? It seems related to that fix https://github.com/introlab/rtabmap/commit/4de2ed767b8e920b39ccbb9bdb35b9ec6b40f088.

cheers

CelestialBoon, 2017-10-20:
It works! Thanks a bunch.

closed 2017-10-20 by CelestialBoon