introlab/rtabmap #242: how to build a map with stereo camera?

Shangareer, 2017-10-20:
When I build a map with ZED stereo camera,  it lose odometry quite often and can't build a good map. How to set the proper parameters to make it work well?

matlabbe, 2017-10-22:
Which rtabmap version are you using? On ros or standalone? Do you use Zed sdk driver (or zed_ros_wrapper) with default parameters? Do you scan mostly indoor or outdoor?

newhouse1991, 2017-10-23:
hey, I also have the same question, I download this project three days ago,  run standalone, when I use Zed or just stereo picture from Zed and parameters define almost as same as your .ini document in wiki, it works fine with your example picture, but when I used with my cam or picture， it's easy lost odometry and 3D map didn't look as well as some video in youtube, I'm very confused about this, I don't know why, can you tell me some point for using this, thank you for your answer. By the way I used mostly  outdoor but I have tested it indoor also.

matlabbe, 2017-10-23:
Can you record and share a sample database? Use "Tool->Data recorder" to record all frames.

Shangareer, 2017-10-24:
I'm using version 0.14.3 standalone without Zed sdk. I scan mostly outdoor. @matlabbe

matlabbe, 2017-10-24:
When zed sdk is not used, a good calibration is very important to get good results. Can youshare asmall databse to actially see if calo rqtion looks good?

newhouse1991, 2017-10-26:
@matlabbe I used zed sdk and got a better result, but not good enough as some video in youtube, Could I get a good enough calibration just use rtabmap? My datebase is a litter large, about 3G memory, so... ,   Alway, thank you for your generous answer，and by the way @Shangareer is my friend and work together. :)

matlabbe, 2017-10-26:
Can you link the video you try to reproduce? The zed sdk doesn't produce disparity images like opencv's cv::StereoBM, so the 3D reconstruction would not look similar than videos using custom calibration and OpenCV like with the Bumblebee2: https://www.youtube.com/watch?v=qpTS7kg9J3A.

Note also that you can re-use your 3G database as source to generate a smaller database.

magicdan19, 2021-04-11:
hello

magicdan19, 2021-04-11:
Hello every body, @matlabbe I from Quebec city. Can i exposed my problem in french? Thank you very much.

matlabbe, 2021-05-04:
@magicdan19  Yes, open a new issue.