introlab/rtabmap #246: run rtabmap segmentation fault (core dumped)

caydyn-skd, 2017-11-20:
Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
Ubuntu 16.04.3 LTS Linux DEV 4.4.0-101-generic #124-Ubuntu SMP Fri Nov 10 18:29:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
`
        caydyn@DEV:~$ ldd /usr/local/bin/rtabmap  
	linux-vdso.so.1 =>  (0x00007ffc48333000)
	librtabmap_gui.so.0.15 => /usr/local/lib/librtabmap_gui.so.0.15 (0x00007f215059c000)
	librtabmap_core.so.0.15 => /usr/local/lib/librtabmap_core.so.0.15 (0x00007f214fd1e000)
	librtabmap_utilite.so.0.15 => /usr/local/lib/librtabmap_utilite.so.0.15 (0x00007f214fafe000)
	libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007f214ee30000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f214e7a3000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f214e2cd000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f214df4b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f214dd35000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f214d96b000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f214d767000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f214d54a000)
	libpcl_common.so.1.8 => /usr/local/lib/libpcl_common.so.1.8 (0x00007f214d297000)
	libpcl_io.so.1.8 => /usr/local/lib/libpcl_io.so.1.8 (0x00007f214ce80000)
	libpcl_search.so.1.8 => /usr/local/lib/libpcl_search.so.1.8 (0x00007f214ca93000)
	libpcl_surface.so.1.8 => /usr/local/lib/libpcl_surface.so.1.8 (0x00007f214c3af000)
	libpcl_visualization.so.1.8 => /usr/local/lib/libpcl_visualization.so.1.8 (0x00007f214c06a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f214bd61000)
	libvtkIOMPIImage-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkIOMPIImage-6.2.so.6.2 (0x00007f214bb38000)
	libvtkRenderingMatplotlib-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.2.so.6.2 (0x00007f214b927000)
	libvtkInteractionImage-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkInteractionImage-6.2.so.6.2 (0x00007f214b700000)
	libvtkRenderingFreeTypeOpenGL-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeOpenGL-6.2.so.6.2 (0x00007f214b4ee000)
	libvtkViewsContext2D-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkViewsContext2D-6.2.so.6.2 (0x00007f214b2db000)
	libvtkRenderingFreeTypeFontConfig-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.2.so.6.2 (0x00007f214b0d1000)
	libQt5Svg.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007f2150dc3000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007f2150d50000)
	libopencv_calib3d3.so.3.2 => /opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2 (0x00007f214ad64000)
	libopencv_objdetect3.so.3.2 => /opt/ros/kinetic/lib/libopencv_objdetect3.so.3.2 (0x00007f214aaff000)
	libopencv_imgcodecs3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2 (0x00007f214a6db000)
	libopencv_imgproc3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.2 (0x00007f2148f24000)
	libvtkRenderingLOD-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2 (0x00007f2148d0a000)
	libvtkGUISupportQt-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQt-6.2.so.6.2 (0x00007f2148abf000)
	libvtkChartsCore-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkChartsCore-6.2.so.6.2 (0x00007f214876e000)
	libvtkRenderingOpenGL-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.2.so.6.2 (0x00007f21482c6000)
	libvtkInteractionStyle-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkInteractionStyle-6.2.so.6.2 (0x00007f214803a000)
	libvtkRenderingAnnotation-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingAnnotation-6.2.so.6.2 (0x00007f2147ce2000)
	libvtkRenderingFreeType-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.2.so.6.2 (0x00007f2147a1f000)
	libvtkRenderingCore-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkRenderingCore-6.2.so.6.2 (0x00007f214758b000)
	libvtkFiltersSources-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkFiltersSources-6.2.so.6.2 (0x00007f21472c5000)
	libvtkCommonExecutionModel-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonExecutionModel-6.2.so.6.2 (0x00007f2147017000)
	libvtkCommonDataModel-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.2.so.6.2 (0x00007f2146aa8000)
	libvtkCommonMath-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.2.so.6.2 (0x00007f2146886000)
	libvtkCommonCore-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so.6.2 (0x00007f2146358000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f2145e10000)
	libopencv_xfeatures2d3.so.3.2 => /opt/ros/kinetic/lib/libopencv_xfeatures2d3.so.3.2 (0x00007f2145971000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007f2145710000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f21454f6000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f21452f2000)
	libpcl_kdtree.so.1.8 => /usr/local/lib/libpcl_kdtree.so.1.8 (0x00007f2144fb9000)
	libpcl_filters.so.1.8 => /usr/local/lib/libpcl_filters.so.1.8 (0x00007f21445ec000)
	libpcl_features.so.1.8 => /usr/local/lib/libpcl_features.so.1.8 (0x00007f21420ef000)
	libpcl_sample_consensus.so.1.8 => /usr/local/lib/libpcl_sample_consensus.so.1.8 (0x00007f214129c000)
	libpcl_segmentation.so.1.8 => /usr/local/lib/libpcl_segmentation.so.1.8 (0x00007f2140356000)
	libOpenNI.so.0 => /usr/lib/libOpenNI.so.0 (0x00007f21400da000)
	libfreenect.so.0.5 => /opt/ros/kinetic/lib/libfreenect.so.0.5 (0x00007f213fecb000)
	librealsense.so.1 => /opt/ros/kinetic/lib/librealsense.so.1 (0x00007f213fbd0000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007f213f95a000)
	liboctomap.so.1.8 => /opt/ros/kinetic/lib/liboctomap.so.1.8 (0x00007f213f712000)
	liboctomath.so.1.8 => /opt/ros/kinetic/lib/liboctomath.so.1.8 (0x00007f213f50b000)
	libopencv_video3.so.3.2 => /opt/ros/kinetic/lib/libopencv_video3.so.3.2 (0x00007f213f187000)
	libopencv_features2d3.so.3.2 => /opt/ros/kinetic/lib/libopencv_features2d3.so.3.2 (0x00007f213eec8000)
	libvtkIOPLY-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkIOPLY-6.2.so.6.2 (0x00007f213ecad000)
	libopencv_photo3.so.3.2 => /opt/ros/kinetic/lib/libopencv_photo3.so.3.2 (0x00007f213e8c2000)
	libopencv_videoio3.so.3.2 => /opt/ros/kinetic/lib/libopencv_videoio3.so.3.2 (0x00007f213e699000)
	libvtkFiltersModeling-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkFiltersModeling-6.2.so.6.2 (0x00007f213e41c000)
	libvtkIOImage-6.2.so.6.2 => /usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2 (0x00007f213e0b5000)
[... 183 lines cut ...]
	libidn.so.11 => /usr/lib/x86_64-linux-gnu/libidn.so.11 (0x00007f2117bcc000)
	libtasn1.so.6 => /usr/lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007f21179b9000)
	libkrb5.so.3 => /usr/lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f21176e7000)
	libk5crypto.so.3 => /usr/lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f21174b8000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f21172b4000)
	libkrb5support.so.0 => /usr/lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f21170a9000)
	liblber-2.4.so.2 => /usr/lib/x86_64-linux-gnu/liblber-2.4.so.2 (0x00007f2116e9a000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f2116c7f000)
	libsasl2.so.2 => /usr/lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007f2116a64000)
	libgssapi.so.3 => /usr/lib/x86_64-linux-gnu/libgssapi.so.3 (0x00007f2116823000)
	libsz.so.2 => /usr/lib/x86_64-linux-gnu/libsz.so.2 (0x00007f2116620000)
	libltdl.so.7 => /usr/lib/x86_64-linux-gnu/libltdl.so.7 (0x00007f2116416000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f2116202000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f2115ffe000)
	libheimntlm.so.0 => /usr/lib/x86_64-linux-gnu/libheimntlm.so.0 (0x00007f2115df5000)
	libkrb5.so.26 => /usr/lib/x86_64-linux-gnu/libkrb5.so.26 (0x00007f2115b6b000)
	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007f21158c9000)
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007f2115696000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007f2115480000)
	libaec.so.0 => /usr/lib/x86_64-linux-gnu/libaec.so.0 (0x00007f2115278000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007f211504f000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f2114e40000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f2114bf5000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f21149bd000)
`

matlabbe, 2017-11-20:
What is the cmake config? (after `cmake ..` in rtabmap/build directory)

caydyn-skd, 2017-11-21:
-- Info :
--   Version : 0.15.0
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--   With g2o                  = YES (License: BSD)
--   With GTSAM                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = YES (License: GPLv2)
--   With libpointmatcher      = YES (License: BSD)
--   With ZED                  = NO (ZED sdk not found)
--   With RealSense            = YES (License: Apache-2)
--   With RealSenseSlam        = NO (WITH_REALSENSE_SLAM=OFF)
--   With OCTOMAP              = YES (License: BSD)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
--   With Qt5                  = YES (License: Open Source or Commercial)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/caydyn/soft/rtabmap/build

caydyn-skd, 2017-11-21:
GDB Info：

`
  Reading symbols from rtabmap...(no debugging symbols found)...done.
  (gdb) r
  Starting program: /usr/local/bin/rtabmap 
  [Thread debugging using libthread_db enabled]
  Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

  [New Thread 0x7fffb4b57700 (LWP 8350)]
  [New Thread 0x7fffb4356700 (LWP 8351)]
  [New Thread 0x7fffafb55700 (LWP 8352)]

  Thread 1 "rtabmap" received signal SIGSEGV, Segmentation fault.
  0x00007fffc6edc5a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
  (gdb) 
  (gdb) bt
  #0  0x00007fffc6edc5a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
  #1  0x00007ffff7de76ba in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffd828, env=env@entry=0x7fffffffd838) at dl-init.c:72
  #2  0x00007ffff7de77cb in call_init (env=0x7fffffffd838, argv=0x7fffffffd828, argc=1, l=<optimized out>) at dl-init.c:30
  #3  _dl_init (main_map=0x7ffff7ffe168, argc=1, argv=0x7fffffffd828, env=0x7fffffffd838) at dl-init.c:120
  #4  0x00007ffff7dd7c6a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
  #5  0x0000000000000001 in ?? ()
  #6  0x00007fffffffdc39 in ?? ()
  #7  0x0000000000000000 in ?? ()
`

matlabbe, 2017-11-21:
Thanks for the gdb log, it seems related to Qt, when creating windows. The cmake log says it is building against Qt5, but the error is in Qt4 version of the library. It looks like on runtime libQtGui.so.4 is used instead of libQt5Widgets.so.5, which will cause the problem. Not sure why the linker, if built with Qt5, would open Qt4 libraries, unless rtabmap has been built with a Qt version, then other Qt version has been installed, then rtabmap rebuilt without cleaning the building directory. Can you make sure you have only one version of Qt installed at the same time? Normally, cmake would manage both versions at the same time but just to make sure.

cheers,
Mathieu

caydyn-skd, 2017-11-22:
@matlabbe  Thank you very much

closed 2017-11-22 by caydyn-skd