introlab/rtabmap #247: Occupancy grid nodes missaligned after reloading rtabmap with GridGlobal/FullUpdate set to true (ROS)

tentone, 2017-12-06:
Hi

When i have the GridGlobal/FullUpdate parameter set to true (which does improve a lot the map quality a lot) after closing and loading again the map the OccupancyGrid gets deformed back to the odom position where the data was captured.

Eventually after running for a while it gets aligned back together.

This is the grid when the mapping finishes.

[image]

This is the grid after reloading it.

[image]

Is there any way to fix this and have it loading properly, im trying to load it so that i can run mapping phase with GridGlobal/FullUpdate set to true but have it set to false when running localization only mode.

Thanks a lot all the work and for the help!

Cumps

matlabbe, 2017-12-06:
`GridGlobal/FullUpdate` should not affect map optimization, just its rendering. When you "reload", is it just after rtabmap is relaunched in localization mode?

I tried to reproduce the problem using the [demo_mapping.bag](http://wiki.ros.org/rtabmap_ros#Robot_mapping_with_RVIZ):
```bash
Mapping phase:
$ roslaunch rtabmap_ros demo_robot_mapping.launch rviz:=true rtabmapviz:=false
$ rosbag play --clock demo_mapping.bag

Localization phase:
$ roslaunch rtabmap_ros demo_robot_mapping.launch rviz:=true rtabmapviz:=false localization:=true
Publish the map to rviz:
$ rosservice call /rtabmap/publish_map  "global_: true
optimized: true
graphOnly: false"
```
Th map looks okay:
[image]

If I publish not optimized:
```bash
$ rosservice call /rtabmap/publish_map  "global_: true
optimized: false
graphOnly: false"
```
[image]

Can you share the database?

cheers,
Mathieu

tentone, 2017-12-07:
Hi @matlabbe Thanks a lot i added the optimized parameter to my launch file and that resolved the "problem".

Thanks a lot, was my fault!

tentone, 2017-12-07:
Well, actually this seems to still happen when i enable robust optimization.

When the mapping process finishes this is the result.

[image]

After closing and rtab and reloading the map for localization (used the service indicated above)

[image]

The db for this map can be donwloaded from  here
https://mega.nz/#!2lxkTCIT!rnZ_7WWok14or21GRHmsi7aqvTQohsNvONkaUpkUg1Q

And here is my launch file

```
<launch>
	<arg name="localization"  default="false"/>
	<arg name="database"      default="/home/tentone/map.db"/>

	<!-- RTAB args -delete_db_on_start to remove old data -->
	<arg if="$(arg localization)"     name="rtabmap_args"  default=""/>
	<arg unless="$(arg localization)" name="rtabmap_args"  default="--delete_db_on_start"/>

	<!-- Frame ID -->
	<arg name="frame_id"               default="base_link"/>
	<arg name="odom_frame_id"          default="odom"/>
	<arg name="map_frame_id"           default="map"/>
	<arg name="camera_frame_id"        default="rgbd_front_top_link"/>

	<!-- Wiigo V1.5 -->
	<arg name="topic_rgb_image"        default="/camera/rgb/image_raw"/>
	<arg name="topic_depth_image"      default="/camera/depth/image_raw"/>
	<arg name="topic_rgb_camera_info"  default="/camera/rgb/camera_info"/>
	<arg name="topic_odom"             default="/wgr/odometry/odom"/>
	<arg name="topic_scan"             default="/scan"/>
	<arg name="topic_map"              default="/rtab_map"/>
	
	<!-- RTAB map group -->
	<group ns="rtabmap">
		<!-- Mapping node -->
		<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg rtabmap_args)">
			<!-- Input -->
			<remap from="rgb/image"        to="$(arg topic_rgb_image)"/>
			<remap from="rgb/camera_info"  to="$(arg topic_rgb_camera_info)"/>
			<remap from="depth/image"      to="$(arg topic_depth_image)"/>
			<remap from="odom"             to="$(arg topic_odom)"/>
			<remap from="scan"             to="$(arg topic_scan)"/>

			<!-- Output -->
			<remap from="grid_map"         to="$(arg topic_map)"/>

			<!-- General settings -->
			<param name="database_path"                   type="string"    value="$(arg database)"/>
			<param name="frame_id"                        type="string"    value="$(arg frame_id)"/>
			<param name="odom_frame_id"                   type="string"    value="$(arg odom_frame_id)"/>
			<param name="map_frame_id"                    type="string"    value="$(arg map_frame_id)"/>
			<param name="queue_size"                      type="int"       value="10"/>
			<param name="use_sim_time"                    type="bool"      value="false"/>
			<param name="subscribe_depth"                 type="bool"      value="true"/>
			<param name="subscribe_scan"                  type="bool"      value="true"/>
			<param name="gen_scan"                        type="bool"      value="false"/>
			<param name="wait_for_transform"              type="bool"      value="true"/>
			<param name="wait_for_transform_duration"     type="double"    value="0.2"/>
			<param name="rgbd_odometry"                   type="bool"      value="false"/>
			<param name="compressed"                      type="bool"      value="false"/>
			<param name="approx_sync"                     type="bool"      value="true"/>
			<param name="latch"                           type="bool"      value="true"/>
			<param name="optimized"                       type="bool"      value="true"/>
			<param name="graphOnly"                       type="bool"      value="false"/>
[... 91 lines cut ...]
			<param name="Grid/ProjRayTracing"             type="bool"       value="true"/>

			<!-- Grid global settings -->
			<param name="GridGlobal/FullUpdate"           type="bool"       value="true"/>
			<param name="GridGlobal/Eroded"               type="bool"       value="false"/>
			<param name="GridGlobal/FootprintRadius"      type="double"     value="0.0"/>

			<!-- Memory Management -->
			<param name="Mem/UseOdomFeatures"             type="bool"       value="false"/>
			<param name="Mem/RehearsalSimilarity"         type="string"     value="0.45"/>
			<param name="Mem/ImageKept"                   type="bool"       value="false"/>
			<param name="Mem/BinDataKept"                 type="bool"       value="true"/>
			<param name="Mem/RawDescriptorsKept"          type="bool"       value="true"/>
			<param name="Mem/NotLinkedNodesKept"          type="bool"       value="true"/>
			<param name="Mem/IntermediateNodeDataKept"    type="bool"       value="false"/>
			<param name="Mem/STMSize"                     type="int"        value="10"/>

			<!-- Localization mode -->
			<param name="Mem/InitWMWithAllNodes" type="bool"                   value="$(arg localization)"/>
			<param if="$(arg localization)"      name="Mem/IncrementalMemory"  type="string" value="false"/>
			<param unless="$(arg localization)"  name="Mem/IncrementalMemory"  type="string" value="true"/>
		</node>
	</group>
</launch>
```

matlabbe, 2017-12-07:
Thx for sharing the database, it is easier to debug. This is mostly caused by g2o optimization. I tried different optimization approaches.
1- Your parameters:
[image]

2- Removing Robust (`Optimizer/Robust=false`):
[image]

3- Using TORO optimization and 100 iterations (`Optimizer/Strategy=0`, `Optimizer/Iterations=100`):
[image]

4- Using GTSAM optimization (`Optimizer/Strategy=2`)::
[image]

5- Changing the optimizer algorithm used by g2o to Gauss-Newton (`g2o/Optimizer=1`): 
[image]

6- Same like 5 but without Robust (`g2o/Optimizer=1`, `Optimizer/Robust=false`)
[image]

Note also the Time optimization on the bottom left for each approach. 6 is faster than 5, but I am not sure which one looks better optimized. Note that using `RGBD/OptimizeMaxError=0.1` could replace `Optimizer/Robust` to get better time performance while still rejecting most bad loop closures.

cheers,
Mathieu

tentone, 2017-12-12:
Hi

I have tested some of these configuration, all of them work nice as well but when i reload the ROS node if robust is set true the map the map load ignoring the odom corrections applied.

But if after mapping i disable robust and reload the map is loads better, not as good as when it was created but usable.

When mapping finished.

[image]

After reloading with robust set to true 

[image]

If i set robust to false and maxerror to 0.1 and reload the same db

[image]

Thanks a lot for your help.

matlabbe, 2017-12-12:
Note that above with g2o graph optimization, you can try "g2o/Optimizer=1" without robust to get better results. The max error doesn't affect the actual map when reloaded, only when mapping online.

tentone, 2017-12-13:
I changed the Optimizer to  GaussNewton again as you said and it works fine now.

Thanks a lot for you help!

Keep up with this awesome work!

closed 2017-12-13 by tentone