introlab/rtabmap #248: Issue running Rtabmap C++ example

BrunoEduardoCSantos, 2017-12-07:
Hi,
I am running the example of RGBD in C++  on Ubuntu 16.04, and I am getting the following error 
_make[2]: *** No rule to make target 'MapBuilder.h', needed by 'moc_MapBuilder.cpp'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rgbd_mapping.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_mapping.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2_

I have no errors installing rtabmap , but I can't understand what is the issue .
When I run the cmake the following message shows:

 Eigen found (include: /usr/local/include/eigen3, version: 3.3.3)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- OpenNI2 found (include: /usr/local/include/ni2, lib: /usr/local/lib/ni2)
-- Could NOT find ENSENSO (missing: ENSENSO_LIBRARIES ENSENSO_INCLUDE_DIRS) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find davidSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: DSSDK_LIBRARIES DSSDK_INCLUDE_DIRS) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

** WARNING ** io features related to libusb-1.0 will be disabled
-- FLANN found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp.so;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- FLANN found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp.so;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- OpenNI2 found (include: /usr/local/include/ni2, lib: /usr/local/lib/ni2)
-- Could NOT find ENSENSO (missing: ENSENSO_LIBRARIES ENSENSO_INCLUDE_DIRS) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find davidSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: DSSDK_LIBRARIES DSSDK_INCLUDE_DIRS) 
** WARNING ** visualization features related to dssdk will be disabled
-- Could NOT find RSSDK (missing: RSSDK_LIBRARIES RSSDK_INCLUDE_DIRS) 
** WARNING ** visualization features related to rssdk will be disabled
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_2D
-- looking for PCL_GEOMETRY
-- looking for PCL_FEATURES
-- looking for PCL_ML
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_SURFACE
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_TRACKING
-- looking for PCL_RECOGNITION
-- looking for PCL_STEREO
-- looking for PCL_APPS
-- looking for PCL_OUTOFCORE
-- looking for PCL_PEOPLE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/brunoeducsanos/eclipse-workspace/CartAttention

Any guidance you can give?

matlabbe, 2017-12-07:
What is the CMakeLists.txt? Make sure you are creating the qt moc for the header like [here](https://github.com/introlab/rtabmap/blob/5f5c74f260098acb8681c004561964fb02263fcc/examples/RGBDMapping/CMakeLists.txt#L63-L67).

BrunoEduardoCSantos, 2017-12-11:
Hi,
CMakeLists is cmake file to import dependencies.
I create the qt moc as you mentioned. 
I am using eclipse. Is there a easy way to run the example?
Or is there other IDE easier to run it?

matlabbe, 2017-12-11:
The easiest way to run the example is:
```
$ cd rtabmap/build
$ cmake ..
$ make
$ cd ../bin
$ ./rtabmap-rgbd_mapping
```

With eclipse, after the `cmake ..` step above, I often create an empty makefile project, set the make directory to build, and make with eclipse.

closed 2017-12-11 by BrunoEduardoCSantos